espressif/arduino-esp32 #4970: How to write batch UUID to a device

asuaka, 2021-03-23:
During mass production, it is necessary to write a batch of UUID to the equipment. How can it be realized on ARDUINO IDE?

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.5
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                      no
Upload Speed:					115200
Computer OS:                                          Windows 10

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

sansillusion, 2021-03-23:
I think you could generate a random one on first boot and store it in
nvs/littlefs or even better use the preferences library to store and
retreive it.

lbernstone, 2021-03-23:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_partition_gen.html

closed 2021-07-11 by stale[bot]