espressif/arduino-esp32 #4972: Update Preferences.cpp to include nvs_commit() on three methods

beamholder, 2021-03-23:
In [Preferences.cpp](https://github.com/espressif/arduino-esp32/blob/master/libraries/Preferences/src/Preferences.cpp),  the functions:
```
Preferences::clear()
Preferences::remove()
Preferences::end()
```
should be revised to include a call to 
`nvs_commit()`
as required per 
[Non-volatile storage library](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_flash.html)
when using 
```
nvs_erase_all()
nvs_erase_key()
nvs_close()
```
Suggested changes:

```
void Preferences::end(){                                         // modified to add an nvs_commit()
    if(!_started){
        return;
    }
    esp_err_t err = nvs_commit(_handle);                         // to undo changes: delete the lines from here...
    if(err){
        log_e("nvs_commit fail: %s %s", key, nvs_error(err));
    }                                                            // to here.
    nvs_close(_handle);
    _started = false;
}
```

```
bool Preferences::clear(){                                  // modified to add an nvs_commit()
    if(!_started || _readOnly){
        return false;
    }
    esp_err_t err = nvs_erase_all(_handle);
    if(err){
        log_e("nvs_erase_all fail: %s", nvs_error(err));
        return false;
    }
    // return true;                                         // to undo changes: uncomment this line and...
    err = nvs_commit(_handle);                              // delete the lines from from here...
    if(err){
        log_e("nvs_commit fail: %s %s", key, nvs_error(err));
        return false;
    }
    return true;                                            // to here.
}
```

```
bool Preferences::remove(const char * key){                 // modified to add an nvs_commit()
    if(!_started || !key || _readOnly){
        return false;
    }
    esp_err_t err = nvs_erase_key(_handle, key);
    if(err){
        log_e("nvs_erase_key fail: %s %s", key, nvs_error(err));
        return false;
    }
    // return true;                                         // to undo changes: uncomment this line and...
    err = nvs_commit(_handle);                              // delete the lines from from here...
    if(err){
        log_e("nvs_commit fail: %s %s", key, nvs_error(err));
        return false;
    }
    return true;                                            // to here.
}
```

Xylopyrographer, 2021-04-12:
Correction to my suggested changes for the end() and clear() methods to fix a coding error in writing to log_e().
```
void Preferences::end(){                                         // modified to add an nvs_commit()
    if(!_started){
        return;
    }
    esp_err_t err = nvs_commit(_handle);                         // to undo changes: delete the lines from here...
    if(err){
        log_e("nvs_commit fail: %s", nvs_error(err));            // 2021-04-12: corrected this line 
    }                                                            // to here.
    nvs_close(_handle);
    _started = false;
}
```
and
```
/*
 * Clear all keys in opened preferences
 * */

bool Preferences::clear(){                                  // modified to add an nvs_commit()
    if(!_started || _readOnly){
        return false;
    }
    esp_err_t err = nvs_erase_all(_handle);
    if(err){
        log_e("nvs_erase_all fail: %s", nvs_error(err));
        return false;
    }
    // return true;                                         // to undo changes: uncomment this line and...
    err = nvs_commit(_handle);                              // delete the lines from from here...
    if(err){
        log_e("nvs_commit fail: %s", nvs_error(err));       // 2021-04-12: corrected this line
        return false;
    }
    return true;                                            // to here.
}

```

Xylopyrographer, 2021-06-19:
Bump.

What is the review process? Thanks.

lbernstone, 2021-06-19:
You closed the pull request.  The review process only happens on active pull requests :smile:

beamholder, 2021-06-19:
Argh! Can  you tell I’m new to GitHub? 😬

closed 2021-09-19 by stale[bot]