espressif/arduino-esp32 #4973: 1.0.5 source code execution on ESP32 much slower than with 1.0.4

corrados, 2021-03-24:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			        1.0.5
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                       no
Upload Speed:					        115200
Computer OS:                                           Windows 10

### Description:
I am developing an open source edrum module (source code at Github: https://github.com/corrados/edrumulus) on my ESP32 dev module (NodeMCU with dual core ESP32). For that I need to sample from multiple ADC at a rate of 8 kHz and do some digital signal processing. I have worked with version 1.0.4 and could successfully process up to 7 channels. After upgrading to version 1.0.5, now I cannot even process just one single channel. So, the same source code executes much slower now.

Basically, I am using a "hw_timer_t" and check with "hw_timer_t" to get a constant sampling rate. Then I am using analogRead(), the MIDI-library and some floating point calculations.

Gamelauncher, 2021-04-06:
Having the same issue unfortunately. 

Hardware:

Board: ESP32 Dev Module
Core Installation version: 1.0.5 & 1.0.6
IDE name: Arduino IDE
Flash Frequency: 40Mhz
PSRAM enabled: no
Upload Speed: 115200
Computer OS: Mac

Loaded Libs:
AsyncTcp
AsyncMqtt
ESPAsyncWebserver
UART driver for RS485 @ 9600
UART driver for RS232 @ 57600

There is huge difference after 1.0.4 where the slowdowns result in panics and corrupt heap. Been searching for ages for a solution as I thought it would be in one of the loaded libs. Reverting back to 1.0.4 makes a huge speed difference in for example mqtt sending, but also all the panics and crashes have disappeared. As soon I compile against 1.0.5, within 15 minutes it crashes again.

```CORRUPT HEAP: Bad tail at 0x3fff0948. Expected 0xbaad5678 got 0x00000000
abort() was called at PC 0x40085c79 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x400889dc:0x3ffe1440 0x40088c59:0x3ffe1460 0x40085c79:0x3ffe1480 0x40085da5:0x3ffe14b0 0x4010746f:0x3ffe14d0 0x40102b79:0x3ffe1790 0x40102a69:0x3ffe17e0 0x4008e0d5:0x3ffe1810 0x400863ba:0x3ffe1830 0x40085b71:0x3ffe1850 0x4000bec7:0x3ffe1870 0x401504c7:0x3ffe1890 0x4015054b:0x3ffe18b0 0x40150587:0x3ffe18d0 0x40152a87:0x3ffe18f0 0x4015fb9c:0x3ffe1910 0x40160a98:0x3ffe1930 0x401614fe:0x3ffe1950 0x4015a0cf:0x3ffe1970 0x4015e9c2:0x3ffe19a0 0x40149fd3:0x3ffe19c0 0x4008a8ba:0x3ffe19f0```

```0x400889dc: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 156
0x40088c59: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 171
0x40085c79: lock_acquire_generic at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c line 143
0x40085da5: _lock_acquire_recursive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c line 171
0x4010746f: _vfiprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 860
0x40102b79: fiprintf at ../../../.././newlib/libc/stdio/fiprintf.c line 50
0x40102a69: __assert_func at ../../../.././newlib/libc/stdlib/assert.c line 59
0x4008e0d5: multi_heap_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c line 218
0x400863ba: heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c line 268
0x40085b71: _free_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c line 42
0x401504c7: mem_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/mem.c line 176
0x4015054b: do_memp_free_pool at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/memp.c line 432
0x40150587: memp_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/memp.c line 489
0x40152a87: tcp_seg_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp.c line 1410
0x4015fb9c: tcp_receive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp_in.c line 1192
0x40160a98: tcp_process at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp_in.c line 878
0x401614fe: tcp_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp_in.c line 372
0x4015a0cf: ip4_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c line 750
0x4015e9c2: ethernet_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/netif/ethernet.c line 184
0x40149fd3: tcpip_thread at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c line 135
0x4008a8ba: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143```

corrados, 2021-04-06:
For my project I found out that it is just the analogRead() function which is much slower in version 1.0.5 and also 1.0.6. As far as I understand, this is the corresponding change in the library: https://github.com/espressif/arduino-esp32/pull/3377.

As a workaround, I have written my own my_analogRead() function now.

simap, 2021-04-07:
for what it's worth, I'm seeing some floating point performance issues with 1.0.5.

Gamelauncher, 2021-04-07:
I am not using analogRead, only 3 digital reads. It's hard to tell where it's coming from. I've just disabled ESPAsyncWebserver and that makes it run more stable, but if in turn I just disable AsyncMQTT, it also runs more stable. But running both will make it crash. I've patched all the libs with the fixes I could find and not able to pinpoint an exact cause. This problem has costed me 2 weeks already and I am glad with 1.0.4 it works perfect. When the project is finished, I'll try to revisit this and see where the hiccups are coming from.

lbernstone, 2021-04-07:
Please keep this on topic.  The issue is about slowdowns from analogRead, not crashes (likely memory issues).  OP has posted a workaround.

martenjacobs, 2021-04-18:
> For my project I found out that it is just the analogRead() function which is much slower in version 1.0.5 and also 1.0.6. As far as I understand, this is the corresponding change in the library: #3377.
> 
> As a workaround, I have written my own my_analogRead() function now.

Just FYI for people stumbling upon this issue: workaround code is [here](https://github.com/corrados/edrumulus/blob/a5393269dba640ceba905957f61102733bad6574/edrumulus_esp32.cpp#L237)

bxparks, 2021-04-18:
I just observed the same when I upgraded from v1.0.4 to v1.0.6. The benchmarks for one of my libraries say that `analogRead()` became 2.5 - 3.5X slower (numbers below in microseconds, mostly due to one call to `analogRead()`):
```
$ git diff
 +---------------------------------------------------+
 | Button Event              | min/avg/max | samples |
 +---------------------------+-------------+---------+
...
-| LadderButtonConfig        |  21/ 24/ 76 | 2002    |
+| LadderButtonConfig        |  64/ 67/266 | 2002    |
```

Interestingly, the new `analogRead()` also consumes about 14kB more flash memory, but 400 bytes fewer static RAM:
```
$ git diff
 +--------------------------------------------------------------+
 | functionality                   |  flash/  ram |       delta |
 |---------------------------------+--------------+-------------|
-| Baseline                        | 206625/14564 |     0/    0 |
+| Baseline                        | 197914/13092 |     0/    0 |
...
-| LadderButtonConfig              | 211773/15452 |  5148/  888 |
+| LadderButtonConfig              | 216918/13504 | 19004/  412 |
```

Fortunately, I need a sampling rate of only about 200Hz, so I don't have any problems. But I can see why a sampling rate of 8kHz would no longer work. That would require `analogRead()` to be faster than 125 micros, which it cannot seem to do anymore.

closed 2021-07-11 by stale[bot]

ch-fb, 2021-08-23:
I have the same problem. 1.0.4 is fast. after upgrading, 1.0.5 and 1.0.6 super slow. I don't have analogRead in my loop(). In the meantime, has anybody figured out what causes the problem?