espressif/arduino-esp32 #4974: BluetoothSerial stops to respond after ESP_SPP_CLOSE_EVT event finished

ELazarMosk, 2021-03-25:
The basic ESP32 Arduino Bluetooth Serial example pairs and connects as expected, BUT once I disconnect the Bluetooth terminal it stops responding to any new events at all, and I can not reconnect anymore.

once I reset the esp32, it starts working again with the same problem. (it doesn't matter if it keeps being paired or not)

I have tested it with the windows Bluetooth terminal and with the Bluetooth Any terminal. (from the windows store)

I have downloaded the latest esp32-Arduino library from GitHub and I still get the same problem.

I have tested it with several esp32 modules, and they all respond with the same problem.

So it looks like a bug in the library code.

See the simple Arduino example code below:

#include "BluetoothSerial.h"

#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
#error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
#endif

BluetoothSerial SerialBT;

void setup() {
  Serial.begin(115200);
  SerialBT.begin("ESP32test"); //Bluetooth device name
  Serial.println("The device started, now you can pair it with bluetooth!");
}

void loop() {
  if (Serial.available()) {
    SerialBT.write(Serial.read());
  }
  if (SerialBT.available()) {
    Serial.write(SerialBT.read());
  }
  delay(20);
}

lbernstone, 2021-03-25:
Likely dupe https://github.com/espressif/arduino-esp32/issues/4915

ELazarMosk, 2021-03-25:
This should also be fixed when installing the ESP32 boards from the Arduino IDE

ELazarMosk, 2021-03-25:
However,

I still have issues with installing it with the instruction in:
https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md

I did everything in the tutorial and I get this err in the Arduino IDE

Arduino: 1.8.13 (Windows Store 1.8.42.0) (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

Warning: Board breadboard:avr:atmega328bb doesn't define a 'build.board' preference. Auto-set to: AVR_ATMEGA328BB

exec: "C:\\Users\\elaza\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++": file does not exist

Error compiling for board ESP32 Dev Module.

Invalid library found in C:\Users\elaza\Documents\Arduino\libraries\AzureIoT: no headers files (.h) found in C:\Users\elaza\Documents\Arduino\libraries\AzureIoT

lbernstone, 2021-03-25:
https://github.com/espressif/arduino-esp32/issues/4915#issuecomment-795615800

closed 2021-07-11 by stale[bot]