espressif/arduino-esp32 #4975: Once exited the light sleep mode the WiFi does not reconnect.

YordanYanakiev, 2021-03-25:
### Hardware:
Board:							Lily Go T-Watch 2020 V1 
Core Installation version:			1.0.5
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                      yes
Upload Speed:					115200
Computer OS:                                          Windows 8.1

### Description:
I am trying to put the watch in light sleep mode, and each 15 seconds to waking it up, and check if there is a web resource available, so I can get it. Anyway only the first time the WiFi is connecting. Once it enter for the first time in light sleep mode - it can not reconnect anymore.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

#include "WiFi.h"
#include <WiFiMulti.h>
#include <HTTPClient.h>
#include "esp_wifi.h"
#include <HTTPClient.h>
 
#define uS_TO_S_FACTOR 1000000ULL  /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP  5        /* Time ESP32 will go to sleep (in seconds) */
 
void print_wakeup_reason()
{
    esp_sleep_wakeup_cause_t wakeup_reason;
    wakeup_reason = esp_sleep_get_wakeup_cause();
 
    switch( wakeup_reason )
    {
    case ESP_SLEEP_WAKEUP_EXT0: Serial.println( "Wakeup caused by external signal using RTC_IO" ); break;
    case ESP_SLEEP_WAKEUP_EXT1: Serial.println( "Wakeup caused by external signal using RTC_CNTL" ); break;
    case ESP_SLEEP_WAKEUP_TIMER: Serial.println( "Wakeup caused by timer" ); break;
    case ESP_SLEEP_WAKEUP_TOUCHPAD: Serial.println( "Wakeup caused by touchpad" ); break;
    case ESP_SLEEP_WAKEUP_ULP: Serial.println( "Wakeup caused by ULP program" ); break;
    default: Serial.printf( "Wakeup was not caused by deep sleep: %d\n", wakeup_reason ); break;
    }
}
 
void setup()
{
    Serial.begin( 9600 );
    while( !Serial )
    {
        ; // wait for serial port to connect. Needed for native USB
    }
    setupNetwork();
 
    setCpuFrequencyMhz( 20 );
 
    delay( 10 );
}
 
void loop()
{
    print_wakeup_reason(); //Print the wakeup reason for ESP32
 
    delay( 100 );
    
 
    esp_sleep_enable_timer_wakeup( TIME_TO_SLEEP * uS_TO_S_FACTOR ); // ESP32 wakes up every 5 seconds
 
    Serial.println( "Going to light-sleep now" );
    Serial.flush();
 
    WiFi.disconnect( true );
    delay( 100 );
    esp_light_sleep_start();
 
    setupNetwork();
}
 
void setupNetwork()
{
    //WiFi.disconnect();
    WiFi.enableSTA( true );
    WiFi.mode( WIFI_STA );
    //WiFi.persistent( false );
    delay( 10 );
 
    WiFi.onEvent( []( WiFiEvent_t event, WiFiEventInfo_t info )
        {
        //wifi_connect_status(true);
        }, WiFiEvent_t::SYSTEM_EVENT_STA_GOT_IP );
 
    delay( 10 );
 
    WiFi.begin( "UnrSoft", "AZRTTT377" );
    while( WiFi.status() != WL_CONNECTED )
    {
        delay( 500 );
        Serial.println( "Connecting to WiFi.. > status : " + String( WiFi.status() ) );
    }
 
  
    delay( 10 );
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

felmue, 2021-03-26:
Hello @YordanYanakiev 

I do not have the exact hardware, but running your code on an M5Core2 works just fine - it reconnects properly after light sleep.
I've also modified your code a bit and instead of calling `setupNetwork()` after light sleep I only enable WiFi and do a reconnect. Maybe that helps in your case.
```
    WiFi.enableSTA( true );
    WiFi.mode( WIFI_STA );
    WiFi.reconnect();
    while( WiFi.status() != WL_CONNECTED )
    {
        delay( 500 );
        Serial.println( "Connecting to WiFi.. > status : " + String( WiFi.status() ) );
    }
```
This variant works for me too.
The only other factor that comes to mind is the router which might not like such a short disconnect / reconnect time? Do you see the same issue when you put your device into light sleep a bit longer?

Thanks
Felix

felmue, 2021-03-26:
Hello @YordanYanakiev 

the issue seem to be the lowering of the CPU frequency. If I uncomment `setCpuFrequencyMhz( 20 );` your code works for me. If you really want to lower the CPU frequency before light sleep you could try to restore it after light sleep and before reconnecting.

Thanks
Felix

YordanYanakiev, 2021-03-26:
Google freakish lords! 
Love You man ! <3 
This thing cost me almost 2 days looking stupid! 
Thank You !

closed 2021-03-26 by YordanYanakiev

felmue, 2021-03-26:
Hi @YordanYanakiev 

you are welcome. BTW: things like that happen to most of us at some time.

Cheers
Felix