espressif/arduino-esp32 #4980: After 1.0.6 update WiFi won't connect to other network

handmade0octopus, 2021-03-26:
Hello.

Using ESP32 with arduino.

Below snippet was working ok on all previous versions. I updated today to 1.0.5 and now it won't connect to my home network, always creates Access Point instead.

```cpp
WiFi.begin(wifiSSID.c_str(), wifiPSWD.c_str());
	while(WiFi.status() != WL_CONNECTED) {
		if(millis() - startTime > 2000) break;	
		delay(100);
		tft.drawCentreString(CONNECTING_WIFI, OFFSET_WIDTH, OFFSET_HEIGHT+WIFI_OFFSET_HEIGHT, 1);
	}
	if(WiFi.status() != WL_CONNECTED) {
		WiFi.disconnect(true);
		WiFi.mode(WIFI_AP);
		delay(100);
//		WiFi.softAPConfig(local_ip, gateway, subnet);
		WiFi.softAP(wifiSSID.c_str(), wifiPSWD.c_str());
		delay(500);
		IPAddress IP = WiFi.softAPIP();
		tft.setCursor(10, 230);
		tft.println(IP);
		wifiOn = false;
	} else {
		wifiOn = true;
		tft.setCursor(10, 230);
		tft.println(WiFi.localIP());
	}
```

Reverted back to 1.0.5 and 1.0.6, power cycled device many times and nothing helps. Once I use 1.0.5 everything works as it should.

me-no-dev, 2021-03-26:
can you enable debug to Verbose and post the log? WiFi is working fine on my end

handmade0octopus, 2021-03-26:
Sure:

1.0.5 log:
```
okoûoûçïÆoûok[E][vfs_api.cpp:64] open(): /sd/update.bin does not exist
        
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
              
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
               
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 4 - STA_CONNECTED
           
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
              
[D][WiFiGeneric.cpp:419] _eventCallback(): STA IP: 192.168.1.79, MASK: 255.255.2
55.0, GW: 192.168.1.254
                                                       
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 3 - STA_STOP
                
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 3 - STA_STOP
    
    ```
    
    
And here is 1.0.6 and only AP working ok:
```
[E][vfs_api.cpp:64] open(): /sd/update.bin does not ex
ist
                                                                           
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
              
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
               
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 14 - AP_START
               
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
              
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 14 - AP_START
               
                                                                                
                                                                                
                                                                                
                                                      
                                                      ```

me-no-dev, 2021-03-26:
I think you should wait a bit longer than 2 seconds :) logs does not show that connect failes

handmade0octopus, 2021-03-26:
> I think you should wait a bit longer than 2 seconds :) logs does not show that connect failes

Fair enough but why I NEVER had any problems with 2s waiting time?

Never ever had any issue and I do it this FW for over a year. Now with 1.0.6 it NEVER connects within this 2s period.

What you changed in the code it lost its performance? I think this is worth finding out to be honest. I am happy to run some tests if you want.

Changing to 3000 "solved" issue but I call it a bodge as performance loss of 1s is big thing don't you think?

I ran some benchmarks and on 1.0.6 I got between 2200-2420ms

On 1.0.5 on other hand I have 220ms - 1100ms (while most being around 260-330ms!)

What is the reason behind this huge performance loss?

ThomasWaldmann, 2021-03-28:
https://github.com/ecocurious2/MultiGeiger/issues/408 we also encounter wifi issues with 1.0.6.

ThomasWaldmann, 2021-03-28:
@handmade0octopus your top post says  `"I updated today to 1.0.5"` - I guess you meant 1.0.6 (see issue title). Can you edit the top post and fix that?

zasnicoff, 2021-04-06:
1.0.6 also broke my code, not much to report at this point, but I'll keep you posted.
PlatformIO, upgraded espressif32 to 3.2.0 / Board: m5stack-core-esp32 / Framework: arduino
- Unable to connect to WiFi Access Point, board in STA or AP-STA modes;
- Unable to access "captive portal" through cell or PC, board in AP mode, using WebServer.h and DNSServer.h;
- User buttons in board seem to be triggering on their own after connection failure.
All the above working fine before upgrade.
As a side note, WiFi Scan is taking considerably longer, I guess it's related to the new DELAY code / looking for stronger APs.

eyecee, 2021-04-29:
Same issue here using WiFiMulti; upgrading to 1.0.6 has broke my WiFi connection code as well with all the same issues listed above.  It feels like the overall connection process in 1.0.6 is taking significantly longer than 1.0.5 which has caused a lot of people's timeouts to trigger. Wifi connections under 1.0.5 was working 99.5% of the time with the 0.5% usually attributable to external factors.

I'll revert to 1.0.5 for now but keen to see this issue solved as there are other upgrades to the ESP32 library that has improved, e.g. SD_MMC library.

zencow, 2021-05-20:
I just wanted to chime in with my own experiences.  I've had code running on a esp32 (Adafruit's Huzzah32) for more than 6 months before breaking my device last night with all the latest versions recently updated in the Arduino interface.  After finding this thread, I did some quick tests rolling back from 1.0.6 to 1.0.5 (from February) and then to 1.0.4 (from October).

- **1.0.6** - connecting with WiFiMulti was taking about 15 seconds.  connecting to Adafruit IO (mqtt service) never reported success.
- **1.0.5** - connecting with WiFiMulti dropped to about 6 seconds.  connection to Adafruit IO never reported success.
- **1.0.4** - connecting with WiFiMulti dropped to about 2.5 seconds (what I expected).  connection to Adafruit IO works again.

I duplicated these results using the stock WiFi example code, using "Simple Time" as a baseline against "WiFiMulti" example code (only adding serial print commands with millis() for showing time before and after connect), and the WiFiMulti connection was slower with 1.0.5 and 1.0.6 quite consistently, with similar times as listed above.  Note, I was using just 2 sets of credentials for WiFiMulti in all of these tests, with only 1 of the 2 ssids within range.

ThomasWaldmann, 2021-07-20:
If nobody fixed this, there is no reason for a stupid stale bot closing this.

enriquewph, 2021-09-02:
Hello, any updates on this? recently updated all the way from 1.0.4 to 2.0.0, and wifi connection time became 10 times slower. I'm not using "WiFiMulti", just the regular "WiFi"

Andr1961, 2021-11-17:
Hi,
the same if the version is older than 1.04 ESP32, then it connects to the router for a very long time or cannot connect at all,
but version 1.02 is faster than 1.04.

perotom, 2022-01-19:
I have the same issue. Connection times got slower.

Andr1961, 2022-01-23:
2.0.2 with all the latest updates on the site on 23.01.2022 nothing has been fixed(((

PilnyTomas, 2022-03-30:
Hello, there has been a recent update of the Arduino-ESP32 core to 2.0.3-RC1 and on my side, it seems that connection time got back to previous values similar to pre-1.0.6
Can you guys please try it out?

Andr1961, 2022-03-31:
Updated through the IDE, it turned out to install only on a clean win10. 
With "debug" it always connects in 15-18 seconds, without "debug" it connects once out of 10
22:55:18.859 -> [    45][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: Ty16
22:55:18.859 -> [    45][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: ssid_from_AP_2
22:55:18.899 -> [    47][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: ssid_from_AP_3
22:55:18.899 -> Connecting Wifi...
22:55:18.899 -> [    84][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 0 - WIFI_READY
22:55:18.979 -> [   172][V][WiFiGeneric.cpp:283] _arduino_event_cb(): STA Started
22:55:19.014 -> [   173][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 2 - STA_START
22:55:22.407 -> [  3599][V][WiFiGeneric.cpp:317] _arduino_event_cb(): SCAN Done: ID: 128, Status: 0, Results: 16
22:55:22.449 -> [  3599][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 1 - SCAN_DONE
22:55:22.449 -> [  3603][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
22:55:22.449 -> [  3608][I][WiFiMulti.cpp:119] run(): [WIFI] 16 networks found
22:55:22.449 -> [  3614][D][WiFiMulti.cpp:149] run():  --->   0: [1][48:57:02:92:A8:66] Ty16 (-40) *
22:55:22.449 -> [  3621][D][WiFiMulti.cpp:151] run():        1: [6][48:57:02:04:38:A1] byfly WIFI (-64)  
22:55:22.449 -> [  3629][D][WiFiMulti.cpp:151] run():        2: [6][48:57:02:04:38:A0] HUAWEI-uCgn (-65) *
22:55:22.489 -> [  3637][D][WiFiMulti.cpp:151] run():        3: [3][60:F1:8A:40:73:F0] Sirius (-71) *
22:55:22.489 -> [  3644][D][WiFiMulti.cpp:151] run():        4: [11][F4:6B:EF:C7:E4:CE] Get-C7E4C8 (-74) *
22:55:22.489 -> [  3652][D][WiFiMulti.cpp:151] run():        5: [4][48:57:02:03:99:71] byfly WIFI (-75)  
22:55:22.489 -> [  3660][D][WiFiMulti.cpp:151] run():        6: [4][48:57:02:03:99:70] HUAWEI-wDqV (-66) *
22:55:22.489 -> [  3668][D][WiFiMulti.cpp:151] run():        7: [7][48:57:02:9C:A9:91] byfly WiFi (-66)  
22:55:22.529 -> [  3676][D][WiFiMulti.cpp:151] run():        8: [7][48:57:02:9C:A9:90] HUAWEI-E9vb (-80) *
22:55:22.529 -> [  3684][D][WiFiMulti.cpp:151] run():        9: [3][7C:C3:85:2B:69:30] Lisa (-81) *
22:55:22.529 -> [  3692][D][WiFiMulti.cpp:151] run():        10: [11][58:90:43:59:B9:56] Get-59B950 (-81) *
22:55:22.529 -> [  3700][D][WiFiMulti.cpp:151] run():        11: [9][48:57:02:03:9C:FD] byfly WIFI (-86)  
22:55:22.529 -> [  3708][D][WiFiMulti.cpp:151] run():        12: [9][48:57:02:03:9C:FC] HUAWEI-AQZG (-87) *
22:55:22.569 -> [  3716][D][WiFiMulti.cpp:151] run():        13: [11][A0:39:EE:B2:44:6E] Get-B24468 (-89) *
22:55:22.569 -> [  3724][D][WiFiMulti.cpp:151] run():        14: [5][48:57:02:04:41:84] HUAWEI-ZPP3 (-90) *
22:55:22.569 -> [  3732][D][WiFiMulti.cpp:151] run():        15: [5][E0:69:95:4B:EF:57] 25453c (-90) *
22:55:22.569 -> [  3739][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: 48:57:02:92:A8:66 SSID: Ty16 Channel: 1 (-40)
22:55:22.569 -> [  3750][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
22:55:22.608 -> [  3791][V][WiFiGeneric.cpp:300] _arduino_event_cb(): STA Disconnected: SSID: Ty16, BSSID: 48:57:02:92:A8:66, Reason: 4
22:55:22.608 -> [  3791][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
22:55:22.649 -> [  3798][W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 4 - ASSOC_EXPIRE
22:55:27.609 -> [  8775][E][WiFiMulti.cpp:187] run(): [WIFI] Connecting Failed (6).
22:55:33.290 -> [ 14481][V][WiFiGeneric.cpp:317] _arduino_event_cb(): SCAN Done: ID: 129, Status: 0, Results: 16
22:55:33.329 -> [ 14482][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 1 - SCAN_DONE
22:55:33.329 -> [ 14486][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
22:55:33.329 -> [ 14491][I][WiFiMulti.cpp:119] run(): [WIFI] 16 networks found
22:55:33.329 -> [ 14496][D][WiFiMulti.cpp:149] run():  --->   0: [1][48:57:02:92:A8:66] Ty16 (-48) *
22:55:33.329 -> [ 14504][D][WiFiMulti.cpp:151] run():        1: [6][48:57:02:04:38:A0] HUAWEI-uCgn (-62) *
22:55:33.329 -> [ 14512][D][WiFiMulti.cpp:151] run():        2: [6][48:57:02:04:38:A1] byfly WIFI (-63)  
22:55:33.369 -> [ 14520][D][WiFiMulti.cpp:151] run():        3: [3][60:F1:8A:40:73:F0] Sirius (-68) *
22:55:33.369 -> [ 14527][D][WiFiMulti.cpp:151] run():        4: [11][F4:6B:EF:C7:E4:CE] Get-C7E4C8 (-75) *
22:55:33.369 -> [ 14535][D][WiFiMulti.cpp:151] run():        5: [11][58:90:43:59:B9:56] Get-59B950 (-77) *
22:55:33.369 -> [ 14543][D][WiFiMulti.cpp:151] run():        6: [4][48:57:02:03:99:70] HUAWEI-wDqV (-66) *
22:55:33.369 -> [ 14551][D][WiFiMulti.cpp:151] run():        7: [4][48:57:02:03:99:71] byfly WIFI (-79)  
22:55:33.409 -> [ 14559][D][WiFiMulti.cpp:151] run():        8: [3][7C:C3:85:2B:69:30] Lisa (-87) *
22:55:33.409 -> [ 14567][D][WiFiMulti.cpp:151] run():        9: [7][48:57:02:9C:A9:91] byfly WiFi (-88)  
22:55:33.409 -> [ 14574][D][WiFiMulti.cpp:151] run():        10: [9][48:57:02:03:9C:FD] byfly WIFI (-88)  
22:55:33.409 -> [ 14582][D][WiFiMulti.cpp:151] run():        11: [11][A0:39:EE:B2:44:6E] Get-B24468 (-88) *
22:55:33.409 -> [ 14591][D][WiFiMulti.cpp:151] run():        12: [5][48:57:02:04:41:84] HUAWEI-ZPP3 (-89) *
22:55:33.449 -> [ 14599][D][WiFiMulti.cpp:151] run():        13: [9][48:57:02:03:9C:FC] HUAWEI-AQZG (-89) *
22:55:33.449 -> [ 14607][D][WiFiMulti.cpp:151] run():        14: [7][48:57:02:9C:A9:90] HUAWEI-E9vb (-90) *
[... cut ...]
22:55:35.969 -> [ 17149][D][WiFiGeneric.cpp:914] _eventCallback(): STA IP: 192.168.100.30, MASK: 255.255.255.0, GW: 192.168.100.1
22:55:36.009 -> [ 17159][I][WiFiMulti.cpp:174] run(): [WIFI] Connecting done.
22:55:36.009 -> [ 17165][D][WiFiMulti.cpp:175] run(): [WIFI] SSID: Ty16
22:55:36.009 -> [ 17170][D][WiFiMulti.cpp:176] run(): [WIFI] IP: 192.168.100.30
22:55:36.009 -> [ 17175][D][WiFiMulti.cpp:177] run(): [WIFI] MAC: 48:57:02:92:A8:66
22:55:36.009 -> [ 17181][D][WiFiMulti.cpp:166] run(): [WIFI] Channel: 1

PilnyTomas, 2022-04-01:
Can you please edit your code to be complete, and without the TFT? I would like to check your code exactly how are you using it.

Andr1961, 2022-04-01:
The last log from example WiFiMulti.ino,
on 1.04 everything works well, problems start from 1.05

PilnyTomas, 2022-04-04:
Yes, there were problems but with the recent 2.0.3-RC1 they should be fixed. I just tried the example sketch `WiFiMulti.ino` on the newest version and it works - no errors, connected in 2.5 sec.
I am unable to replicate your error.
Arduino IDE: 1.8.19
Arduino-ESP32 core: 2.0.3-RC1
HW: DevKitC ESP32-WROVER-B
OS: Linux Fedora 34

Andr1961, 2022-04-05:
I will wait, maybe you can find a reason,
noticed, if the sketch with "VERBOSE," then always connects on the second attempt 17-18 sec,
without "VERBOSE" almost never connects.

PilnyTomas, 2022-04-06:
With or without the debug logs it connects fast for me on the core version 2.0.3-RC1.
I am sorry, but I cannot replicate your issue.
@handmade0octopus, @ThomasWaldmann, @zasnicoff, @eyecee, @zencow, @enriquewph, @perotom Can you please try the new core if the updates fixed your issue?

perotom, 2022-04-06:
On my end, connecting worked good and fast but I tried it very little. In one week I will have 30 devices in use which constantly connect. I will get back with some impressions.

zencow, 2022-04-06:
I still get "slow" connects and sometimes "exceedingly slow" connects with 2.0.3-RC1.  When it works, connection times are between those I got for version 1.0.4 (~2.3-2.5 seconds) and version 1.0.5 (~6.5 seconds).  However, I also often get errors, which extend connection times over 30-60 seconds.

* "slow" is typically taking ~4 to 5 seconds, but I've seen it as fast as 3.6 seconds.

* "exceedingly slow" is over 30 seconds, sometimes 45+ or 60+ seconds.  This happens when there is an error during the initial connection, which starts a long connection cycle with multiple failures before it finally succeeds.

I often see *errors* when I power up the board from scratch or I hit the on-board Reset (which includes the first run right after flashing the device with new code).  It tends to not have errors if I do a WiFi.disconnect() followed by esp_restart() (but this isn't guaranteed).  This seems to indicate that when it simply disappears from the network that it leaves the AP in a bad state, but... I never see these errors, nor long connect times with the same code compiled against version 1.0.4.

The error output starts with ```wifi:Association refused temporarily, comeback time 0 mSec``` (which it spews out for a couple seconds) and then goes through a series of connect steps, which fail multiple times until it finally succeeds, taking upwards of 30-45, sometimes 65+ seconds.  This connection example with Verbose output *only* took about 32 seconds:

My test code is based on the sample code for WiFiMulti.  I've been gradually modifying it to refine the output.  I have added a delay after starting Serial (for getting the serial monitor open after flashing the device), added a printf statement with ARDUINO_ESP32_GIT_DESC and use esp_timer_get_time() for timing the connection from start to end.  My main loop calls wifiMulti.run() and also checks for serial input which triggers a software reset (so I can test WiFi.disconnect() with esp_restart() results vs using the hardware Reset button).

```
12:37:27.836 -> ets Jul 29 2019 12:21:46
12:37:27.836 -> 
12:37:27.836 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
12:37:27.836 -> configsip: 0, SPIWP:0xee
12:37:27.836 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
12:37:27.836 -> mode:DIO, clock div:1
12:37:27.836 -> load:0x3fff0030,len:1184
12:37:27.836 -> load:0x40078000,len:12812
12:37:27.836 -> load:0x40080400,len:3032
12:37:27.836 -> entry 0x400805e4
12:37:28.105 -> [(...garbled serial connection mess snipped to allow paste of content...)2-hal-cpu.c:214] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
12:37:28.137 -> [    54][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 0 - WIFI_READY
12:37:28.239 -> [   153][V][WiFiGeneric.cpp:283] _arduino_event_cb(): STA Started
12:37:28.239 -> [   153][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 2 - STA_START
12:37:33.244 -> 
12:37:33.244 -> ARDUINO_ESP32_GIT_DESC = 2.0.3-RC1
12:37:33.244 -> [5153911] Connecting Wifi...
12:37:33.244 -> [  5154][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: trapdoor
12:37:36.093 -> [  7983][V][WiFiGeneric.cpp:317] _arduino_event_cb(): SCAN Done: ID: 128, Status: 0, Results: 22
12:37:36.093 -> [  7983][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 1 - SCAN_DONE
12:37:36.093 -> [  7988][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
12:37:36.093 -> [  7992][I][WiFiMulti.cpp:119] run(): [WIFI] 22 networks found

...snip scan results, first entry index 0 has "--->" pointing at it...

12:37:36.294 -> [  8172][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: 78:8A:20:4C:49:EC SSID: trapdoor Channel: 6 (-63)
12:37:36.294 -> [  8183][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
12:37:36.294 -> E (16610) wifi:Association refused temporarily, comeback time 0 mSec
12:37:36.328 -> E (16619) wifi:Association refused temporarily, comeback time 0 mSec

...snip ~140 repeats...

12:37:37.333 -> E (17625) wifi:Association refused temporarily, comeback time 0 mSec
12:37:37.333 -> E (17631) wifi:Association refused temporarily, comeback time 0 mSec
12:37:38.370 -> [ 10243][V][WiFiGeneric.cpp:300] _arduino_event_cb(): STA Disconnected: SSID: trapdoor, BSSID: 78:8a:20:4c:49:ec, Reason: 203
12:37:38.370 -> [ 10244][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
12:37:38.370 -> [ 10251][W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 203 - ASSOC_FAIL
12:37:38.370 -> [ 10258][D][WiFiGeneric.cpp:889] _eventCallback(): WiFi Reconnect Running
12:37:38.370 -> [ 10266][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
12:37:40.849 -> [ 12709][V][WiFiGeneric.cpp:300] _arduino_event_cb(): STA Disconnected: SSID: trapdoor, BSSID: 00:00:00:00:00:00, Reason: 201
12:37:40.849 -> [ 12710][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
12:37:40.849 -> [ 12717][W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 201 - NO_AP_FOUND
12:37:40.849 -> [ 12724][D][WiFiGeneric.cpp:898] _eventCallback(): WiFi AutoReconnect Running
12:37:40.849 -> [ 12731][E][WiFiMulti.cpp:181] run(): [WIFI] Connecting Failed AP not found.
12:37:40.849 -> [ 12732][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[... cut ...]
12:38:05.492 -> [ 37374][D][WiFiGeneric.cpp:914] _eventCallback(): STA IP: 10.101.0.207, MASK: 255.255.255.0, GW: 10.101.0.1
12:38:05.492 -> [ 37383][I][WiFiMulti.cpp:174] run(): [WIFI] Connecting done.
12:38:05.492 -> [ 37389][D][WiFiMulti.cpp:175] run(): [WIFI] SSID: trapdoor
12:38:05.492 -> [ 37394][D][WiFiMulti.cpp:176] run(): [WIFI] IP: 10.101.0.207
12:38:05.492 -> [ 37399][D][WiFiMulti.cpp:177] run(): [WIFI] MAC: 78:8A:20:4C:49:EC
12:38:05.525 -> [ 37405][D][WiFiMulti.cpp:178] run(): [WIFI] Channel: 6
12:38:05.525 -> 
12:38:05.525 -> [5155018]-[37410863] CONNECTED in 32.26 seconds
12:38:05.525 -> 
```

PilnyTomas, 2022-04-12:
Hi @zencow, is this behavior consistent across various routers/APs and various distances from them?

zencow, 2022-04-14:
@PilnyTomas My tests have mainly been against the one AP that I have (Ubiquiti UniFi), using several different dev boards... all with the WROOM-32E chip (Adafruit's Huzzah32 Feather and their breakout board version).  I think I had originally tested against my phone's hotspot (the reason I'm using WifiMulti), with similar results, but I haven't done that recently.  Maybe I'll do a quick test now to see if that still yields similar results.

zencow, 2022-04-14:
@PilnyTomas  I just did some tests against my Mobile HotSpot and I do see similar connect time patterns as with my primary AP for versions tested (1.0.4, 1.0.5, 1.0.6, 2.0.3-RC1), but it's also a lot less reliable with the HotSpot regardless of version.

Here's how I'm doing the timing measurement:
```
#include <WiFi.h>
#include <WiFiMulti.h>
#include <core_version.h>

WiFiMulti wifiMulti;
#define XSTR(x) #x
#define STR(x) XSTR(x)

void setup() {
    Serial.begin(115200);
    delay(1000);

    WiFi.mode(WIFI_STA);
    WiFi.disconnect();
    delay(5000);

    Serial.printf("\n");
    Serial.printf("ARDUINO_ESP32_GIT_DESC = %s\n", STR(ARDUINO_ESP32_GIT_DESC));
    Serial.printf("[%lld] Connecting Wifi...\n", esp_timer_get_time());

    wifiMulti.addAP("ssid", "passphrase");

    // ## START CONNECT TIME MEASUREMENT ############################
    int64_t connect_start = esp_timer_get_time();
    // ##############################################################

    WiFi.mode(WIFI_STA);
    
    while(wifiMulti.run(WIFI_CONNECTION_TIMEOUT) != WL_CONNECTED) {
      delay(10);
    }

    // ## END CONNECT TIME MEASUREMENT ##############################
    int64_t connect_end = esp_timer_get_time();
    Serial.printf("\n[%lld]-[%lld] CONNECTED in %4.2f seconds\n", connect_start, connect_end, (connect_end-connect_start)/1e6 );
    // ##############################################################

    Serial.printf("\n");
    Serial.printf("[%lld] WiFi Connected to [%s]  ", esp_timer_get_time(), WiFi.SSID().c_str());
    Serial.print("IP [");
    Serial.print(WiFi.localIP());
    Serial.println("]");
}

void loop()
{
    if(wifiMulti.run(WIFI_CONNECTION_TIMEOUT) != WL_CONNECTED) {
        Serial.printf("[%lld] WiFi not connected\n", esp_timer_get_time());
    } else {
        Serial.printf("[%lld] WiFi Connected to [%s]  ", esp_timer_get_time(), WiFi.SSID().c_str());
        Serial.print("IP [");
        Serial.print(WiFi.localIP());
        Serial.println("]");
    }
    delay(2000);
    if (Serial.available()) {
      WiFi.disconnect();
      Serial.printf("Restarting...\n");
      delay(1000);
      esp_restart();
    }
}
```

On a hardware reset of the device, the connecting time measurement (for both my UniFi AP and the mobile HotSpot) are calculated to be about:
* 1.0.4 - ~2.3-2.6 seconds
* 1.0.5 - ~5.5-6.5 seconds
* 1.0.6 - ~10.5-11.5 seconds
* 2.0.3-RC1 - ~3.6 seconds

**The rest of this observation doesn't directly relate to this issue, but here's what I noticed with the HotSpot reliability:**

The Mobile HotSpot seems to drop the devices often (maybe because they're not really passing any traffic?).

The devices running 1.0.4 don't seem to notice that they have been dropped by the HotSpot.  I suspect this is because 1.0.4 doesn't have the same callback system and will (re)determine its connection status when the device actually needs to use the connection.

The devices running 2.0.3-RC1 detect the connection drop every ~10-30 seconds and reconnect, which seems to take ~7 seconds from the first "STA_DISCONNECTED" message to the "STA_CONNECTED" and "STA_GOT_IP" messages, mostly (~6 seconds) between the "STA_DISCONNECTED" message and "SCAN_DONE" (I trimmed out the scan results and such, keeping the event messages and codes):

```
[823396][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[823403][W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 7 - NOT_ASSOCED
[830196][V][WiFiGeneric.cpp:317] _arduino_event_cb(): SCAN Done: ID: 148, Status: 0, Results: 19
[830197][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 1 - SCAN_DONE
```
...
```
[830501][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[830556][V][WiFiGeneric.cpp:305] _arduino_event_cb(): STA Got Same IP:192.168.43.4
[830556][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
```

zencow, 2022-04-14:
Hmmm... I just noticed that my 1.0.4 device was still connected solidly all the time I was writing up the last message, and it was listed in the HotSpot's "Connected devices" and was not dropping it, like the 2.0.3-RC1 device.

To see if this was the version or the devices themselves, I re-flashed the 2 test devices I'm using today, swapping the version running on them.

Again... whichever device is running 1.0.4 stays solidly connected to the HotSpot, while the 2.0.3-RC1 device frequently disconnects and re-connects, regardless of the physical device.  Maybe this is due to lack of traffic and attempts to save power?  I dunno, but those delays taking 5+ seconds to get reconnected are not ideal.

In case there were conflicts, I powered off the 1.0.4 device and still see the frequent connection drops in the 2.0.3-RC1 device (with this test code, that isn't actually using the connection once established).

zencow, 2022-04-15:
The connection times I've listed previously are for ```wifiMulti```, even when setting just a single set of credentials.  When using ```WiFi.begin(ssid, password);```, connect times are much faster, but 1.0.4 is still the clear winner.

Connect times with ```WiFi.begin```:
* 1.0.4 - ~0.3 seconds (normalized to 1x)
* 1.0.5 - ~0.7 seconds (~2.3x)
* 1.0.6 - ~1.5 seconds (~5x)
* 2.0.3-RC1 - ~2.5 seconds (~8.3x)

Andr1961, 2022-04-16:
The same problem on esp8266, on the version after
2.5.2

gonzabrusco, 2022-04-30:
Can you test with latest master? @me-no-dev recently pushed some changes about this.

zencow, 2022-05-04:
It didn't compile for me.

To install the latest version, I followed instructions at:
https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#windows-manual-installation

...but instead of using GitGUI, I used ```git``` from cygwin.  I don't use git often, but I think I picked the right commands out of the linux instructions and got the files into the right folders with these commands.

```
cd c/Users/chris/Documents/Arduino
mkdir -p hardware/espressif
cd hardware/espressif
git clone https://github.com/espressif/arduino-esp32.git esp32
cd esp32/
git submodule update --init --recursive
cd tools
chmod u+x get.exe
```

I then ran get.exe by double-clicking it in the Explorer and after it was done, I started the Arduino IDE, chose my board from the list in ```ESP32 Arduino (in sketchbook)``` and attempted to compile.

Let me know if you need debug output and I'll copy the gobs of that info, but this is the standard. 

```
C:\Users\chris\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi\src\WiFiGeneric.cpp: In static member function 'static bool WiFiGenericClass::setDualAntennaConfig(uint8_t, uint8_t, wifi_rx_ant_t, wifi_tx_ant_t)':
C:\Users\chris\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi\src\WiFiGeneric.cpp:1265:5: sorry, unimplemented: non-trivial designated initializers not supported
     };
     ^
C:\Users\chris\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi\src\WiFiGeneric.cpp:1265:5: sorry, unimplemented: non-trivial designated initializers not supported
C:\Users\chris\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi\src\WiFiGeneric.cpp:1265:5: sorry, unimplemented: non-trivial designated initializers not supported
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\chris\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
Error compiling for board Adafruit ESP32 Feather.
```

zencow, 2022-05-04:
Above, I was using Arduino version ```1.8.13``` and just updated to ```1.8.19``` to see if that would make a difference, but I got the same error.  Switching over to esp32 release version ```1.0.4``` compiled ok, as I expected.

VojtechBartoska, 2022-05-04:
Hello folks, please retest this with 2.0.3.-stable. This is supposed to be solved.

Thanks!

PilnyTomas, 2022-05-05:
@zencow if you have problems with installation, please open a new issue. This one is about WiFi connection time.
You can ping us about it here, but let's solve it in a different thread.

Andr1961, 2022-05-05:
I have been connected for longer than in 1.0.4, but it always connects, this is already good, I often didn't connect at all before. It seems that he goes through all the ssids in turn until he gets to my settings. There were problems with reading files from SD, but this is another topic

zencow, 2022-05-05:
@PilnyTomas Thanks... I'm not experienced enough to know if it was an installation issue or an issue with the "cutting edge update" from the repository that I was asked to try out.  I just reported my negative results with that test, and tried to include enough info on what I did, so that experienced folks could use to evaluate those results.  I typically try to keep to release versions, since they're packaged and don't require many error-prone manual steps to install.

The error message ``sorry, unimplemented: non-trivial designated initializers not supported`` looks (to my untrained eye) like an inconsistency caused by code that is in transition, but isn't completely integrated.  If the issue was actually the installation process, then I'll just steer clear of attempting to use anything other than a tagged release/release-candidate that the Arduino-IDE or PlatformIO can manage directly.

lbernstone, 2022-05-05:
Yes, dual antenna is new code, and may have a mismatch on the struct elements.  @VojtechBartoska was referring to this when he said he thought this was fixed in 2.0.3rc1

zencow, 2022-05-05:
Ok...  does ```VojtechBartoska removed this from the [2.0.3] milestone [14 days ago]``` mean that increased connection times were not addressed in the 2.0.3 release?

I just updated to ```2.0.3``` in the Arduino IDE, changed my code that determines version since core_version.h was removed and esp_arduino_version.h was added, and tested again.  I see similar results to timing with ```wifiMulti``` that I previously reported for ```2.0.3-RC1``` (5-7 seconds, compared to 2.3 seconds in ```1.0.4```), and lots of ```E (23079) wifi:Association refused temporarily, comeback time 0 mSec``` on hard resets (resulting in >25-30 seconds to connect... tho sometimes stalling indefinitely).  I don't get that error when doing a ("graceful") software reset using ```WiFi.disconnect()``` followed by ```esp_restart()```.

VojtechBartoska, 2022-05-06:
@zencow It was removed from 2.0.3 milestone because we weren't able to test this properly but it's still supposed to be fixed.

Due to that this issue was moved to 2.0.4 and @PilnyTomas is going to test this in next 2 weeks. Up to that we will set up next steps.

handmade0octopus, 2022-05-08:
@PilnyTomas Since 1.0.5 I tested on 2.0.3 and I believe it's back on the track now. Seems to be fixed!