espressif/arduino-esp32 #4984: SD/FS card issues with batch operations

brunojoyal, 2021-03-29:
### Hardware:
Board:							ESP32 DEVKIT V1
Core Installation version:		1.0.6
IDE name:						PlatformIO
Peripherals:                    SD Card (SPI)
Partition Scheme:               FAT32

### Description:
There seem to be a number of serious issues with batch operations on an SD card.

Here is a simple script which attempts to write 513 gibberish files when you press "a", and cleans up all gibberish files when you press "b".

Some observations:

1. Pressing 'a', we see that the write operation fails on the 513th file, with the Serial output

```
...
508 : /691358.TXT 191 ms
509 : /735921.TXT 191 ms
510 : /866109.TXT 237 ms
511 : /343629.TXT 191 ms
[E][vfs_api.cpp:270] VFSFileImpl(): fopen(/sd/420985.TXT) failed
```

2. Even resetting the board and pressing "a" again yields the same error immediately - it seems impossible to write a 513th file to the root directory until the files have been erased.

3. Pressing 'b' to clear the root directory, we can see that the `SD.remove()` calls get slower and slower in a non-sustainable manner:

> Removed /373736.TXT 9 ms
> Removed /778743.TXT 9 ms
> Removed /836301.TXT 9 ms
> Removed /472533.TXT 9 ms
> ...
> Removed /691358.TXT 140 ms
> Removed /735921.TXT 140 ms
> Removed /866109.TXT 141 ms
> Removed /343629.TXT 141 ms

Perhaps a separate issue, but illustrated by the same code, and I suspect having to do with some of the same memory management issues lurking in the FS library.

Thanks for any help. 

### Sketch: 
```cpp

void loadGibberish(void *parameter);
void removeSomeGibberish(void *parameter);
#include <Arduino.h>
#include <SD.h>

void setup()
{
  Serial.begin(115200);

  while (!SD.begin())
  {
    Serial.println("Loading SD card...");
    delay(200);
  }
  Serial.println("Card loaded.");
}

void loop()
{
  if (Serial.available())
  {
    switch (Serial.read())
    {
    case 'a':
      xTaskCreate(loadGibberish, "Load Gibberish", 10000, NULL, 1, NULL);
      break;
    case 'b':
      xTaskCreate(removeSomeGibberish, "Remove Some Gibberish", 10000, NULL, 1, NULL);
      break;
    }
  }
}

#define LEN 1000

int counter = 0;
char buf[100];
void loadGibberish(void *parameter) //create 513 2kb gibberish files.
{ 
  while (true)
  {
    for (int i = 0; i < 513; i++)
    {

      long timer = millis();
      File newImportantFile;
      char filename[32];

      sprintf(filename, "/%i.TXT", esp_random() % 999999); //random filename

      newImportantFile = SD.open(filename, "w");

      if (newImportantFile)
      {
        for (int j = 0; j <= 2; j++)
        {
          uint8_t gibberish[LEN];
          newImportantFile.write(gibberish, LEN);
        }
        newImportantFile.close();
        long duration = millis() - timer;
        sprintf(buf, "%i : %s %li ms", i, filename, duration);
        Serial.println(buf);
      }
      else
        vTaskDelete(NULL);
      vTaskDelay(1);
    }
    vTaskDelete(NULL);
  }
}

void removeSomeGibberish(void *parameter)
{

  while (true)
  {

    File root = SD.open("/");
    File file = root.openNextFile();

    while (file)
    {
      long timer = millis();
      char filename[32];
      memcpy(filename, file.name(), strlen(file.name()) + 1);
      file.close();
      if (!SD.remove(filename))
      {
        Serial.println("Problem. Aborting.");
      }
      long duration = millis() - timer;
      sprintf(buf, "Removed %s %li ms", filename, duration);
      Serial.println(buf);

      file = root.openNextFile();
      vTaskDelay(1);
    }

    vTaskDelete(NULL);
  }
}

brunojoyal, 2021-03-30:
The issue with 512 files is probably a hard-coded limit which applies only to root. i.e. not really an issue.
I will open a new issue which better illustrates my problem. Thank you.

closed 2021-03-30 by brunojoyal

chegewara, 2021-03-30:
It is actually FAT fs standard.