espressif/arduino-esp32 #4987: The SDK includes an old version of Unity which doesn't include TEST_MESSAGE

nietaki, 2021-03-29:
### Hardware:
|||
|:---|---|
|<B>Board</B>|ESP32 Dev Module|
|<B>Version/Date</B>|3.1.1|
|<B>IDE name</B>|Neovim + platformio|
|<B>Flash Frequency</B>|40Mhz|
|<B>PSRAM enabled</B>|?|
|<B>Upload Speed</B>|460800|
|<B>Computer OS</B>|Ubuntu 20.04|

### Description:

I have the same some common tests running in `native` and `esp32dev` environments. I realized using `TEST_MESSAGE` works well in `native`, but doesn't work when the tests run on the devboard.

It seems like the [bundled version of unity](https://github.com/espressif/arduino-esp32/tree/master/tools/sdk/include/unity) is relatively old, or maybe just with a different interface than the main repo: https://github.com/ThrowTheSwitch/Unity/tree/master/src

Is there a way to get the unity interfaces more aligned? Is it worth it to push for `arduino-esp32` to pull in the latest version of unity / align the interface with the one that runs in `native`?

Thanks.

### Test file - minimum reproduction:  
```cpp
#include <unity.h>
#include <StandardHeaders.h> // relevant to my project, conditionally includes Arduino.h

// void setUp(void) {
// // set stuff up here
// }

// void tearDown(void) {
// // clean stuff up here
// }

void test_message(void) {
  TEST_MESSAGE("foo");
}

void process() {
    UNITY_BEGIN();
    RUN_TEST(test_message);
    UNITY_END();
}

#ifdef ARDUINO
void setup() {
  process();
}

void loop() {
}
#else
int main(int argc, char **argv) {
    process();
    return 0;
}
#endif

```

### Test output in `native` environment:

```
# platformio run -e native --target clean
platformio test -e native
Verbose mode can be enabled via `-v, --verbose` option
Collected 5 items

Processing test_generic_utils in native environment
-------------------------------------------------------------------------------------------------------------------------------------------------
Building...
Testing...
(some test output cut)
test/test_generic_utils/test_generic_utils.cpp:21:test_message:INFO: foo
test/test_generic_utils/test_generic_utils.cpp:131:test_message [PASSED]

```

### Test output in `esp32dev` environment

```
Processing test_generic_utils in esp32dev environment
-------------------------------------------------------------------------------------------------------------------------------------------------
Building...
test/test_generic_utils/test_generic_utils.cpp: In function 'void test_message()':
test/test_generic_utils/test_generic_utils.cpp:21:21: error: 'TEST_MESSAGE' was not declared in this scope
   TEST_MESSAGE("foo");
                     ^
*** [.pio/build/esp32dev/test/test_generic_utils/test_generic_utils.cpp.o] Error 1
```

lbernstone, 2021-03-29:
The unity library comes from upstream in the [esp-idf](https://github.com/espressif/esp-idf/tree/release/v3.3/components/unity), so this repo does not control the version.  You can include whatever version you would like.  Just put the code in your sketch directory, and `#include "unity.h"`.  quotes get you the local version, brackets will get you the framework version.

nietaki, 2021-03-30:
Thanks @lbernstone, what you suggested seemed like a very reasonable idea and I gave it a try.

I put the unity files under `vendor/unity/` and added `lib_extra_dirs = vendor` to `platformio.ini`.

It seems like the compilation and linking tries to use both the `unity` included by the framework and the one added by me and they are conflicting:

```
$ platformio test -e esp32dev -f test_geometry -v                                                                                          1 ↵
Collected 5 items

Processing test_geometry in esp32dev environment
-------------------------------------------------------------------------------------------------------------------------------------------------
Building...
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.1.1) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10005.210308 (1.0.5) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <BasicLinearAlgebra> 2.3.0 (/home/nietaki/repos/moving_pieces/archimedes/.pio/libdeps/esp32dev/BasicLinearAlgebra)
|-- <TMCStepper> 0.7.1 (/home/nietaki/repos/moving_pieces/archimedes/.pio/libdeps/esp32dev/TMCStepper)
|   |-- <SPI> 1.0 (/home/nietaki/.platformio/packages/framework-arduinoespressif32/libraries/SPI)
|-- <SD(esp32)> 1.0.5 (/home/nietaki/.platformio/packages/framework-arduinoespressif32/libraries/SD)
|   |-- <FS> 1.0 (/home/nietaki/.platformio/packages/framework-arduinoespressif32/libraries/FS)
|   |-- <SPI> 1.0 (/home/nietaki/.platformio/packages/framework-arduinoespressif32/libraries/SPI)
|-- <logic> (/home/nietaki/repos/moving_pieces/archimedes/lib/logic)
|   |-- <BasicLinearAlgebra> 2.3.0 (/home/nietaki/repos/moving_pieces/archimedes/.pio/libdeps/esp32dev/BasicLinearAlgebra)
|-- <embedded_utils> (/home/nietaki/repos/moving_pieces/archimedes/lib/embedded_utils)
|   |-- <TMCStepper> 0.7.1 (/home/nietaki/repos/moving_pieces/archimedes/.pio/libdeps/esp32dev/TMCStepper)
|   |   |-- <SPI> 1.0 (/home/nietaki/.platformio/packages/framework-arduinoespressif32/libraries/SPI)
|   |-- <FS> 1.0 (/home/nietaki/.platformio/packages/framework-arduinoespressif32/libraries/FS)
|   |-- <logic> (/home/nietaki/repos/moving_pieces/archimedes/lib/logic)
|   |   |-- <BasicLinearAlgebra> 2.3.0 (/home/nietaki/repos/moving_pieces/archimedes/.pio/libdeps/esp32dev/BasicLinearAlgebra)
|-- <behaviours> (/home/nietaki/repos/moving_pieces/archimedes/lib/behaviours)
|   |-- <logic> (/home/nietaki/repos/moving_pieces/archimedes/lib/logic)
|   |   |-- <BasicLinearAlgebra> 2.3.0 (/home/nietaki/repos/moving_pieces/archimedes/.pio/libdeps/esp32dev/BasicLinearAlgebra)
|   |-- <embedded_utils> (/home/nietaki/repos/moving_pieces/archimedes/lib/embedded_utils)
|   |   |-- <TMCStepper> 0.7.1 (/home/nietaki/repos/moving_pieces/archimedes/.pio/libdeps/esp32dev/TMCStepper)
|   |   |   |-- <SPI> 1.0 (/home/nietaki/.platformio/packages/framework-arduinoespressif32/libraries/SPI)
|   |   |-- <FS> 1.0 (/home/nietaki/.platformio/packages/framework-arduinoespressif32/libraries/FS)
|   |   |-- <logic> (/home/nietaki/repos/moving_pieces/archimedes/lib/logic)
|   |   |   |-- <BasicLinearAlgebra> 2.3.0 (/home/nietaki/repos/moving_pieces/archimedes/.pio/libdeps/esp32dev/BasicLinearAlgebra)
|   |-- <SD(esp32)> 1.0.5 (/home/nietaki/.platformio/packages/framework-arduinoespressif32/libraries/SD)
|   |   |-- <FS> 1.0 (/home/nietaki/.platformio/packages/framework-arduinoespressif32/libraries/FS)
|   |   |-- <SPI> 1.0 (/home/nietaki/.platformio/packages/framework-arduinoespressif32/libraries/SPI)
|-- <unity> (/home/nietaki/repos/moving_pieces/archimedes/vendor/unity)
|-- <calculator> (/home/nietaki/repos/moving_pieces/archimedes/lib/calculator)
Building in release mode

(many lines cut)
xtensa-esp32-elf-g++ -o .pio/build/esp32dev/test/test_geometry/test_geometry.cpp.o -c -fno-rtti -fno-exceptions -std=gnu++11 -Os -g3 -Wall -nostdlib -Wpointer-arith -Wno-error=unused-but-set-variable -Wno-error=unused-variable -mlongcalls -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wno-error=deprecated-declarations -Wno-error=unused-function -Wno-unused-parameter -Wno-sign-compare -fstack-protector -fexceptions -Werror=reorder -DPLATFORMIO=50101 -DARDUINO_ESP32_DEV -DESP32 -DESP_PLATFORM -DF_CPU=240000000L -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DARDUINO=10805 -DARDUINO_ARCH_ESP32 -DARDUINO_VARIANT=\"esp32\" "-DARDUINO_BOARD=\"Espressif ESP32 Dev Module\"" -DUNIT_TEST -DUNITY_INCLUDE_CONFIG_H -Iinclude -Isrc -Ilib/calculator -Ivendor/unity -Ilib/behaviours -Ilib/embedded_utils -Ilib/logic -I/home/nietaki/.platformio/packages/framework-arduinoespressif32/libraries/SD/src -I/home/nietaki/.platformio/packages/framework-arduinoespressif32/libraries/FS/src -I.pio/libdeps/esp32dev/TMCStepper/src -I/home/nietaki/.platformio/packages/framework-arduinoespressif32/libraries/SPI/src -I.pio/libdeps/esp32dev/BasicLinearAlgebra -I/home/nietaki/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/config -I/home/nietaki/.platformi [... cut ...]
.pio/build/esp32dev/libUnityTestLib.a(unity.c.o):(.rodata.UnityStrErrShorthand+0x0): multiple definition of `UnityStrErrShorthand'
.pio/build/esp32dev/lib09c/libunity.a(unity.c.o):(.rodata.UnityStrErrShorthand+0x0): first defined here
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32dev/firmware.elf] Error 1
```

I realise this might be more of a `platformio` issue than `arduino-esp32` issue, but currently I don't have an idea of how to resolve it without updating upstream `esp-idf` or sticking to common `unity` APIs.

nietaki, 2021-04-07:
I was confident it wouldn't work because I remembered [platformio docs](https://docs.platformio.org/en/latest/plus/unit-testing.html#shared-code) mentioning that by default code under `src/` doesn't get built when running tests, but it did work when I put those files directly in `src/` (not in `src/unity/`).

The problem is now when I'm building the main project I'm getting the conflict:

```
Building in release mode
xtensa-esp32-elf-g++ -o .pio/build/esp32dev/firmware.elf -T esp32_out.ld -nostdlib -Wl,-static -u call_user_start_cpu0 -Wl,--undefined=uxTopUsedPriority -Wl,--gc-sections -Wl,-EL -T esp32.project.ld -T esp32.rom.ld -T esp32.peripherals.ld -T esp32.rom.libgcc.ld -T esp32.rom.spiram_incompatible_fns.ld -u ld_include_panic_highint_hdl -u __cxa_guard_dummy -u __cxx_fatal_exception .pio/build/esp32dev/src/main.cpp.o .pio/build/esp32dev/src/unity.c.o -L.pio/build/esp32dev -L/home/nietaki/.platformio/packages/framework-arduinoespressif32/tools/sdk/lib -L/home/nietaki/.platformio/packages/framework-arduinoespressif32/tools/sdk/ld -Wl,--start-group .pio/build/esp32dev/libd1d/libBasicLinearAlgebra.a .pio/build/esp32dev/lib440/libSPI.a .pio/build/esp32dev/libbdc/libTMCStepper.a .pio/build/esp32dev/lib4fa/libFS.a .pio/build/esp32dev/lib12f/libSD.a .pio/build/esp32dev/lib919/liblogic.a .pio/build/esp32dev/lib582/libembedded_utils.a .pio/build/esp32dev/libbc0/libbehaviours.a .pio/build/esp32dev/libFrameworkArduinoVariant.a .pio/build/esp32dev/libFrameworkArduino.a -lgcc -lapp_trace -llibsodium -lbt -lesp-tls -lmdns -lconsole -ljsmn -lesp_ringbuf -lpthread -ldriver -ldetection -lsoc -lc -lmesh -lwpa2 -ljson -ldl -lwear_levelling -lmicro-ecc -lcoexist -lface_detection -lnvs_flash -lwifi_provisioning -lfr -lnghttp -lesp32 -lnet80211 -lesp_http_server -ltcp_transport -llog -lespnow -lhal -lmqtt -lesp_websocket_client -lesp_http_client -lvfs -lbtdm_app -lapp_update -lpe -lprotocomm -lwps -lsdmmc -lesp_adc_cal -lwpa -lefuse -lcoap -lsmartconfig -limage_util -lspiffs -lulp -lunity -lface_recognition -lesp_https_server -lethernet -lspi_flash -lpp -lexpat -lfatfs -ltcpip_adapter -llwip -lcxx -lfreertos -lesp32-camera -lmbedtls -ldetection_cat_face -lm -lc_nano -lesp_event -lnewlib -lcore -lopenssl -lsmartconfig_ack -lwpa_supplicant -lbootloader_support -lasio -lesp_https_ota -lod -lespcoredump -lheap -lrtc -lprotobuf-c -lfb_gfx -lfreemodbus -lfd -lphy -lxtensa-debug-module -lstdc++ -Wl,--end-group
/home/nietaki/.platformio/packages/framework-arduinoespressif32/tools/sdk/lib/libunity.a(unity.o): In function `UnityPrintNumberUnsigned':
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/unity/unity/src/unity.c:192: multiple definition of `UnityPrintNumberUnsigned'
.pio/build/esp32dev/src/unity.c.o:/home/nietaki/repos/moving_pieces/archimedes/src/unity.c:257: first defined here
/home/nietaki/.platformio/packages/framework-arduinoespressif32/tools/sdk/lib/libunity.a(unity.o): In function `UnityPrintNumber':
unity.c:(.text.UnityPrintNumber+0x0): multiple definition of `UnityPrintNumber'
```

I tried getting rid of the `-lunity` flag by putting it in `build_unflags` in `platformio.ini`, but it didn't seem to work.

If I could get it to work, the solution seems hacky to me and I can't really understand why it works, I think I'd go with using the bundled version of unity in the end if I couldn't figure anything else out.

closed 2021-07-11 by stale[bot]

Bootstrap360, 2021-12-31:
Did anyone find a solution to this? I am having the same issue. :(

ankostis, 2022-04-30:
I'm having the same issue with latest *dev* `platformio-core-v6.0.0a2`, and had to revert back to latest *stable* to 5.2.5.
I guess i should open a bug-report in platformio/platformio-core.