espressif/arduino-esp32 #4990: Compilation breaks when using -DUSE_ESP_IDF_LOG

salmon-charles, 2021-03-30:
Hello,

First, I really appreciate that someone is working on this topic: I need to redirect the logs to a telnet console and without your change it is difficult without having to refactor my code. 

I am using PlatformIO with Expressif 32 Platform@3.1.1.
If I look the esp32-hal-log.h file (in /.platformio/packages/framework-arduinoespressif32/cores/esp32), I can see that the USE_ESP_IDF_LOG is handled.
When I compiled my project, I have the following errors:

```
In file included from /Users/ces/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal.h:45:0,
                 from /Users/ces/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:35,
                 from src/main.cpp:1:
/Users/ces/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-log.h:167:13: error: expected unqualified-id before string constant
 #define TAG "ARDUINO"
             ^
lib/status/StatusManager.h:24:22: note: in expansion of macro 'TAG'
         const char * TAG = "StatusManager";
                      ^
In file included from lib/module/Module.h:3:0,
                 from src/main.cpp:2:
lib/status/StatusManager.h:24:20: error: expected ';' at end of member declaration
         const char * TAG = "StatusManager";
```

StatusManager.h

```cpp

#pragma once

#include <Arduino.h>

#define STATUS_LED_PIN 12

class StatusManager {
    public:
        StatusManager();
        ~StatusManager();

        // States
        enum State {
            awake,
            asleep
        };

        // API
        bool setState(State state);
        State getState();

     private:
        // Constants
        const char * TAG = "StatusManager";
        
        // Members
        State state;
        
        // State handlers
        bool _handleAwakeState();
        bool _handleAsleepState();
};
}
```

How should I declare my class TAG ?

Moreover, I have a **lot** of warnings like this:

```
In file included from /Users/ces/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal.h:45:0,
                 from /Users/ces/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:35,
                 from /Users/ces/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiUdp.h:38,
                 from /Users/ces/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiUdp.cpp:19:
/Users/ces/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiUdp.cpp: In member function 'uint8_t WiFiUDP::begin(IPAddress, uint16_t)':
/Users/ces/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-log.h:142:84: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
 #define log_e(format, ...) do {log_to_esp(TAG, ESP_LOG_ERROR, format, ##__VA_ARGS__);}while(0)
```

For a lot of included files that comes from external libraries (ie not related with my project code).

Thanks a lot
Charles

cyberman54, 2022-02-26:
same problem here. Any solution to this?

VojtechBartoska, 2022-04-06:
Hello @salmon-charles and @cyberman54. Are you able to test your issue on development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) to check if this is still valid? You can take a look on [Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release version in Arduino IDE.

cyberman54, 2022-04-06:
@VojtechBartoska 
Same problem, see logs below.

Test environment:

IDE: platformIO with arduino-espressif32 as framework
PLATFORM: Espressif 32 (3.3.1+sha.3784198)
PACKAGES:
 - framework-arduinoespressif32 2.0.3+sha.1e388a24 
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - tool-openocd-esp32 2.1000.20210721 (10.0)
 - toolchain-xtensa-esp32 8.4.0+2021r2-patch3
build_flags = '-D USE_ESP_IDF_LOG'

Result:

```
/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiServer.h: In constructor 'WiFiServer::WiFiServer(uint16_t, uint8_t)':   
/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-log.h:99:69: error: 'TAG' was not declared in this scope
 #define log_v(format, ...) do {ESP_LOG_LEVEL_LOCAL(ESP_LOG_VERBOSE, TAG, format, ##__VA_ARGS__);}while(0)
                                                                     ^~~
/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/log/include/esp_log.h:421:81: note: in definition of macro 'ESP_LOG_LEVEL'
         if (level==ESP_LOG_ERROR )          { esp_log_write(ESP_LOG_ERROR,      tag, LOG_FORMAT(E, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
                                                                                 ^~~
/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-log.h:99:32: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
 #define log_v(format, ...) do {ESP_LOG_LEVEL_LOCAL(ESP_LOG_VERBOSE, TAG, format, ##__VA_ARGS__);}while(0)
                                ^~~~~~~~~~~~~~~~~~~
/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiServer.h:42:7: note: in expansion of macro 'log_v'
       log_v("WiFiServer::WiFiServer(port=%d, ...)", port);
       ^~~~~
```

etc. etc.

me-no-dev, 2022-05-09:
There are misuses in the different posts.
In the OP, a line above above the class declaration with content `#undef TAG` would have solved the issue.

In between the versions that the OP tested agains and @cyberman54 tested agains, `TAG` was commented out from `esp32-hal-log.h`: https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-log.h#L193
So now it actually MUST be defined in one way or another in the file that would be logging, for this to work.
So add something like `const char * TAG = "YourDebugTAG";` to the file and it will work.

closed 2022-05-09 by me-no-dev

cyberman54, 2022-05-10:
If i try to compile a project which is using arduino style logging (e.g. using `log_e("xyz");`) with build flag `'-D USE_ESP_IDF_LOG'` set and `static const char TAG[] = __FILE__;` set, i still get lots of macro expansion errors like this:

```
/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/cores/esp32/esp32-hal-log.h:163:67: error: 'TAG' was not declared in this scope
 #define log_e(format, ...) do {ESP_LOG_LEVEL_LOCAL(ESP_LOG_ERROR, TAG, format, ##__VA_ARGS__);}while(0)
                                                                   ^~~
/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/esp32/include/log/include/esp_log.h:425:81: note: in definition of macro 'ESP_LOG_LEVEL'
         else                                { esp_log_write(ESP_LOG_INFO,       tag, LOG_FORMAT(I, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
                                                                                 ^~~
/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/cores/esp32/esp32-hal-log.h:163:32: note: in expansion of 
macro 'ESP_LOG_LEVEL_LOCAL'
 #define log_e(format, ...) do {ESP_LOG_LEVEL_LOCAL(ESP_LOG_ERROR, TAG, format, ##__VA_ARGS__);}while(0)
                                ^~~~~~~~~~~~~~~~~~~
/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/libraries/FS/src/vfs_api.cpp:194:9: note: in expansion of 
macro 'log_e'
         log_e("xyz");
         ^~~~~
```

@me-no-dev At this point i am lost what to do?

salmon-charles, 2022-06-15:
I have started a fresh new project and I still have errors.

platform.ini
```
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git
board = esp32dev
framework = arduino
platform_packages =
    framework-arduinoespressif32@https://github.com/espressif/arduino-esp32#master
build_flags = -DUSE_ESP_IDF_LOG=1
```

main.cpp
```
#include <Arduino.h>

static const char * TAG = "YourDebugTAG";

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

Output:
```
Compiling .pio/build/esp32dev/FrameworkArduino/esp32-hal-bt.c.o
In file included from /Users/ces/.platformio/packages/framework-arduinoespressif32@src-e36c2260a49b9bf6768237494cebb35d/cores/esp32/esp32-hal-log.h:189,
                 from /Users/ces/.platformio/packages/framework-arduinoespressif32@src-e36c2260a49b9bf6768237494cebb35d/cores/esp32/WMath.cpp:30:
/Users/ces/.platformio/packages/framework-arduinoespressif32@src-e36c2260a49b9bf6768237494cebb35d/cores/esp32/WMath.cpp: In function 'long int map(long int, long int, long int, long int, long int)':
/Users/ces/.platformio/packages/framework-arduinoespressif32@src-e36c2260a49b9bf6768237494cebb35d/cores/esp32/esp32-hal-log.h:163:67: error: 'TAG' was not declared in this scope
 #define log_e(format, ...) do {ESP_LOG_LEVEL_LOCAL(ESP_LOG_ERROR, TAG, format, ##__VA_ARGS__);}while(0)
                                                                   ^~~
/Users/ces/.platformio/packages/framework-arduinoespressif32@src-e36c2260a49b9bf6768237494cebb35d/tools/sdk/esp32/include/log/include/esp_log.h:421:81: note: in definition of macro 'ESP_LOG_LEVEL'
         if (level==ESP_LOG_ERROR )          { esp_log_write(ESP_LOG_ERROR,      tag, LOG_FORMAT(E, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
                                                                                 ^~~
/Users/ces/.platformio/packages/framework-arduinoespressif32@src-e36c2260a49b9bf6768237494cebb35d/cores/esp32/esp32-hal-log.h:163:32: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
 #define log_e(format, ...) do {ESP_LOG_LEVEL_LOCAL(ESP_LOG_ERROR, TAG, format, ##__VA_ARGS__);}while(0)
                                ^~~~~~~~~~~~~~~~~~~
/Users/ces/.platformio/packages/framework-arduinoespressif32@src-e36c2260a49b9bf6768237494cebb35d/cores/esp32/WMath.cpp:72:9: note: in expansion of macro 'log_e'
         log_e("map(): Invalid input range, min == max");
         ^~~~~
/Users/ces/.platformio/packages/framework-arduinoespressif32@src-e36c2260a49b9bf6768237494cebb35d/cores/esp32/esp32-hal-log.h:163:67: error: 'TAG' was not declared in this scope
 #define log_e(format, ...) do {ESP_LOG_LEVEL_LOCAL(ESP_LOG_ERROR, TAG, format, ##__VA_ARGS__);}while(0)
```

What should I do ?

Thanks !

salmon-charles, 2022-06-15:
In fact, keeping

```
#ifndef TAG
#define TAG "ARDUINO"
#endif
```

in esp32-hal-log.h works (assuming TAG is not defined elsewhere in the code).

What I dont understand, is that `-DTAG="DEBUG"` in `build_flags` does not work.

salmon-charles, 2022-06-15:
`-DTAG="\"DEBUG\""` seems to work

salmon-charles, 2022-06-15:
Still one error in my project:

```
compiling .pio/build/esp32dev/lib4be/WiFiClientSecure/ssl_client.cpp.o
<command-line>: error: expected identifier or '(' before string constant
/Users/ces/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/esp_crt_bundle.c:25:20: note: in expansion of macro 'TAG'
 static const char *TAG = "esp-x509-crt-bundle";
```

salmon-charles, 2022-06-21:
See #6893