espressif/arduino-esp32 #4991: esp_ota_write_with_offset doesn't seemed to work on Arduino

gpfreund, 2021-03-31:
### Hardware:
Board:							Heltec WiFi Lora 32v2/ESP32 Wrover
Core Installation version:			1.0.6
IDE name:						Visual Micro/Arduino IDE 1.83
Flash Frequency:					40Mhz
PSRAM enabled:                                    No
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
We are implementing an Ota update for a rather slow connection (GPRS/IoT-NP) using the esp_ota API. The update should potentially span multiple boots. However the version on the Arduino branch seems rather outdated and doesn't support `esp_ota_write_with_offset()` which is what we really need.

Is there any way to updated the library at least locally?

Thanks...

gpfreund, 2021-03-31:
Thanks @lbernstone - that worked. Much appreciated. The adafruit guide is concise and helpful.

Some caveats: 

- Switching to the v4,2 branch changes the settings of the c/c++ compiler - it's becoming a lot stricter (at least with Visual Micro - I didn't check the Arduino IDE). I didn't even know that you actually can do that via a library. Things like `char* = "myString";` instead of `const char* = "myString";`, functions with missing returns or signed/unsigned mismatches create warnings or errors where there were none before. Non of it is exactly wrong or bad but it might trip up a lot of users (unless this is just a temporary part of the active development process). 
- `esp_ota_write_with_offset()` is waiting for `esp_ota_end()`  to check the magic byte (after the whole image is written). I understand why but IMHO it should check every time when writing to offset 0. 
- The image size is significantly bigger with v4.2. Mine went from about 840k to 960k. Not sure if that due to additional functionality, temporary debug code, different code generator or???

lbernstone, 2021-03-31:
1) New version = newer gcc version.  This required some changes to behavior.  You should always be using const char when possible.  That maps the string in the code space instead of active memory. Proper coding should be encouraged.
2) That function comes from upstream.  You can submit your suggestion at https://github.com/espressif/esp-idf/issues
3) Windows 3.1 was about 30MB.  Windows 10 is ~30GB :smile:

gpfreund, 2021-03-31:
@Ibernstone: No argument re. validity of those coding standards. Arduino has been pretty lenient (or sloppy, depending on you perspective). It's just unusual to have changes in behavior triggered by a lib upgrade. Right now it's not a big deal - the upgrade is not available from within Arduino's IDE library manager but once you take that step I would at least prepare an FAQ.

Speaking of which (and off topic): Since the upgrade to 4.2 the Arduino IDE won't start any more. The flash screen shows "Loading libraries" and "Preparing boards" and then just dies - the process goes away completely. Not absolutely sure if this is related but possible. I tried a reboot with no success. It's the latest version from arduino.cc, not the Windows App shop. Visual Micro seems happy so it's no big deal for me. I'll leave it alone in case you need to do any trouble shooting/diagnostics. Happy to write up a separate ticket if need be.

lbernstone, 2021-03-31:
I haven't heard of straight up crashes from a sketchbook repo install. Oddly, this gets a lot more testing in linux & mac.
Can you try running arduino from a terminal and see if Windows behaves enough to output an error message there?

gpfreund, 2021-03-31:
I ran `arduino_debug.exe` on the console and this seems to be the culprit:
```
Preparing boards...
2021-03-31T19:39:25.060Z INFO c.a.u.n.HttpConnectionManager:153 [cc.arduino.packages.discoverers.serial.SerialDiscovery] Connect to https://builder.arduino.cc/builder/v1/boards/0x10C4/0xEA60, method=GET, request id=AF86983267784A75
java.lang.NullPointerException
        at processing.app.Base.rebuildProgrammerMenu(Base.java:1705)
        at processing.app.Base.<init>(Base.java:286)
        at processing.app.Base.main(Base.java:150)
```
Not sure what I'm looking at but happy to upload the whole log if needed. I would suggest to possibly do a new clean install and see if it goes away. I did some other changes in the way libraries load so entirely possible that I screwed something up.

closed 2021-04-01 by gpfreund