espressif/arduino-esp32 #4994: Cant include custom lib from /components/arduino/libraries

Trackhe, 2021-03-31:
Where i need to change something that i can self puttet librarys include?

i have done 
mkdir -p components && \
cd components && \
git clone https://github.com/espressif/arduino-esp32.git arduino && \
cd arduino && \
git submodule update --init --recursive && \

copie 2 Arduino libraries to the libraries dictory and 

cd ../.. && \
make menuconfig

i also changed the CMakeLists.txt but i cant include the 2 librarys to my main.cpp

chegewara, 2021-03-31:
2 things
1) paste logs
2) use idf.py instead of make

lbernstone, 2021-03-31:
3rd party libraries go in your sketchbook folder.  Typically <HOME>/Arduino/libraries. See File->Preferences in Arduino IDE.

Trackhe, 2021-03-31:
thank you for your help but i dont use the arduino ide i use the visual studio also also used idf.py but while i try to reproduce my problem i get other things that not work. if i get the same problem and i can reproduce it i reopen the ticket.

closed 2021-03-31 by Trackhe

chegewara, 2021-03-31:
You can use visual studio or even notepad, but without logs we cant help you

Trackhe, 2021-03-31:
actually i hang on ``` Linking .pio/build/esp32dev/firmware.elf
/Users/trackhe/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/esp-idf/arduino/libarduino.a(SPIFFS.o):(.rodata._ZTI10SPIFFSImpl[typeinfo for SPIFFSImpl]+0x8): undefined reference to `typeinfo for VFSImpl'
/Users/trackhe/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/esp-idf/arduino/libarduino.a(SPIFFS.o):(.rodata._ZTV10SPIFFSImpl[vtable for SPIFFSImpl]+0x10): undefined reference to `VFSImpl::open(char const*, char const*)'
/Users/trackhe/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/esp-idf/arduino/libarduino.a(SPIFFS.o):(.rodata._ZTV10SPIFFSImpl[vtable for SPIFFSImpl]+0x18): undefined reference to `VFSImpl::rename(char const*, char const*)'
/Users/trackhe/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/esp-idf/arduino/libarduino.a(SPIFFS.o):(.rodata._ZTV10SPIFFSImpl[vtable for SPIFFSImpl]+0x1c): undefined reference to `VFSImpl::remove(char const*)'
/Users/trackhe/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/esp-idf/arduino/libarduino.a(SPIFFS.o):(.rodata._ZTV10SPIFFSImpl[vtable for SPIFFSImpl]+0x20): undefined reference to `VFSImpl::mkdir(char const*)'
/Users/trackhe/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/esp-idf/arduino/libarduino.a(SPIFFS.o):(.rodata._ZTV10SPIFFSImpl[vtable for SPIFFSImpl]+0x24): undefined reference to `VFSImpl::rmdir(char const*)'
/Users/trackhe/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/esp-idf/arduino/libarduino.a(SD.o):(.literal._Z41__static_initialization_and_destruction_0ii+0x4): undefined reference to `vtable for VFSImpl'
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32dev/firmware.elf] Error 1```

chegewara, 2021-03-31:
Did you copy sdkconfig from arduino-esp32 to your project?

Trackhe, 2021-03-31:
No because it work before. and i thougt idf.py menuconfig generate the sdkconfig. i dont want to override my setted settings. but i try it. thanks

Trackhe, 2021-03-31:
i copied and build direktly with the untouched copied sdk from arduino-esp32 

and i get nearly the same
```
/Users/trackhe/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/esp-idf/arduino/libarduino.a(main.o): in function `app_main':
/Users/trackhe/Documents/SHP/htk/components/arduino/cores/esp32/main.cpp:41: multiple definition of `app_main'; .pio/build/esp32dev/main/main.o:/Users/trackhe/Documents/SHP/htk/main/main.cpp:34: first defined here
/Users/trackhe/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/esp-idf/arduino/libarduino.a(SPIFFS.o):(.rodata._ZTV10SPIFFSImpl[vtable for SPIFFSImpl]+0x10): undefined reference to `VFSImpl::open(char const*, char const*)'
/Users/trackhe/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/esp-idf/arduino/libarduino.a(SPIFFS.o):(.rodata._ZTV10SPIFFSImpl[vtable for SPIFFSImpl]+0x18): undefined reference to `VFSImpl::rename(char const*, char const*)'
/Users/trackhe/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/esp-idf/arduino/libarduino.a(SPIFFS.o):(.rodata._ZTV10SPIFFSImpl[vtable for SPIFFSImpl]+0x1c): undefined reference to `VFSImpl::remove(char const*)'
/Users/trackhe/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/esp-idf/arduino/libarduino.a(SPIFFS.o):(.rodata._ZTV10SPIFFSImpl[vtable for SPIFFSImpl]+0x20): undefined reference to `VFSImpl::mkdir(char const*)'
/Users/trackhe/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/esp-idf/arduino/libarduino.a(SPIFFS.o):(.rodata._ZTV10SPIFFSImpl[vtable for SPIFFSImpl]+0x24): undefined reference to `VFSImpl::rmdir(char const*)'
/Users/trackhe/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32dev/esp-idf/arduino/libarduino.a(SD.o):(.literal.startup._GLOBAL__sub_I__ZN2fs4SDFSC2ESt10shared_ptrINS_6FSImplEE+0x4): undefined reference to `vtable for VFSImpl'
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32dev/firmware.elf] Error 1
```

chegewara, 2021-03-31:
When you copy sdkconfig you can enter menuconfig and change most settings, but to make arduino as component working you have to start with that sdkconfig.

chegewara, 2021-03-31:
Try to do `idf.py fullclean` or even delete build directory and rerun.

chegewara, 2021-03-31:
I see one more thing i missed earlier. Why you are mixing arduino as component with platformio? I did never try that, but it most likely wont work.

chegewara, 2021-03-31:
You are right, it is not important, but as i said, it is not compatible.
You can easy move code between platformio and arduino as component.

Trackhe, 2021-03-31:
maybe i should reprogramm my arduino files to the esp functions