espressif/arduino-esp32 #4998: Missing #idf-release/v4.0 branch

enriquewph, 2021-04-01:
Hello there.

It seems the branch  **#idf-release/v4.0** got deleted a few hours ago.

I was working yesterday with this branch and everything worked ok.
However, i cant get anything to compile with this new branch **#idf-master**

- Is **#idf-release/v4.0** coming back?
- What esp-idf version/release am i supossed to use to work with **#idf-master**?

Best Regards, Enrique.

h2zero, 2021-04-01:
Looks like #idf-master branch is compiled using https://github.com/espressif/esp-idf/commit/2bfdd036b2dbd07004c8e4f2ffc87c728819b737

castorw, 2021-04-02:
Hi there, same here. My project is now broken, since it relied on the mentioned branch of arduino-esp32. Using idf-master results in: `failed to resolve component 'tinyusb'` error.

Can anybody please explain why the branch was removed and how to get this working again?

nlac, 2021-04-02:
Looking into [this header](https://github.com/espressif/arduino-esp32/blob/idf-master/tools/sdk/esp32/include/config/sdkconfig.h) in branch `idf-master`, there are two rows:
`#define CONFIG_ARDUINO_IDF_COMMIT "1067b2870"`
`#define CONFIG_ARDUINO_IDF_BRANCH "master"`

Until now, it was a working reference for me (meaning branch `idf-master` is compiled using ESP-IDF [commit](https://github.com/espressif/esp-idf/tree/1067b28707e527f177752741e3aa08b5dc64a4d7) "1067b2870") but it is not any more. For example esp_littlefs is not recent in that commit but branch `idf-master` has references to that.

VioletGiraffe, 2021-04-02:
According to various discussions I've found, `#idf-release/v4.0` is required to compile using `-std=c++17` (and it sucks we even have to fiddle so much to get C++17 to work in 2021!). Is there a current (and, preferably, future-proof) solution for C++17 on ESP32 + Arduino framework?

chegewara, 2021-04-02:
1) idf v4.0 is not a long term release, so the next arduino version (v2.0.x) will be based on v4.3 most likely (idf-release/v4.2 has been merged to master recently)
2) esp-idf API is based on C++11, thats why arduino is also based on that version, even if it is possible to enable c++17 in idf projects

Im not saying it is good or bad that some branches has been deleted, but you have to understand that all those branches were development branches and no longer maintained.

VioletGiraffe, 2021-04-02:
I understand that was a development branch, obviously we as developers shouldn't have to need _that_ branch, we just need our client code to work (preferably with the stable release version of the platform).
The current compiler itself is GCC 8.2.0 which is good enough (it knows `-std=c++2a`), but the standard library is missing stuff and I can't get it all to compile correctly. For example, I can't find `std::size` anywhere. Any hints for me on how to fixed would be highly appreciated!

nlac, 2021-04-03:
I'm really wondering if someone could suggest here the most recent pair of commit numbers of arduino-esp32 and esp-idf that surely work together for a new "arduino_as_component" project? I'm over days of trials of combinations but i always run into "Failed to resolve component *'" (ethernet, arduino_tinyusb etc) issues when trying to build the simplest "arduino-as- component" hello world project.

chegewara, 2021-04-03:
I cant say it is most recent, because its at least few months old, but works good to me in commercial app:
https://github.com/espressif/esp-idf/commit/b0150615dff529662772a60dcb57d5b559f480e2
https://github.com/espressif/arduino-esp32/tree/71e2142c7f73a043331c42ecdd900b3c2c81b84a

The only thing you have to do is to edit CMakeLists.txt in arduino-esp32, to fix issue with 1 or 2 libraries:
https://github.com/Othernet-Project/Dreamcatcher/blob/master/CMakeLists.txt_arduino-esp32

marcovannoord, 2021-04-04:
Because i had to get this working for a commercial application too, i have forked a recent fork of this repo, that still had the #idf-release/4.0 branch available.  
https://github.com/marcovannoord/arduino-esp32/tree/idf-release/v4.0  
Since i rely on this 4.0 branch for an esp32-arduino build using PlatformIO, to get the project to build, 
i just replace:  
```platform_packages =  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#idf-release/v4.0```  
with:  
```platform_packages =  framework-arduinoespressif32 @ https://github.com/marcovannoord/arduino-esp32.git#idf-release/v4.0``` in platformio.ini

VioletGiraffe, 2021-04-04:
@marcovannoord, you're the savior! And what are your `framework` and `platform` set to in `platformio.ini`?

marcovannoord, 2021-04-04:
I've left those at:
```
platform = espressif32
framework = arduino, espidf```

VioletGiraffe, 2021-04-04:
This is what I've tried as per your advice, and it doesn't work:

```[env:esp32]
platform_packages = framework-arduinoespressif32 @ https://github.com/marcovannoord/arduino-esp32.git#idf-release/v4.0
platform = espressif32
framework = arduino
board = esp32dev
```

Specifically, it compiles but still uses the current release version of the build tools (ancient GCC 5.2.0). I also didn't notice PlatformIO downloading anything new from git.

marcovannoord, 2021-04-04:
Hmm that's odd, after replacing the platform_packages, i did notice PlatformIO downloading the other branch, and also in the Github Actions script, i see that it now builds successfully again (GH actions was how i first noticed my broken build, since it does not cache any repositories, and will always start with a clean slate).
Perhaps removing the framework-arduinoespressif32 from the .platformio\packages folder (in your home folder )and cleaning the .pio folder in the project directory will trigger a re-download?

marcovannoord, 2021-04-04:
To demonstrate a working version, i have placed 
https://gist.github.com/marcovannoord/4275a25639427e13d6e74f17636bbd5f
in the .github\workflows folder and that shows the project compiling successfully.

VioletGiraffe, 2021-04-04:
@marcovannoord, thank you very much! It does indeed work. Turns out, I didn't see the new platform being downloaded because it doesn't happen when you build the project or issue "Platformio: Update all" command; it happens right away as soon as PIO parses the changed .ini file (which, apparently, it does as you edit it). 
Also, I needed first and foremost the new(er) GCC 8 compiler, which is fully C++17-capable unlike the old one (GCC 5.2.0), for that I also added `toolchain-xtensa32 @ 3.80200.200512` to `platform_packages`. So it looks like this now:

```
platform_packages = 
	framework-arduinoespressif32 @ https://github.com/marcovannoord/arduino-esp32.git#idf-release/v4.0
	toolchain-xtensa32 @ 3.80200.200512
```
And now all my build errors due to non-C++17-conformant compiler and standard library are gone!

Also, thank you for the Github Actions script, I'll use it as a template for all my PlatformIO projects (which currently use Travis CI).

enriquewph, 2021-04-04:
UPDATE: I'm still trying @chegewara solution, but platform io clones whatever it wants and not the full repo.. i'm working on it.
Also, the fork made by @marcovannoord and the c++17 fix from @VioletGiraffe got me working again. Thanks!

chegewara, 2021-04-04:
Well, in my case posted links are actually from old idf-release/v4.2 and idf master commit compatible with it. It is because i am using it with S2.

enriquewph, 2021-04-08:
looks like #idf-master branch got deleted for version 2.0.0 #4996 as of now, building using a **previously forked repository** seems to work ok, but things are moving fast.. cant wait to see what 2.0.0 brings to us!

closing this issue.

> Because i had to get this working for a commercial application too, i have forked a recent fork of this repo, that still had the #idf-release/4.0 branch available. [...]

closed 2021-04-08 by enriquewph

VioletGiraffe, 2021-04-08:
> cant wait to see what 2.0.0 brings to us
I hope it brings C++17 support out of the box without having to spend 2 days to get it working using unofficial forks!

EmilianoQS, 2021-04-14:
Same issue here.

I used marcovannoord branch, but I get the following error:
```
-- Configuring incomplete, errors occurred!
See also "C:/Users/Emiliano/.platformio/platforms/espressif32/examples/espidf-ar
duino-blink/.pio/build/mhetesp32devkit/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any of the parent directories): .git
CMake Error at C:/Users/Emiliano/.platformio/packages/framework-espidf@3.40001.2
00521/tools/cmake/build.cmake:185 (message):
  Failed to resolve component 'ethernet'.
```

Hope this gets fixed soon....

luckwaski, 2021-04-21:
> Well, in my case posted links are actually from old idf-release/v4.2 and idf master commit compatible with it. It is because i am using it with S2.

Hi - would you be able to share your working configuration of platformio for a esp32-s2 ardiuno+esp-idf? I was fallowing your advices but could get it working. 
Thank you!

chegewara, 2021-04-21:
I did not try to use S2 with platformio, sorry.

MikaelFerland, 2021-05-01:
Is anyone had any issue with ESP32 not defined following marcovannoord solution?

I noticed that ESP32 is defined in arduino only framework (without platform_packages).

I noticed also that it should be already defined here:
https://github.com/marcovannoord/arduino-esp32/blob/idf-release/v4.0/tools/platformio-build.py#L102

Is it supposed to be called when I click build in VSCODE using PlatformIO?
Should I define ESP32 manually when we use arduino and  espidf  together?

Thank you!

marcovannoord, 2021-05-02:
Hmm i'm not sure if i follow entirely, but building using Platformio using this config should do all the work for you, without having to define anything else:
```
platform_packages = framework-arduinoespressif32 @ https://github.com/marcovannoord/arduino-esp32.git#idf-release/v4.0
platform = espressif32
framework = arduino, espidf
board = esp32dev
```

MikaelFerland, 2021-05-02:
Sorry about my explanation.
I'm using `board = featheresp32` and tried `board = esp32dev` still the same.

Strangely ESP32 is not defined as the line 82 is greyed out.
[image]

luckwaski, 2021-05-02:
I had the same observations. You may wanna use :

```
build_flags = 
	-D ESP32=1
```

at your platformio.ini

matthew798, 2021-11-06:
@marcovannoord Thanks for providing the branch. 

Does all of this mean that there is no combination of versions (arduino/idf) that work with PIO? If I din't care what IDF version PIO uses, is there a compatible arduino version I can use instead of the forked 4.0?

enriquewph, 2021-11-06:
> @marcovannoord Thanks for providing the branch.
> 
> Does all of this mean that there is no combination of versions (arduino/idf) that work with PIO? If I don't care what IDF version PIO uses, is there a compatible arduino version I can use instead of the forked 4.0?

yes. start with a blank esp-idf project:
```ini
platform = espressif32@3.3.2
board = esp32dev
framework = espidf
```
now make sure you clone arduino-esp32 into the components folder, you should add this as a submodule:
```bash
git submodule add https://github.com/espressif/arduino-esp32.git ./components/arduino
git submodule update --init
```

lastly, create a file named: **sdkconfig.defaults** in the root directory of the project.
```ini
# Override some defaults to enable Arduino framework
CONFIG_ENABLE_ARDUINO_DEPENDS=y
CONFIG_AUTOSTART_ARDUINO=y
CONFIG_ARDUINO_RUN_CORE1=y
CONFIG_ARDUINO_RUNNING_CORE=1
CONFIG_ARDUINO_EVENT_RUN_CORE1=y
CONFIG_ARDUINO_EVENT_RUNNING_CORE=1
CONFIG_ARDUINO_UDP_RUN_CORE1=y
CONFIG_ARDUINO_UDP_RUNNING_CORE=1
CONFIG_DISABLE_HAL_LOCKS=y
CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_ERROR=y
CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL=1
CONFIG_ARDUHAL_PARTITION_SCHEME_DEFAULT=y
CONFIG_ARDUHAL_PARTITION_SCHEME="default"
CONFIG_AUTOCONNECT_WIFI=y
CONFIG_ARDUINO_SELECTIVE_WiFi=y
CONFIG_MBEDTLS_PSK_MODES=y
CONFIG_MBEDTLS_KEY_EXCHANGE_PSK=y
```

it works ok, to access menuconfig run:
```bash
pio run -t menuconfig
```

matthew798, 2021-11-06:
@enriquewph Woah! I never thought to do it that way. I guess I just assumed I _had_ to do it the PIO way. Thanks!

VojtechBartoska, 2021-11-08:
Yes, IDF 4.4 is now supported.

VioletGiraffe, 2021-11-08:
This is how my Pl;atformIO build was set up, and it worked half a year ago. I did not need to invoke `git` manually or anything like that.

```
[env:esp32]
platform_packages = 
	framework-arduinoespressif32 @ https://github.com/marcovannoord/arduino-esp32.git#idf-release/v4.0
	toolchain-xtensa32 @ 3.80200.200512
	
platform = espressif32
framework = arduino
board = esp32dev
```

the-potato-church, 2022-09-21:
Hi guys! I am having a very similar/ the same problem - since sometime around August 2022 I cannot initialise my ESP32 Arduino project. I can still build and upload it on machines with the project initialised already.

The project configuration:
```
[env:esp32dev]
platform = espressif32
board = esp32dev

framework = arduino
monitor_speed = 115200
board_build.f_cpu = 240000000L

build_unflags = 
     -std=gnu++11   ; Disable the use of gnu++11, as we want to move to c++17
build_flags = 
     -std=c++17     ; Enable the use of c++17
     -I "test/integration"
     -I "test"
     ;-DCORE_DEBUG_LEVEL=5

; Ensure that we are using esp idf 4.0 - which has c++17 support
platform_packages =
     toolchain-xtensa32 @ 3.80200.200512
     framework-arduinoespressif32 @ https://github.com/marcovannoord/arduino-esp32.git#idf-release/v4.0
```

When I try to build the project, PIO spits out this:
```
 *  The terminal process "C:\Users\***\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

 
Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------Error! Failed to extract upstream toolchainconfigurations:
Bad package version `1.22.0-80-g6c4433a-5.2.0`
You can disable this feature via the `board_build.arduino.upstream_packages = no` setting in your `platformio.ini` file. 
```

After some digging, figured out that the error likely comes back to [this](https://github.com/marcovannoord/arduino-esp32/blob/master/package/package_esp32_index.template.json#L39). I have tried playing around with the versions of all the packages and platform but without any luck. Any ideas?

Thanks for the help!