espressif/arduino-esp32 #5000: Bug in delayMicroseconds()

hzoli17, 2021-04-01:
ESP32 crashes when I call the function after ~1 hour 20 minutes of usage, because esp_timer_get_time() return a int64_t value and is parsed as uint32_t in delayMicroseconds().

maxgerhardt, 2021-04-01:
Do you have a crashlog for that? And minimal example code? 

the way I read it in 

https://github.com/espressif/arduino-esp32/blob/8dc70e0add06e75da5a255ba94fe86df98ae5ef1/cores/esp32/esp32-hal-misc.c#L149-L163

https://github.com/espressif/arduino-esp32/blob/8dc70e0add06e75da5a255ba94fe86df98ae5ef1/cores/esp32/esp32-hal-misc.c#L134-L137

https://github.com/espressif/arduino-esp32/blob/af119215353b26d944b80d7126bed68e060e6697/tools/sdk/include/esp32/esp_timer.h#L185-L190

The `int64_t` is indeed downcasted to `unsigned long` (which should be `uint32_t`) but then the value will just internally overflow and `delayMicroseconds` still handles the overflow case.

gin66, 2021-04-05:
One issue in the code is this:

Assume m=micros() = 0xffff0000 and us = 0xffff.
So the calculated value will be e=0xffffffff and no overflow has occurred.
Consequently the second while loop will be executed.
If now an interrupt or a high priority task squeezes inbetween and returns after a couple of us, then the micros() may have wrapped around. So it will need wait 4 billion us = 4000s, which could trigger a watchdog.

Better is a while loop without overflow check:
```
     uint32_t m = micros(); 
     if(us){    
         while((uint32_t)(micros() - m) < us) { 
             NOP();
          }
    }
```

closed 2021-04-05 by me-no-dev

gin66, 2021-04-05:
now I understand, why esp32 is not running stable.