espressif/arduino-esp32 #5001: ESP32-S2 "get.exe" not installing xtensa-esp32s2-elf-gcc

lewispg228, 2021-04-01:
Dear Espressif,
I have noticed something that I think may need to be updated. The windows "get.exe" file does not seem to download the necessary tools to compile for the EXP32-S2.

[image]

However, when I run the python module "get.py" it works wonderfully.

[image]

We are happy to instruct users to use the python module for installation, but I thought it was worth bringing to your attention.

Please let me know if there is anything else I can do to help resolve this issue.

Cheers and thank you very much!

Pete Lewis
SparkFun Electronics

me-no-dev, 2021-04-01:
are you sure that you ran the commands on the same branch? (`idf-master`)

lewispg228, 2021-04-01:
Whoops. I was running it on the "master" branch. Sorry for my mistake. I must have switched branches in between running get.exe and get.py. 

Thanks for getting back to me so quickly. 
Have a great rest of your day!

closed 2021-04-01 by lewispg228