espressif/arduino-esp32 #5006: ESP32 Arduino IDE esp_https_server.h: No such file or directory

idvbu, 2021-04-03:
### Hardware:
Board:							ESP32 Dev Module WEMOS LOLIN32
Core Installation version:			
IDE name:						Arduino IDE
Flash Frequency:					240Mhz
PSRAM enabled:                                     
Upload Speed:				        9600
Computer OS:                                          Windows 10

Additional Boards Manager URLs:
https://arduino.esp8266.com/stable/package_esp8266com_index.json
https://dl.espressif.com/dl/package_esp32_index.json
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json

### Description:
In Arduino sketch included library esp_https_server.h.

ISSUE:
However when run the sketch compiler it returns me following error:
esp_https_server.h: No such file or directory

Please, advice what might be the root cause of this issue.

Remark:
Note that with the other HTTP "non-ssl" library esp_http_server.h" I have no such issue and sketch is able to find reference to it.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <WiFi.h>
#include "esp_https_server.h"

void setup() {
}

void loop() {
}

chegewara, 2021-04-03:
HTTPS server is not enabled in sdkconfig, because arduino is using its own library for https server.
This is required:
https://github.com/espressif/esp-idf/blob/master/examples/protocols/https_server/simple/sdkconfig.defaults#L1

idvbu, 2021-04-11:
Hi,
Thank for your very much for your  prompt answer.

However, correct me if I am wrong but the link which you referred is for: espressif /esp-idf 
https://github.com/espressif/esp-idf/blob/master/examples/protocols/https_server/simple/sdkconfig.defaults#L1

In my case I need a solution for Arduino IDE.

Said otherwise where/how in Arduino IDE I can apply these settings referred at the previous posting:
CONFIG_ESP_HTTPS_SERVER_ENABLE=y
CONFIG_ESP_NETIF_TCPIP_ADAPTER_COMPATIBLE_LAYER=n

?

Regards.

chegewara, 2021-04-11:
You cant use it in arduino IDE. You can do it with arduino as component or with platformio or you have to use arduino-lib-builder to rebuild libraries and replace those in arduino-esp32.

idvbu, 2021-04-11:
Hi che,

Thank you for your last answer. Based on that I will continue in ESP with some other alternative for HTTPS web server and will not try to use anymore the Espressif native "esp_https_server.h" library.

Regards

closed 2021-07-11 by stale[bot]