espressif/arduino-esp32 #5012: esp 32 1.0.6 installation problem

RajeshRanjanmdp, 2021-04-04:
Board: ESP32
Core Installation version: ?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3?
IDE name: Arduino IDE

Computer OS: Windows 10

Description:
i cant install esp32 on my arduino IDE
when i try to install it an error occour
''Error downloading https://github.com/espressif/arduino-esp32/releases/download/1.0.6/esp32-1.0.6.zip''
can any one fix it
[image]

maxgerhardt, 2021-04-04:
In the lower portion there's a Java trace log talking about `readRecord`.. can you expand that? I can download the linked zip just fine.

RajeshRanjanmdp, 2021-04-04:
Error downloading https://github.com/espressif/arduino-esp32/releases/download/1.0.6/esp32-1.0.6.zip
java.lang.RuntimeException: java.lang.Exception: Error downloading https://github.com/espressif/arduino-esp32/releases/download/1.0.6/esp32-1.0.6.zip
at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:175)
at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Error downloading https://github.com/espressif/arduino-esp32/releases/download/1.0.6/esp32-1.0.6.zip
at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:149)
at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:83)
at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:60)
at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:105)
at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)
... 1 more
Caused by: javax.net.ssl.SSLException: Tag mismatch!
at sun.security.ssl.Alerts.getSSLException(Alerts.java:208)
at sun.security.ssl.SSLSocketImpl.fatal(SSLSocketImpl.java:1946)
at sun.security.ssl.SSLSocketImpl.readRecord(SSLSocketImpl.java:1022)
at sun.security.ssl.SSLSocketImpl.readDataRecord(SSLSocketImpl.java:933)
at sun.security.ssl.AppInputStream.read(AppInputStream.java:105)
at java.io.BufferedInputStream.read1(BufferedInputStream.java:284)
at java.io.BufferedInputStream.read(BufferedInputStream.java:345)
at sun.net.www.MeteredStream.read(MeteredStream.java:134)
at java.io.FilterInputStream.read(FilterInputStream.java:133)
at sun.net.www.protocol.http.HttpURLConnection$HttpInputStream.read(HttpURLConnection.java:3444)
at sun.net.www.protocol.http.HttpURLConnection$HttpInputStream.read(HttpURLConnection.java:3437)
at cc.arduino.utils.network.FileDownloader.readStreamCopyTo(FileDownloader.java:268)
at cc.arduino.utils.network.FileDownloader.openConnectionAndFillTheFile(FileDownloader.java:247)
at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:182)
at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:129)
at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:147)
... 5 more
Caused by: javax.crypto.AEADBadTagException: Tag mismatch!
at com.sun.crypto.provider.GaloisCounterMode.decryptFinal(GaloisCounterMode.java:578)
at com.sun.crypto.provider.CipherCore.finalNoPadding(CipherCore.java:1049)
at com.sun.crypto.provider.CipherCore.doFinal(CipherCore.java:985)
at com.sun.crypto.provider.AESCipher.engineDoFinal(AESCipher.java:491)
at javax.crypto.Cipher.doFinal(Cipher.java:2376)
at sun.security.ssl.CipherBox.decrypt(CipherBox.java:461)
at sun.security.ssl.InputRecord.decrypt(InputRecord.java:172)
at sun.security.ssl.SSLSocketImpl.readRecord(SSLSocketImpl.java:1017)
... 18 more
Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
at cc.arduino.contributions.packages.ui.ContributedPlatformTableCellJPanel.update(ContributedPlatformTableCellJPanel.java:226)
at cc.arduino.contributions.packages.ui.ContributedPlatformTableCellRenderer.getTableCellRendererComponent(ContributedPlatformTableCellRenderer.java:47)
at javax.swing.JTable.prepareRenderer(JTable.java:5729)
at javax.swing.plaf.basic.BasicTableUI.paintCell(BasicTableUI.java:2114)
at javax.swing.plaf.basic.BasicTableUI.paintCells(BasicTableUI.java:2016)
at javax.swing.plaf.basic.BasicTableUI.paint(BasicTableUI.java:1812)
at javax.swing.plaf.ComponentUI.update(ComponentUI.java:161)
at javax.swing.JComponent.paintComponent(JComponent.java:780)
at javax.swing.JComponent.paint(JComponent.java:1056)
at javax.swing.JComponent.paintChildren(JComponent.java:889)
at javax.swing.JComponent.paint(JComponent.java:1065)
at javax.swing.JViewport.paint(JViewport.java:728)
at javax.swing.JComponent.paintChildren(JComponent.java:889)
at javax.swing.JComponent.paint(JComponent.java:1065)
at javax.swing.JComponent.paintToOffscreen(JComponent.java:5210)
at javax.swing.RepaintManager$PaintManager.paintDoubleBuffered(RepaintManager.java:1579)
at javax.swing.RepaintManager$PaintManager.paint(RepaintManager.java:1502)
at javax.swing.RepaintManager.paint(RepaintManager.java:1272)
at javax.swing.JComponent._paintImmediately(JComponent.java:5158)
at javax.swing.JComponent.paintImmediately(JComponent.java:4969)
at javax.swing.RepaintManager$4.run(RepaintManager.java:831)
at javax.swing.RepaintManager$4.run(RepaintManager.java:814)
at java.security.AccessController.doPrivileged(Native Method)
at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
at javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:814)
at javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:789)
at javax.swing.RepaintManager.prePaintDirtyRegions(RepaintManager.java:738)
at javax.swing.RepaintManager.access$1200(RepaintManager.java:64)
at javax.swing.RepaintManager$ProcessingRunnable.run(RepaintManager.java:1732)
at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
[... 161 lines cut ...]
at sun.security.ssl.SSLSocketImpl.readRecord(SSLSocketImpl.java:1022)
at sun.security.ssl.SSLSocketImpl.readDataRecord(SSLSocketImpl.java:933)
[... cut ...]
at com.sun.crypto.provider.GaloisCounterMode.decryptFinal(GaloisCounterMode.java:578)
at com.sun.crypto.provider.CipherCore.finalNoPadding(CipherCore.java:1049)
at com.sun.crypto.provider.CipherCore.doFinal(CipherCore.java:985)
at com.sun.crypto.provider.AESCipher.engineDoFinal(AESCipher.java:491)
at javax.crypto.Cipher.doFinal(Cipher.java:2376)
at sun.security.ssl.CipherBox.decrypt(CipherBox.java:461)
at sun.security.ssl.InputRecord.decrypt(InputRecord.java:172)
at sun.security.ssl.SSLSocketImpl.readRecord(SSLSocketImpl.java:1017)
... 18 more

maxgerhardt, 2021-04-04:
> Caused by: java.lang.Exception: Error downloading https://github.com/espressif/arduino-esp32/releases/download/1.0.6/esp32-1.0.6.zip
> Caused by: javax.net.ssl.SSLException: Tag mismatch!
> Caused by: javax.crypto.AEADBadTagException: Tag mismatch!
>at com.sun.crypto.provider.GaloisCounterMode.decryptFinal(GaloisCounterMode.java:578)

Interesting, a crypto error while downloading. Do you have any weird Antivirus that could interfere? Can you download the linked ZIP file manually?

RajeshRanjanmdp, 2021-04-04:
from where i can download zip file?
can u send me the link?

maxgerhardt, 2021-04-04:
The link is already in the quote above, esp32-1.0.6.zip..

RajeshRanjanmdp, 2021-04-04:
it works
thanks

maxgerhardt, 2021-04-04:
Well but if you can download the above zip file this doesn't resolve the error in the Arduino IDE though? Or is it now magically working?

closed 2021-07-11 by stale[bot]