espressif/arduino-esp32 #5013: ESP32 with BLE and LCD Interface

Cheems109, 2021-04-04:
I tried to run a program which uses BLE on the ESP32, and takes a message from the phone connected to it and prints that on the serial monitor. That part of the program work properly. But I also want to interface an LCD screen to it so that I can also display the message on the LCD Screen. I am new to this and I want to learn how to convert or change the data type I received as the message from the phone to be able to display on the LCD Screen. Please provide me hints rather than the complete solution, because I want to learn it on my own. Thank you 

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3?
IDE name:						Arduino IDE? 
Flash Frequency:					?40Mhz
PSRAM enabled:                                    ?no? ?yes
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
I need to get the message I receive in "value" to be displayed on a 16x2 LCD Screen. I'm pretty new to Bluetooth Connections.
Please help.

### Sketch:  

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>

#define SERVICE_UUID        " "
#define CHARACTERISTIC_UUID "  "

 class MyCallbacks: public BLECharacteristicCallbacks {
    void onWrite(BLECharacteristic *pCharacteristic) {
      std::string value = pCharacteristic->getValue();

      if (value.length() > 0) {
        Serial.println("*********");
        Serial.print("New value: ");
        for (int i = 0; i < value.length(); i++)
          Serial.print(value[i]);

        Serial.println();
        Serial.println("*********");
}
}
};

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");

  BLEDevice::init("Pikachu");
  BLEServer *pServer = BLEDevice::createServer();
  BLEService *pService = pServer->createService(SERVICE_UUID);
  BLECharacteristic *pCharacteristic = pService->createCharacteristic(
                                         CHARACTERISTIC_UUID,
                                         BLECharacteristic::PROPERTY_READ |
                                         BLECharacteristic::PROPERTY_WRITE
                                       );

  pCharacteristic->setValue("I am right here!");
  pCharacteristic->setCallbacks(new MyCallbacks());
  pService->start();
  // BLEAdvertising *pAdvertising = pServer->getAdvertising();  // this still is working for backward compatibility
  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  pAdvertising->addServiceUUID(SERVICE_UUID);
  pAdvertising->setScanResponse(true);
  pAdvertising->setMinPreferred(0x06);  // functions that help with iPhone connections issue
  pAdvertising->setMinPreferred(0x12);
  BLEDevice::startAdvertising();
  Serial.println("Characteristic defined! Now you can read it in your phone!");
  
  
}
}

lbernstone, 2021-04-04:
Set a global variable to your "value".
Have a separate [Ticker](https://github.com/espressif/arduino-esp32/blob/master/libraries/Ticker/examples/Blinker/Blinker.ino) that updates your LCD every second with that variable.

Cheems109, 2021-04-10:
> Set a global variable to your "value".
> Have a separate [Ticker](https://github.com/espressif/arduino-esp32/blob/master/libraries/Ticker/examples/Blinker/Blinker.ino) that updates your LCD every second with that variable.

Thanks for the tip, it really helped me understand the new concept of Tickers and I also got the code working the way I wanted.

closed 2021-04-10 by Cheems109