espressif/arduino-esp32 #5014: Build error with platformio/framework-arduinoespressif32 @ ~3.10006.0

gin66, 2021-04-05:
Build error with `framework-arduinoespressif32 @ ~3.10006.0`  as per this [log](https://github.com/gin66/FastAccelStepper/runs/2261202956?check_suite_focus=true). Works well with `framework-arduinoespressif32 @ 3.10005.210308` (see [log](https://github.com/gin66/FastAccelStepper/runs/2261259799?check_suite_focus=true) ):
```
Tool Manager: Installing platformio/framework-arduinoespressif32 @ ~3.10006.0
123
Downloading...
124
Unpacking...
125
Tool Manager: framework-arduinoespressif32 @ 3.10006.210326 has been installed!
126
/home/runner/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c: In function 'spiTransferBytesNL':
127
/home/runner/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:922:39: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
128
                 uint8_t * last_out8 = &result[c_longs-1];
129
                                       ^
130
/home/runner/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:923:40: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
131
                 uint8_t * last_data8 = &last_data;
132
                                        ^
133
cc1: all warnings being treated as errors
134
*** [.pio/build/esp32/FrameworkArduino/esp32-hal-spi.c.o] Error 1
```

me-no-dev, 2021-04-05:
you would need to lower the log/debug level of the build `cc1: all warnings being treated as errors` is the problem

gin66, 2021-04-05:
Not a good idea to lower warning level. Never mind. The latest (huge) commit from #4996 fixes this problem by adding a `(uint8_t *)` before the &. So the next release will resolve the issue.

closed 2021-04-05 by gin66