espressif/arduino-esp32 #5016: In ledc_types.h, LEDC_CHANNEL_6 is skipped

Dlloydev, 2021-04-06:
In this typedef in [ledc_types.h](https://github.com/OS-Q/E52A/blob/b07fb5cff19c4d90774849cf9bbf7d7755cdb87c/components/hal/include/hal/ledc_types.h):

typedef enum {
    LEDC_CHANNEL_0 = 0, /*!< LEDC channel 0 */
    LEDC_CHANNEL_1,     /*!< LEDC channel 1 */
    LEDC_CHANNEL_2,     /*!< LEDC channel 2 */
    LEDC_CHANNEL_3,     /*!< LEDC channel 3 */
    LEDC_CHANNEL_4,     /*!< LEDC channel 4 */
    LEDC_CHANNEL_5,     /*!< LEDC channel 5 */
#if SOC_LEDC_CHANNEL_NUM > 6
    LEDC_CHANNEL_6,     /*!< LEDC channel 6 */
    LEDC_CHANNEL_7,     /*!< LEDC channel 7 */
#endif
    LEDC_CHANNEL_MAX,
} ledc_channel_t;

Channel 6 is skipped ... should be CHANNEL_NUM > 5

lbernstone, 2021-04-06:
I haven't looked thoroughly, but SOC_LEDC_CHANNEL_NUM probably isn't indexed at 0, while the LEDC_CHANNEL_X is.

Dlloydev, 2021-04-06:
Thanks, I'll open an issue there.
Just a note that in the following typedef (ledc_timer_bit_t), they used SOC_LEDC_CHANNEL_NUM being indexed at 0.

closed 2021-04-06 by Dlloydev