espressif/arduino-esp32 #5018: fatal error occurred: Timed out waiting for packet header

lroboredo, 2021-04-06:
I am tying up load a sketch and I get the following result:

[image]

Please help how I can fixit?

[image]

lbernstone, 2021-04-06:
Bad usb cable

lroboredo, 2021-04-06:
Thank you for your sugestion.
I change the usb cable by a ew one certified 480Mbps and I got the same!

lroboredo, 2021-04-06:
[image]
This is the configuration.

lroboredo, 2021-04-06:
[image]
When I press Boot button I get this answer in the COM8 monitor

lbernstone, 2021-04-06:
Verify that nothing is using gpio pins 6-11.  Try setting flash to DIO/40MHz. It seems like your flash may be damaged.

lroboredo, 2021-04-06:
Now I got
[image]

lroboredo, 2021-04-06:
This is the board

lroboredo, 2021-04-06:
[image]

closed 2021-07-11 by stale[bot]