espressif/arduino-esp32 #5019: SD open(): behavior on missing path on write

fabiuz7, 2021-04-06:
Hi, 

I had noted that `SD.open(/path/to/file/, "w")` does not create the path to file if missing. I think this behavior is intuitive if you know the underlying implementation fopen, which does not create missing folders. However, I was used to ESP8266 Arduino core (SDFS and SPIFFS), and the common behavior creates any missing path. 
Is this an intended behavior or a bug? Is there a parameter to force automatic creation of missing path?

IMHO, create missing path is very intuitive and more suitable for Arduino users.

Villaquiranm, 2021-08-31:
Hello @me-no-dev this is marked as closed, but it is still in the *To-Do (Seeking Contributors)* list.

Do you want someone working on this or this is actually the expected behavior ?

me-no-dev, 2021-09-01:
@Villaquiranm seems that Stale BOT closed it. Surely needs work though :)

Villaquiranm, 2021-09-01:
Great @me-no-dev thanks a lot for your response, I would like to start working on this (maybe I'll need some mentoring). For the moment I will try to reproduce, understand the code on the library, and compare it to the ` ESP8266 Arduino core` that the author mentioned.

Just to check 
_desired behavior_: to create all the folders on the path if there are not already present ?

P-R-O-C-H-Y, 2021-09-13:
Hello @Villaquiranm and @fabianoriccardi,
in the official Arduino reference of the SD library there is this description of SD.Open().

_"Opens a file on the SD card. If the file is opened for writing, it will be created if it doesn't already exist (but the directory containing it must already exist)."_

It will create a file if not exist but the directory must already exist.
I checked it and it is acting exactly same as written in Arduino reference and we should stay with the same behavior as the Arduino lib.

Link: [https://www.arduino.cc/en/Reference/SDopen](url)

VojtechBartoska, 2021-09-29:
@fabianoriccardi PTAL, Is the answer above sufficient?

closed 2021-10-01 by me-no-dev