espressif/arduino-esp32 #5021: Self signed TLS certificate issue for MQTT broker.

norbertg1, 2021-04-06:
### Hardware:
Board:							?ESP32 Wemos Lollin32?
Core Installation version:			?v3.3.5-1-g85c43024c?
IDE name:						?Platform.io?
Computer OS:                                         ?Ubuntu?

### Description:
I would like to make a TLS secured connection between MQTT Mosquitto broker and ESP32 board (Wemos Lollin32) with self-signed certificates, but it is not going. See the error message below the code.

For generating certificates I used the following commands:
```
openssl req -new -x509 -days 365 -extensions v3_ca -keyout ca.key -out ca.crt

openssl genrsa -out mosquitto.key 2048
openssl req -out mosquitto.csr -key mosquitto.key -new
openssl x509 -req -in mosquitto.csr -CA ca.crt -CAkey ca.key -CAcreateserial -out mosquitto.crt -days 365

openssl genrsa -out esp.key 2048
openssl req -out esp.csr -key esp.key -new
openssl x509 -req -in esp.csr -CA ca.crt -CAkey ca.key -CAcreateserial -out esp.crt -days 365
```

I tried to connect with [MQTTfx](https://mqttfx.jensd.de/) to my mosquito server with the same certificates, it works well.

### Sketch:
```
#include <PubSubClient.h>
#include "WiFiClientSecure.h"

const char* CA_cert = \
"-----BEGIN CERTIFICATE-----\n" \
"################################################################\n" \
"################################################################\n" \
"################################################################\n" \
"-----END CERTIFICATE-----";

const char* ESP_CA_cert = \
"-----BEGIN CERTIFICATE-----\n" \
"################################################################\n" \
"################################################################\n" \
"################################################################\n" \
"-----END CERTIFICATE-----";

const char* ESP_RSA_key= \
"-----BEGIN RSA PRIVATE KEY-----\n" \
"################################################################\n" \
"################################################################\n" \
"################################################################\n" \
"-----END RSA PRIVATE KEY-----";

const char* ssid     = "wifi";     // your network SSID (name of wifi network)
const char* password = "password"; // your network password

const char* server = "IP adress";
const char* mqtt_user = "user";
const char* mqtt_pass = "secret";

WiFiClientSecure client;
PubSubClient mqtt_client(client); 

void setup() {
  Serial.begin(115200);
  delay(100);

  Serial.print("ESP.getSdkVersion():");
  Serial.println(ESP.getSdkVersion());

  Serial.print("Attempting to connect to SSID: ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);

  // attempt to connect to Wifi network:
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    // wait 1 second for re-trying
    delay(1000);
  }

  Serial.print("Connected to ");
  Serial.println(ssid);

  client.setCACert(CA_cert);
  client.setCertificate(ESP_CA_cert); // for client verification
  client.setPrivateKey(ESP_RSA_key);	// for client verification
  mqtt_client.setServer(server, 8883);
  
}

void loop() {
  Serial.println("\nStarting connection to server...");
  
  if (mqtt_client.connect("ESP32_client", mqtt_user , mqtt_pass)) {
    Serial.println("connected, mqtt_client state:");
    Serial.print(mqtt_client.state());
  }
  else {
    Serial.println("Connected failed!  mqtt_client state:");
    Serial.print(mqtt_client.state());
  }
  delay(10000);
}

```

### Debug Messages:
```
[E][ssl_client.cpp:36] _handle_error(): [start_ssl_client():216]: (-9984) X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
[E][WiFiClientSecure.cpp:133] connect(): start_ssl_client: -9984
```

The only solution I found is using old WiFiClientSecure.h files from this project:
[https://github.com/debsahu/ESP_MQTT_Secure/tree/master/ESP32_MQTT_SSL/Arduino/ESP32_PubSubClient_SSL](https://github.com/debsahu/ESP_MQTT_Secure/tree/master/ESP32_MQTT_SSL/Arduino/ESP32_PubSubClient_SSL
)
Probably I am not the only one suffering from this bug:
[https://stackoverflow.com/questions/65218865/esp32-mqtt-with-tls](https://stackoverflow.com/questions/65218865/esp32-mqtt-with-tls)

rin67630, 2021-04-10:
I have a similar issue: struggling since week with my ESP compiler which binaries do not work:
I have now used a pristine computer, windows 10 reset to defaults, updated.
-freshly installed the latest Arduino IDE 1.8.13
-entered the board address: https://dl.espressif.com/dl/package_esp32_index.json
-installed the boards ESP32 and ESP8266 from the board manager.
-installed thinger.io from the library manager.
Nothing else!
-Called the thinger.io ESP32 example from the library.
-Entered my credentials that work with an ESP8266
-Compiled without error (verbose, all warnings on), loaded the code on a fresh Wemos32 board.
Upon running, I get exactly these diagnostic messages: the console device never connects.
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:419] _eventCallback(): STA IP: 192.168.188.76, MASK: 255.255.255.0, GW: 192.168.188.1
[V][ssl_client.cpp:265] stop_ssl_socket(): Cleaning SSL connection.
[V][ssl_client.cpp:59] start_ssl_client(): Free internal heap before TLS 281100
[E][WiFiClientSecure.cpp:133] connect(): start_ssl_client: -1
[V][ssl_client.cpp:265] stop_ssl_socket(): Cleaning SSL connection.
[V][ssl_client.cpp:265] stop_ssl_socket(): Cleaning SSL connection.
[V][ssl_client.cpp:59] start_ssl_client(): Free internal heap before TLS 281104
[E][WiFiClientSecure.cpp:133] connect(): start_ssl_client: -1
[V][ssl_client.cpp:265] stop_ssl_socket(): Cleaning SSL connection.

the last repeats every 2-3 seconds…

rin67630, 2021-04-10:
Can it be that a certificate expired?

norbertg1, 2021-04-10:
After several working hours, I realized that the problem was in the generating of the certificates, exactly in the CN field. If you issue self-signed certificates the CN field in the root CA.crt can be anything or maybe even you can leave it blank because you don't have Trusted certificate authority. 
But much more important is the CN field of the certificate for the Mosquito broker (mosquitto.crt, .key and .csr) in which you __must provide your (Mosquitto) server DNS or IP adress:__ 
`
openssl req -out mosquitto.csr -key mosquitto.key -new -subj '/CN=Mosquitto_borker_adress
`
Because the client (in this case the ESP32) checking if this adress really belongs to the server (in this case the Mosquitto broker). Probably with the mentioned old WiFiClientSecure library in my first comment, this checking is missing, also I guess same is the situation with the ESP8266.

I put here my certificate generator commands for openSSL, I hope its can be helpful for similar problems in the future:
```
###########################################################################################################################################################################################################################################
#Certificate generator for TLS encryption
###########################################################################################################################################################################################################################################
openssl req -new -x509 -days 365 -extensions v3_ca -keyout ca.key -out ca.crt -passout pass:1234 -subj '/CN=TrustedCA.net'
#If you generating self-signed certificates the CN can be anything

openssl genrsa -out mosquitto.key 2048
openssl req -out mosquitto.csr -key mosquitto.key -new -subj '/CN=Mosquitto_borker_adress'
openssl x509 -req -in mosquitto.csr -CA ca.crt -CAkey ca.key -CAcreateserial -out mosquitto.crt -days 365 -passin pass:1234
#Mostly the client verifies the adress of the mosquitto server, so its necessary to set the CN to the correct adress (eg. yourserver.com)!!!

###########################################################################################################################################################################################################################################
#These certificates are only needed if the mosquitto broker require certificate for client autentithication (require_certificate is set to true in mosquitto config)
###########################################################################################################################################################################################################################################
openssl genrsa -out esp.key 2048
openssl req -out esp.csr -key esp.key -new -subj '/CN=localhost'
openssl x509 -req -in esp.csr -CA ca.crt -CAkey ca.key -CAcreateserial -out esp.crt -days 365 -passin pass:1234
#If the server (mosquitto) identifies the clients based on CN key, its necessary to set it to the correct value, else it can be blank. See the Mosquitto config.
```
The C code for the ESP32:
```
WiFiClientSecure client;

client.setCACert(root_CA_cert);
client.setCertificate(ESP_CA_cert); // for client verification if the require_certificate is true setting in the mosquitto broker config
client.setPrivateKey(ESP_RSA_key); // for client verification if the require_certificate is true setting in the mosquitto broker config

```

Also the Mosquito broker you have to start with configuration:
```
listener 8883
cafile path/to/ca.crt
keyfile path/to/mosquitto.key
certfile path/to/mosquitto.crt
require_certificate true or false
```

TheKaese, 2021-04-30:
@norbertg1, thanks for doing all of this work.. Can you talk a bit more about what you did for debugging on this? I have been struggling to get this to work for weeks now. Unfortunately your method of generating the certificates produced no change in behavior on my end, still getting the "Certificate verification failed, e.g. CRL, CA or signature check failed" errors.

rin67630, 2021-05-10:
Anyhow: i have a found very simple workaround on the Arduino IDE:
I have reverted to version 1.0.4 instead of 1.0.6.

mrdc, 2021-05-26:
@norbertg1 

Hi, have you tried to use IP address in CN? For me it fails with the same error:
`[start_ssl_client():216]: (-9984) X509 - Certificate verification failed, e.g. CRL, CA or signature check failed`
With a name it works. The same cert works fine with multiple MQTT clients.
Looks like it can be the issue with WiFiClientSecure...

norbertg1, 2021-05-26:
If you used an IP address in CN you have to use this IP address (and not DNS) also in ESP code.

Or maybe I am wrong.

norbertg1, 2021-05-26:
> > If you used an IP address in CN you have to use this IP address (and not DNS) also in ESP code.
> > Or maybe I am wrong.
> 
> For sure, I have it in both places: in `server.crt` and in the code. For some reason, it fails when `CN=192.168.1.5`, for example. For other clients: desktop - no issues.

Paste here the ESP32 code please, with the IP address.

mrdc, 2021-05-26:
> Paste here the ESP32 code please, with the IP address.

It's close to your code, the only difference is here:

mrdc, 2021-05-26:
I have found roots: it's related to subject alt name - removing it solves the issues with IP in CN.

closed 2021-08-21 by stale[bot]

tsctrl, 2021-12-25:
i have the same issue, but my case is to connect to node.js server with https enabled and the node requesting for a cert. tried the above regenerating and importing the cert in esp without success with error as below:

[ 73564][E][ssl_client.cpp:36] _handle_error(): [start_ssl_client():210]: (-9984) X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
[ 73624][E][WiFiClientSecure.cpp:132] connect(): start_ssl_client: -9984
[ 73719][W][HTTPClient.cpp:1400] returnError(): error(-1): connection refused