espressif/arduino-esp32 #5022: SD card or Ethernet not working when we use BLE and WIFI and SD card and Ethernet all together.

akshar001, 2021-04-07:
I have esp32-pico-d4 running with LAN8720 ic and GPIO0 is used as a clock source for IC.
We are using internal clock from ESP32 for PHY IC .

The problem we are having is when we try to call WiFi, BLE, SD and Ethernet All togheter, SD card is not being mounted.
If we comment out either BLE or WiFI initialisation it's being mounted.

This is the example code to try it out.

`#include <WiFiClient.h>
#include <ETH.h>
#include <Arduino.h>
#include "SD.h"
#include "SPI.h"
#include "FS.h"
#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>

#ifdef ETH_CLK_MODE
#undef ETH_CLK_MODE
#endif
#define ETH_CLK_MODE    ETH_CLOCK_GPIO0_OUT
// Pin# of the enable signal for the external crystal oscillator (-1 to disable for internal APLL source)
#define ETH_POWER_PIN   -1
// Type of the Ethernet PHY (LAN8720 or TLK110)
#define ETH_TYPE        ETH_PHY_LAN8720
// I²C-address of Ethernet PHY (0 or 1 for LAN8720, 31 for TLK110)
#define ETH_ADDR        0
// Pin# of the I²C clock signal for the Ethernet PHY
#define ETH_MDC_PIN     18
// Pin# of the I²C IO signal for the Ethernet PHY
#define ETH_MDIO_PIN    5

void WiFiEvent(WiFiEvent_t event) {
  switch (event) {
    case SYSTEM_EVENT_ETH_START:
      Serial.println("ETH Started");
      //set eth hostname here
      ETH.setHostname("esp32-ethernet");
      break;
    case SYSTEM_EVENT_ETH_CONNECTED:
      Serial.println("ETH Connected");
      break;
    case SYSTEM_EVENT_ETH_GOT_IP:
      Serial.print("ETH MAC: ");
      Serial.print(ETH.macAddress());
      Serial.print(", IPv4: ");
      Serial.print(ETH.localIP());
      if (ETH.fullDuplex()) {
        Serial.print(", FULL_DUPLEX");
      }
      Serial.print(", ");
      Serial.print(ETH.linkSpeed());
      Serial.println("Mbps");

      break;
    case SYSTEM_EVENT_ETH_DISCONNECTED:
      Serial.println("ETH Disconnected");

      break;
    case SYSTEM_EVENT_ETH_STOP:
      Serial.println("ETH Stopped");

      break;
    default:
      break;
  }
}

void setup() {
  Serial.begin(115200);

  // If you comment out either mode card mount is being failed.

  WiFi.mode(WIFI_AP_STA); 

  BLEDevice::init("ESP32_INIT");
  Serial.println("Waiting for 5 seconds so you can read this info ^^^");
  Serial.println(ESP.getFreeHeap());
  ETH.begin(ETH_ADDR, ETH_POWER_PIN, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_TYPE, ETH_CLK_MODE);
  WiFi.onEvent(WiFiEvent);
  delay(5000);   
  SPI.begin(14,12,13);
  if(!SD.begin(15))
  {
     Serial.println("Card Mount Failed");
  }else{
    Serial.println("Card Mount SUCCESS");
  }               

}

void loop() {
  delay(1000);
  // put your main code here, to run repeatedly:
}`

akshar001, 2021-04-08:
@me-no-dev

akshar001, 2021-04-13:
I see no update about this?

akshar001, 2021-06-22:
I still see no update! Waiting for an answer. It's been 2 months already

me-no-dev, 2021-06-24:
I can not replicate this issue. My board is slightly different, but I ran the same code. Card mounts fine
```cpp
#include <Arduino.h>
#include <WiFiClient.h>
#include <ETH.h>
#include "SD.h"
#include "SPI.h"
#include "FS.h"
#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>

#define ETH_CLK_MODE    ETH_CLOCK_GPIO0_IN
#define ETH_POWER_PIN   -1
#define ETH_TYPE        ETH_PHY_LAN8720
#define ETH_ADDR        0
#define ETH_MDC_PIN     23
#define ETH_MDIO_PIN    18

void setup() {
  Serial.begin(115200);
  delay(1000);
  WiFi.mode(WIFI_AP_STA); 
  BLEDevice::init("ESP32_INIT");
  Serial.println("Waiting for 5 seconds so you can read this info ^^^");
  ETH.begin(ETH_ADDR, ETH_POWER_PIN, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_TYPE, ETH_CLK_MODE);
  delay(5000);
  Serial.println(ESP.getFreeHeap());
  SPI.begin(14,2,15);
  if(!SD.begin(13)){
     Serial.println("Card Mount Failed");
  } else {
    Serial.println("Card Mount SUCCESS");
  }               
}

void loop() {
  delay(1000);
}
```

```
[  1145][V][WiFiGeneric.cpp:273] _arduino_event_cb(): STA Started
[  1145][D][WiFiGeneric.cpp:804] _eventCallback(): Arduino Event: 2 - STA_START
[  1146][V][WiFiGeneric.cpp:315] _arduino_event_cb(): AP Started
[  1463][D][WiFiGeneric.cpp:804] _eventCallback(): Arduino Event: 10 - AP_START
Waiting for 5 seconds so you can read this info ^^^
[  5780][V][WiFiGeneric.cpp:353] _arduino_event_cb(): Ethernet Started
[  5780][D][WiFiGeneric.cpp:804] _eventCallback(): Arduino Event: 18 - ETH_START
61687
Card Mount SUCCESS
```

akshar001, 2021-08-30:
@me-no-dev There has been no update about this issue since last 3 months!

VojtechBartoska, 2022-02-15:
@akshar001 Hi, can you try to test this issue on the latest version of Arduino? 

There was no discussion as we weren't able to reproduce the issue.

Thanks for your contribution.

VojtechBartoska, 2022-03-03:
@akshar001 Pinging you once again, we weren't able to reproduce this. If you are able to provide more info, please do so. Thanks.

VojtechBartoska, 2022-03-23:
Closing as Expired.

closed 2022-03-23 by VojtechBartoska