espressif/arduino-esp32 #5024: Bluedroid crashing after shutting it down before deep sleep

cyberman54, 2021-04-08:
### Hardware:
Board:							TTGO T-Beam v.1.1
Core Installation version:			 3.10006.210326 (1.0.6)
IDE name:						framework-arduinoespressif32

### Description:
Bluetooth stack crashes sometimes / in certain situations, while shutting it down before device enters deep sleep. (see code below). In an experiment with ~200 shutdowns (caused by entering deepsleep) this happened ~10 times.

### Sketch:  
```
ESP_LOGI(TAG, "Shutting down bluetooth scanner ...");
ESP_ERROR_CHECK(esp_ble_gap_register_callback(NULL));
ESP_ERROR_CHECK(esp_bluedroid_disable());
ESP_ERROR_CHECK(esp_bluedroid_deinit());
btStop(); 
ESP_ERROR_CHECK(esp_coex_preference_set(ESP_COEX_PREFER_WIFI));
ESP_LOGI(TAG, "Bluetooth scanner stopped");
[...]
esp_deep_sleep_start();
```

### Debug Messages:
```
#0  0x400953dc:0x3ffe9890 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #1  0x40095655:0x3ffe98b0 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #2  0x40096385:0x3ffe98d0 in xQueueGenericReceive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:2520
  #3  0x40135275:0x3ffe9910 in osi_sem_take at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/common/osi/semaphore.c:60
  #4  0x40163c9a:0x3ffe9930 in fixed_queue_enqueue at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/common/osi/fixed_queue.c:137
  #5  0x4017986d:0x3ffe9950 in transmit_command at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/hci/hci_layer.c:273
  #6  0x40149e0d:0x3ffe9970 in btu_hcif_send_cmd at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/btu/btu_hcif.c:418
  #7  0x40152746:0x3ffe9990 in btsnd_hcic_ble_update_adv_report_flow_control at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/hcic/hciblecmds.c:1020
  #8  0x4019a609:0x3ffe99b0 in hci_update_adv_report_flow_control at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/hci/hci_hal_h4.c:252
  #9  0x4019a6c0:0x3ffe99d0 in hci_hal_h4_hdl_rx_packet at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/hci/hci_hal_h4.c:319
  #10 0x4019a6f0:0x3ffe99f0 in event_uart_has_bytes at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/hci/hci_hal_h4.c:338
  #11 0x40163f93:0x3ffe9a10 in fixed_queue_process at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/common/osi/fixed_queue.c:289
  #12 0x4019a411:0x3ffe9a30 in hci_hal_h4_rx_handler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/hci/hci_hal_h4.c:177
  #13 0x400966be:0x3ffe9a60 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

FlorianLudwig, 2021-06-22:
@VojtechBartoska How do we get someone to look into this so it does not get marked as stale?

chegewara, 2021-06-22:
Did you try to remove this line of code (just blind shot)?
`ESP_ERROR_CHECK(esp_ble_gap_register_callback(NULL));`

closed 2021-09-06 by stale[bot]