espressif/arduino-esp32 #5025: Cannot compile developmental version if stable is installed

fabiuz7, 2021-04-08:
Hi,
I was used to having the latest version of esp32-core through Arduino Board Manager, and the developmental version in Arduino sketchbook to test new features or do experiment. I was able to switch between them seamlessly in Arduino IDE. However, checking out the latest commit, the developmental version fails to compile since Arduino IDE searches the toolchain in the path of **stable** esp32-core, which is not more compatible with the latest commit 371f382db7dd36c470bb2669b222adf0a497600d. If I delete the stable version, I can compile the developmental version.

### Hardware:
Core Installation version:			1.0.6
IDE name:						Arduino IDE
Computer OS:                                        Windows 10

Log when stable is installed:

> c:/users/user/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find crt1-sim.o: No such file or directory
> c:/users/user/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find _vectors.o: No such file or directory
> collect2.exe: error: ld returned 1 exit status
> exit status 1
> Errore durante la compilazione per la scheda WEMOS LOLIN32.

me-no-dev, 2021-04-08:
what you have been doing has always been a bad idea :) tools are not necessarily the same between versions and Arduino will always fall-back on the one from the Board Manager URL

closed 2021-04-08 by me-no-dev

fabiuz7, 2021-04-08:
> Arduino will always fall-back on the one from the Board Manager URL

Didn't know this. I was thinking the different cores were well-isolated like in "sandboxes". Maybe a note in the installation guide would be helpful.

Thanks!

RudyFiero, 2021-04-09:
I make a portable install of Arduino for different versions.

fabiuz7, 2021-04-09:
@RudyFiero link?

lbernstone, 2021-05-28:
This can also be fixed by adding a platform.local.txt in your sketchbook folder (eg, \<HOME\>/Arduino/hardware/espressif/esp32) with the following:
`compiler.path={runtime.platform.path}/tools/{build.tarch}-{build.target}-elf/bin/`