espressif/arduino-esp32 #5026: serial.begin(baudrate) resets the full configuration of serial

fabiuz7, 2021-04-08:
Hi,
Looking at HardwareSerial.h:

    serial.begin(9600);

This calls does not only set the baud rates, but it resets the full configuration to the default values (even rx and tx pins). Is it an intended behavior or is a bug? 

I think that if I do not specify further parameters, these shouldn't be changed.

I tested this behavior on latest commit 371f382db7dd36c470bb2669b222adf0a497600d

lbernstone, 2021-04-08:
The function has default parameters.  So, if you don't set them, it will use the defaults.  If you want to call begin again, you must supply those custom paramteres again.  If you just want to change frequency, use updateBaudRate.

chegewara, 2021-04-08:
Why dont use `Serial.setBauderate()` instead?

fabiuz7, 2021-04-08:
I have read the header file, I know I can do everything.
The problem is that I'm using a library which call begin(baudrate) and it resets also the rx and tx pins. This is a side unintended effect since in other Arduino cores there is no way to change serial pins if not in Class constructor.

Nothing wrong with this implementation, but in this case portability among cores seems not trivial.

closed 2021-07-11 by stale[bot]