espressif/arduino-esp32 #5027: Connection to Enterprise WPA2 network fails

dfmcwhir, 2021-04-08:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.06
IDE name:						Arduino IDE
Flash Frequency:					80mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                       Windows 10

### Description:
Describe your problem here
I'm trying to connect to a network using WPA2 Enterprise authentication (PEAP+MsChapv2) using the arduino IDE. I have been trying to do this off and on for ~1.5years. Recently I found that I can connect to my network using the Espressif IDF and Visual Studio Code (using their example code), so I'm pretty sure this something with the Arduino core. When I saw that core version had been updated to 1.06 I tried again and the results are below.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

#include <WiFi.h> //Wifi library
#include "esp_wpa2.h" //wpa2 library for connections to Enterprise networks
#define EAP_IDENTITY "_domain\\username_" //This is the format of domain and user name that works with the IDF and Visual Studio Code on my network
#define EAP_PASSWORD "_password_" //your Eduroam password
const char* ssid = "_MY_NETWORK_SSID_"; // Eduroam SSID
const char* host = "_companywebsite.com_"; //external server domain for HTTP connection after authentification
int counter = 0;
void setup() {
  Serial.begin(115200);
  delay(10);
  Serial.println();
  Serial.print("Connecting to network: ");
  Serial.println(ssid);
  WiFi.disconnect(true);  //disconnect form wifi to set new wifi connection
  WiFi.mode(WIFI_STA); //init wifi mode
  esp_wifi_sta_wpa2_ent_set_identity((uint8_t *)EAP_IDENTITY, strlen(EAP_IDENTITY)); //provide identity
  esp_wifi_sta_wpa2_ent_set_username((uint8_t *)EAP_IDENTITY, strlen(EAP_IDENTITY)); //provide username --> identity and username is same
  esp_wifi_sta_wpa2_ent_set_password((uint8_t *)EAP_PASSWORD, strlen(EAP_PASSWORD)); //provide password
  esp_wpa2_config_t config = WPA2_CONFIG_INIT_DEFAULT(); //set config settings to default
  esp_wifi_sta_wpa2_ent_enable(&config); //set config settings to enable function
  WiFi.begin(ssid); //connect to wifi
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
    counter++;
    if(counter>=240){ //updated to 2 minute time out.
        Serial.println("IP address set: "); 
        Serial.println(WiFi.localIP()); //print LAN IP
        ESP.restart();
    }
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address set: "); 
  Serial.println(WiFi.localIP()); //print LAN IP
}
void loop() {
  if (WiFi.status() == WL_CONNECTED) { //if we are connected to Eduroam network
    counter = 0; //reset counter
    Serial.println("Wifi is still connected with IP: "); 
    Serial.println(WiFi.localIP());   //inform user about his IP address
  }else if (WiFi.status() != WL_CONNECTED) { //if we lost connection, retry
    WiFi.begin(ssid);      
  }
  while (WiFi.status() != WL_CONNECTED) { //during lost connection, print dots
    delay(500);
    Serial.print(".");
    counter++;
    if(counter>=30){ //30 seconds timeout - reset board
    ESP.restart();
    }
  }
  Serial.print("Connecting to website: ");
  Serial.println(host);
  WiFiClient client;
  if (client.connect(host, 80)) {
    String url = "/rele/rele1.txt";
    client.print(String("GET ") + url + " HTTP/1.1\r\n" + "Host: " + host + "\r\n" + "User-Agent: ESP32\r\n" + "Connection: close\r\n\r\n");

    while (client.connected()) {
      String line = client.readStringUntil('\n');
      if (line == "\r") {
        break;
      }
    }
    String line = client.readStringUntil('\n');
   Serial.println(line);
  }else{
      Serial.println("Connection unsucessful");
    }  
}

```

### Debug Messages:
```

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4

Connecting to network: _MY_NETWORK_SSID_
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
..........................................................................................................................[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:391] _eventCallback(): Reason: 204 - HANDSHAKE_TIMEOUT
........[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:391] _eventCallback(): Reason: 4 - ASSOC_EXPIRE
..............................................................................................................IP address set: 
0.0.0.0
ets Jun  8 2016 00:22:57

```

thezenox, 2021-05-06:
I had it working for around 1 year with this code:

```
#include <WiFi.h>
#include <WiFiClientSecure.h>
#include "esp_wpa2.h"
#include "esp_wifi.h"
...
    WiFi.disconnect(true);  //disconnect form wifi to set new wifi connection
    WiFi.mode(WIFI_STA); //init wifi mode
    esp_wifi_sta_wpa2_ent_set_identity((uint8_t *)EAP_IDENTITY, strlen(EAP_IDENTITY)); //provide identity
    esp_wifi_sta_wpa2_ent_set_username((uint8_t *)EAP_IDENTITY, strlen(EAP_IDENTITY)); //provide username --> identity and username is same
    esp_wifi_sta_wpa2_ent_set_password((uint8_t *)EAP_PASSWORD, strlen(EAP_PASSWORD)); //provide password
    esp_wpa2_config_t config = WPA2_CONFIG_INIT_DEFAULT(); //set config settings to default
    esp_wifi_sta_wpa2_ent_enable(&config); //set config settings to enable function
    WiFi.begin("eduroam"); //connect to wifi
```

but with it stopped working with #4996 
changing to:

```
    esp_wifi_sta_wpa2_ent_set_identity((uint8_t *)EAP_IDENTITY, strlen(EAP_IDENTITY)); //provide identity
    esp_wifi_sta_wpa2_ent_set_username((uint8_t *)EAP_IDENTITY, strlen(EAP_IDENTITY)); //provide username --> identity and username is same
    esp_wifi_sta_wpa2_ent_set_password((uint8_t *)EAP_PASSWORD, strlen(EAP_PASSWORD)); //provide password
    esp_wifi_sta_wpa2_ent_enable();
    WiFi.begin("eduroam"); //connect to wifi
```

results in a successful connection, but only after the third try. (each dot is 200ms delay)

```
[   841][D][esp32-hal-psram.c:84] psramInit(): PSRAM enabled
[   989][V][WiFiGeneric.cpp:273] _arduino_event_cb(): STA Started
[   990][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[   990][D][WiFiGeneric.cpp:795] _eventCallback(): Arduino Event: 2 - STA_START
...........
[  3061][V][WiFiGeneric.cpp:290] _arduino_event_cb(): STA Disconnected: SSID: eduroam, BSSID: 00:00:00:00:00:00, Reason: 201
[  3062][D][WiFiGeneric.cpp:795] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[  3069][W][WiFiGeneric.cpp:816] _eventCallback(): Reason: 201 - NO_AP_FOUND
[  3077][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
.........
[  5016][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
...........
[  7079][V][WiFiGeneric.cpp:290] _arduino_event_cb(): STA Disconnected: SSID: eduroam, BSSID: 80:8d:XX:XX:XX:XX, Reason: 203
[  7080][D][WiFiGeneric.cpp:795] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[  7087][W][WiFiGeneric.cpp:816] _eventCallback(): Reason: 203 - ASSOC_FAIL
[  7095][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
.........
[  9024][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
............
[ 11999][V][WiFiGeneric.cpp:285] _arduino_event_cb(): STA Connected: SSID: eduroam, BSSID: 80:8d:XX:XX:XX:XX, Channel: 1, Auth: WPA2_ENTERPRISE
[ 12001][D][WiFiGeneric.cpp:795] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
........
[ 13740][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[ 13741][V][WiFiGeneric.cpp:290] _arduino_event_cb(): STA Disconnected: SSID: eduroam, BSSID: 80:8d:XX:XX:XX:XX, Reason: 8
[ 13751][D][WiFiGeneric.cpp:795] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 13759][W][WiFiGeneric.cpp:816] _eventCallback(): Reason: 8 - ASSOC_LEAVE
...........
[ 15840][V][WiFiGeneric.cpp:285] _arduino_event_cb(): STA Connected: SSID: eduroam, BSSID: 80:8d:XX:XX:XX:XX, Channel: 1, Auth: WPA2_ENTERPRISE
[ 15842][D][WiFiGeneric.cpp:795] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
.........
[ 17769][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[ 17770][V][WiFiGeneric.cpp:290] _arduino_event_cb(): STA Disconnected: SSID: eduroam, BSSID: 80:8d:XX:XX:XX:XX, Reason: 8
[ 17780][D][WiFiGeneric.cpp:795] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 17788][W][WiFiGeneric.cpp:816] _eventCallback(): Reason: 8 - ASSOC_LEAVE
...........
[ 19868][V][WiFiGeneric.cpp:285] _arduino_event_cb(): STA Connected: SSID: eduroam, BSSID: 80:8d:XX:XX:XX:XX, Channel: 1, Auth: WPA2_ENTERPRISE
[ 19870][D][WiFiGeneric.cpp:795] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[ 19884][V][WiFiGeneric.cpp:295] _arduino_event_cb(): STA Got New IP:10.___.___.28
[ 19884][D][WiFiGeneric.cpp:795] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[ 19891][D][WiFiGeneric.cpp:844] _eventCallback(): STA IP: 10.__.___.28, MASK: 255.255.240.0, GW: 10.___.___.254
```

Any idea?

dfmcwhir, 2021-05-07:
I don't know, but I believe that this is a completely different issue that what I'm seeing. Looking through older issues that people have had connecting with Enterprise WPA2, some/most people have had success connecting with Eduroam networks like you are. Others have been completely unable Enterprise WPA2 networks at businesses like I me. It has been speculated that it is related to the business networks using Cisco switches.

The problem I am having, I have never been able to connect to my network throughout the ~1.5years have been trying with several different versions of the Arduino core. It looks like you have been historically able to connect to the Eduroam network and are still able to connect to it intermittently .

closed 2021-07-23 by stale[bot]