espressif/arduino-esp32 #5033: Want upgrade eclipse/esp-idf/arduino-esp32 versions

r2d2r01, 2021-04-09:
Hi, my new goal assigned is to upgrade an platform with eclipse 2018/esp-idf 3.3/arduino component and all projects inside.
I have followed/readen all existing documentations (espressig(esp-idf)/arduino/..).
A this time eclipse is 202012r with plugin esp-idf (master) / arduino-esp32 is 4.2 (as indicated in a post by lberstone) branch installed as component.
After reading posts from lberstone/me_to_dev/..  it exists always a little blocking build problem (gpio, ethernet, wifi_provisionning, etc.. ). I have found presque an answer for each one, but it's hard (actually it's  an error with wifi_provisioning/manager.h: No such file or directory)
I think to restart installation of platform. Is exist a way/doc/link to have quickly a production platform installed with eclipse/arduino-esp32 ?
Win10pro/Alien m15r3/32gb blank.
Thank's for all help
Frederic

lbernstone, 2021-04-09:
I provide docker containers with the exact framework files needed to compile from a specific branch/tag.  You can copy the files onto a system with eclipse.  Integrating with eclipse is beyond my knowledge.
https://hub.docker.com/r/lbernstone/esp32-arduino-lib-builder

r2d2r01, 2021-04-09:
Thank's lbesrstone for dock i go to upload but for somewho which like understand , how u explain it's hard to obtain same result with a procedure ? For others  soc not same problems (esp is young product,..)

lbernstone, 2021-04-09:
The source scripts are at https://github.com/lbernstone/arduino-esp32-dockerfiles/

r2d2r01, 2021-04-09:
Thank's i take a look on this this week-end

closed 2021-04-09 by r2d2r01