espressif/arduino-esp32 #5034: Crash when trying to use BLE and BT

aguaviva, 2021-04-09:
### Hardware:
Board:							TTGO T8 V1.7
Core Installation version:			ESP32 Arduino Release 1.0.6 based on ESP-IDF v3.3.5
IDE name:						Arduino 1.8.13
Flash Frequency:					80Mhz
PSRAM enabled:                                    disabled
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
I am trying to use BT to read the buttons and joystick postion of a PS3 controller and generate HID joystick commands so I can use the PS3 controller on cellphones/linux and windows.

That is why I need BLE and BT working at the same time.

### Sketch:  
(note that this is just a repro case)
```cpp
#include <BleGamepad.h> 

BleGamepad bleGamepad;

#include "esp_bt.h"
#include "esp_bt_main.h"
#include "esp_bt_device.h"

bool InitBluetooth()
{
  Serial.println("printBTAddress()");

  Serial.print("Initializing controller...");
  if (!btStart()) {                    //<---------------- crashes here, if I comment our this line the esp_bluedroid_init() call fails
    Serial.println("ERR");
    return false;
  }
  Serial.println("OK");
  
  Serial.print("Initializing bluedroid...");
  if (esp_bluedroid_init() != ESP_OK) {    
    Serial.println("ERR!");
    return false;
  }
  Serial.println("OK");

  Serial.print("Enabling bluedroid...");
  if (esp_bluedroid_enable() != ESP_OK) {
    Serial.println("ERR!");
    return false;
  }
  Serial.println("OK");

  Serial.print("Getting BT address...");
  const uint8_t* point = esp_bt_dev_get_address();
  if (point=NULL) {
    Serial.println("ERR!");
    return false;   
  }
  
  {
    char str[256];
    sprintf(str, "%02X:%02X:%02X:%02X:%02X:%02X\n", (int)point[0], (int)point[1], (int)point[2], (int)point[3], (int)point[4], (int)point[5]);
    Serial.println(str);
  }

  return true;
}

void setup() 
{
  Serial.begin(115200);
  Serial.println("Starting BLE work!");
  bleGamepad.begin();
  InitBluetooth();
}

void loop() 
{
  if(bleGamepad.isConnected()) 
  {
    Serial.println("Press buttons 5 and 32. Move all axes to max. Set DPAD (hat 1) to down right.");
    bleGamepad.press(BUTTON_5);
    bleGamepad.press(BUTTON_32);
    bleGamepad.setAxes(32767, 32767, 32767, 32767, 65535, 65535, 65535, 65535, DPAD_DOWN_RIGHT); //(can also optionally set hat2/3/4 after DPAD/hat1 as seen below)
    // All axes, sliders, hats can also be set independently. See the IndividualAxes.ino example
    delay(500);

    Serial.println("Release button 5. Move all axes to min. Set DPAD (hat 1) to centred.");
    bleGamepad.release(BUTTON_5);
    bleGamepad.setAxes(-32767, -32767, -32767, -32767, 0, 0, 0, 0, DPAD_CENTERED, HAT_CENTERED, HAT_CENTERED, HAT_CENTERED);
    delay(500);
  }
}
```

### Debug Messages:
```
entry 0x400806b4
Starting BLE work!
printBTAddress()
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40166b3f  PS      : 0x00060a30  A0      : 0x800ef9e6  A1      : 0x3ffcc100  
A2      : 0x3ffc3268  A3      : 0x3ffc3360  A4      : 0x00000000  A5      : 0x400ef664  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x80166b2e  A9      : 0x3ffcc0e0  
A10     : 0x3ffcc2b0  A11     : 0x000004f0  A12     : 0x00000400  A13     : 0x3ffbaec8  
A14     : 0x00060820  A15     : 0x00000003  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000014  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x40166b3f:0x3ffcc100 0x400ef9e3:0x3ffcc140 0x400dac48:0x3ffcc160 0x400d5fe2:0x3ffcc1a0 0x400d1d2e:0x3ffcc1d0 0x4008ff9e:0x3ffcc210

Rebooting...
```

### Decoded stack trace

```
Decoding stack results
0x400ef9e3: esp_bt_controller_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bt.c line 1434
0x400dac48: btStart at C:\Users\amdev02\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.6\cores\esp32\esp32-hal-bt.c line 41
0x400d5fe2: BLEDevice::init(std::__cxx11::basic_string  , std::allocator  >) at C:\Users\amdev02\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.6\libraries\BLE\src\BLEDevice.cpp line 339
0x400d1d2e: BleGamepad::taskServer(void*) at C:\Users\amdev02\Documents\Arduino\libraries\ESP32-BLE-Gamepad\BleGamepad.cpp line 815
0x4008ff9e: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

chegewara, 2021-04-09:
It crashes here because you already have bluetooth initialized , also BLE library is releasing bt classic memory so it cant be combined with bt classic.

https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLEDevice.cpp#L351

aguaviva, 2021-04-09:
Can it be fixed? If so, how?

aguaviva, 2021-04-11:
I got this working, awesomeness!! I will close the bug when I put the repo on line

closed 2021-04-11 by aguaviva