espressif/arduino-esp32 #5036: LAN8742 initialisation

NickB1, 2021-04-10:
### Hardware:
Core Installation version:			1.0.6
IDE name:						Arduino IDE

### Description:
I'm trying to use the LAN8742, it should be compatible with LAN8720 software.
When trying to run the ETH_LAN8720 example I get the following error:

```
Timed out waiting for PHY register 0x3 to have value 0xc0f0(mask 0xfff0). Current value 0xc131
```

Register 3 is the identifier 2 register, the LAN8742 has identifier 0xC131 vs the LAN8720 which has 0xC0F0.
I looked for a location to change this value to be able to pass the check and do some further testing but can't seem to find it in the core.

The only check I can find in the ESP-IDF is the following:
[](https://github.com/espressif/esp-idf/blob/6e776946d01ec0d081d09000c36d23ec1d318c06/components/esp_eth/src/esp_eth_phy_lan8720.c#L429)
But not sure if this is the one that's used in the Arduino core.

lbernstone, 2021-04-10:
It is compiled into the libraries from upstream.  There is a fix [there](https://github.com/espressif/esp-idf/issues/5092), so you will either have to wait for that to get into esp-idf or compile the libraries yourself.

NickB1, 2021-04-11:
Thanks for the help!
I somehow got it to work by putting a copy of the phy_lan8720.c file in my arduino project and changing the LAN8720_PHY_ID2 define to 0xc130.
Was expecting the linker to complain about multiple definitions of the same function but it gave no error.
Any idea why this would work?

closed 2021-07-11 by stale[bot]

pablofr918, 2022-01-14:
@NickB1 Hello there!

I'm going to start a proyect using ESP32 and LAN8742, and I would like to know how you made it to get it working. I have been investigating and I have found that Espressif has added the driver esp_eth_phy_lan87xx.c so using esp-idf it is possibe to program this lan, but I would like to use the Arduino framework and it does not seem to be included in this framework, so could you tell me how you got it working? I can't finde the file you said "phy_lan8720.c".

Thank you very much!

NickB1, 2022-01-14:
I recently tested Arduino core v2.0.2 and the LAN8742 phy works by using the default LAN8720 settings.

pablofr918, 2022-01-17:
Thank you very much! I will try it

pablofr918, 2022-01-24:
@NickB1 Hello Nick. I'm testing the example on the LAN8742, but the DHCP does not give me any IP. I have tried to give it an static IP and it does, but when it tries to connect to google, fails. Did you need to modify anything?

Thank you very much.

NickB1, 2022-01-24:
Are you trying the Arduino v2.0.2 core or an older core with the modified phy_lan8720.c file?

pablofr918, 2022-01-24:
@NickB1 I thought you meant that the LAN8720 arduino example could be ran without any modifications for the LAN8742. So I have to use the modified phy_lan8720.c file, but I couldn't find where to replace it. Do you know the directory?

Thanks a lot :)

NickB1, 2022-01-24:
Sorry, I meant that the LAN8742 works on Arduino v2.0.2 core by just using the LAN8720 default settings.
On the older v1.0.6 core you have to add the modified phy_lan8720.c file to your project directory.

pablofr918, 2022-01-25:
Okay, thank you very much! I'm trying the 2.0.2 core with the LAN8720 settings, but running the code, it prints the ETH Started, ETH connected, but after that, it does not take any IP from DHCP. I guess it can be a bug in the hardware because this code should work properly within the LAN8742.

I'll keep updating. :)

pablofr918, 2022-02-04:
@NickB1 Hello Nick, Could you share your schemmatic in order to check possible errors in my connections between ESP and LAN8742?

Thanks so much! :)

NickB1, 2022-02-09:
I have added my email to my account, you may always contact me with your schematic for a review.

pablofr918, 2022-02-11:
Hi @NickB1. I contacted you via email, I'm just checking if you recieved it.

NickB1, 2022-02-11:
Hi, no can't seen to find any mail from you.
Nothing to be found in spam also.

NickB1, 2022-02-15:
The schematic looks good at first glance.
Do you see activity on the rj45 leds when connected to the network?