espressif/arduino-esp32 #5038: SoftAP suddenly change back to no password protected (Arduino Release 1.0.6 based on ESP-IDF v3.3.5)

Luke-KH, 2021-04-11:
I have tried below code from ESP32 Arduino boards manager v1.0.6 library. I found that similar problem that password protected AP will disappear to become an OPEN softAP after reset. It will happen if softAP restarted, even reburn program (no change flash setting area).

It is working fine if using ESP32 Arduino boards manager v1.0.4 library.
Any guys know what is it going on?

```
#include <WiFi.h>
#include <WiFiClient.h>
#include <WiFiAP.h>

#define LED_BUILTIN 2   // Set the GPIO pin where you connected your test LED or comment this line out if your dev board has a built-in LED

// Set these to your desired credentials.
const char *ssid = "yourAP";
const char *password = "yourPassword";
bool wifi_onoff_toggle = true;
WiFiServer server(80);

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);

  Serial.begin(115200);
  Serial.println();
  Serial.println("Configuring access point...");

  // You can remove the password parameter if you want the AP to be open.
  WiFi.softAP(ssid, password);
  IPAddress myIP = WiFi.softAPIP();
  Serial.print("AP IP address: ");
  Serial.println(myIP);
  server.begin();

  Serial.println("Server started");
}

void loop() {
  WiFiClient client = server.available();   // listen for incoming clients

  if (client) {                             // if you get a client,
    Serial.println("New Client.");           // print a message out the serial port
    String currentLine = "";                // make a String to hold incoming data from the client
    while (client.connected()) {            // loop while the client's connected
      if (client.available()) {             // if there's bytes to read from the client,
        char c = client.read();             // read a byte, then
        Serial.write(c);                    // print it out the serial monitor
        if (c == '\n') {                    // if the byte is a newline character

          // if the current line is blank, you got two newline characters in a row.
          // that's the end of the client HTTP request, so send a response:
          if (currentLine.length() == 0) {
            // HTTP headers always start with a response code (e.g. HTTP/1.1 200 OK)
            // and a content-type so the client knows what's coming, then a blank line:
            client.println("HTTP/1.1 200 OK");
            client.println("Content-type:text/html");
            client.println();

            // the content of the HTTP response follows the header:
            client.print("Click <a href=\"/H\">here</a> to turn ON the LED.<br>");
            client.print("Click <a href=\"/L\">here</a> to turn OFF the LED.<br>");
            client.print("Click <a href=\"/T\">here</a> to turn OFF the WiFi in 10 second & resume .<br>");
            // The HTTP response ends with another blank line:
            client.println();
            // break out of the while loop:
            break;
          } else {    // if you got a newline, then clear currentLine:
            currentLine = "";
          }
        } else if (c != '\r') {  // if you got anything else but a carriage return character,
          currentLine += c;      // add it to the end of the currentLine
        }

        // Check to see if the client request was "GET /H" or "GET /L":
        if (currentLine.endsWith("GET /H")) {
          digitalWrite(LED_BUILTIN, HIGH);               // GET /H turns the LED on
        }
        if (currentLine.endsWith("GET /L")) {
          digitalWrite(LED_BUILTIN, LOW);                // GET /L turns the LED off
        }
        if (currentLine.endsWith("GET /T")) {
            WiFi.softAPdisconnect(false);
            delay(10000);
            if(WiFi.enableAP(true)) Serial.println("WiFi.enableAP(true) == TRUE");
            else Serial.println("WiFi.enableAP(true) == false");
        }
      
      }
    }
    // close the connection:
    client.stop();
    Serial.println("Client Disconnected.");
  }
}
```

societyofrobots, 2021-04-19:
Can confirm. I'm seeing the same security issue.

```
Chipset: eps32
Compiler: Arduino IDE
Partition: min_spiffs
```

Custom board, using IotWebConf library.

prampec, 2021-04-19:
This is a huge vulnerability issue. Please fix this ASAP!

me-no-dev, 2021-04-19:
if this is true, the problem is outside Arduino. You have two options: use Arduino from git + `idf-release/v3.3` branch and see if any IDF updates fix it or move to 2.0.0 (Arduino master)

societyofrobots, 2021-04-19:
> use Arduino from git + idf-release/v3.3 branch and see if any IDF updates fix it

To verify I had the latest, I copied all from [idf-release/v3.3](https://github.com/espressif/arduino-esp32/tree/idf-release/v3.3) and put it into the Arduino core folder:
..\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6

Nope, didn't fix the issue. AP is still unsecured.

> or move to 2.0.0 (Arduino master)

Unfortunately this isn't feasible for me because multiple library conflicts, and I'm not confident the bug wasn't carried into v2.0.0.

1.0.6 isn't useable with this major security bug. I'm available to test any fixes you guys have.

lbernstone, 2021-04-19:
OP, please explain what the expected behavior is.  You are disconnecting the AP (which should clear the ssid and password), and then force enabling it without providing any credentials.  Should this return an error, since the AP can't be started without an ssid, or should it do the same thing as WiFi.softAP()?

societyofrobots, 2021-04-19:
Not the OP here, but I put something together using 2.0.0 alpha and it doesn't have the security bug like 1.0.6 does.

The problem with 1.0.6 is that you can log into AP without ever entering a password. [Rotzbua above had a great description of the issue](https://github.com/espressif/arduino-esp32/issues/5038#issuecomment-821654227).

societyofrobots, 2021-06-13:
Rotzbua, your https://github.com/espressif/arduino-esp32/commit/5b5cbb49264d3415e8c29efdf21b63266c764944 commit from a few days ago fixes this problem for me.

The `pairwise_cipher` check gave me a compile error, so I could only included the `authmode` check... solved! =)

```
C:\Users\pikachu\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi\src\WiFiAP.cpp: In function 'bool softap_config_equal(const wifi_config_t&, const wifi_config_t&)':
C:\Users\pikachu\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi\src\WiFiAP.cpp:79:15: error: 'const struct wifi_ap_config_t' has no member named 'pairwise_cipher'
     if(lhs.ap.pairwise_cipher != rhs.ap.pairwise_cipher) {
               ^
C:\Users\pikachu\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi\src\WiFiAP.cpp:79:41: error: 'const struct wifi_ap_config_t' has no member named 'pairwise_cipher'
     if(lhs.ap.pairwise_cipher != rhs.ap.pairwise_cipher) {
```

Rotzbua, 2021-06-26:
@societyofrobots  Glad I could help. The `pairwise_cipher` is new in `2.0.0`.
I created a backport. I think it would be good to release a fixed v1 until v2 is ready. What is your opinion @me-no-dev ?

closed 2021-09-19 by stale[bot]

tablatronix, 2022-03-30:
Yeah this is bizarre bug has to be in SDK, or flash, I have not gotten that far into checking mem or structs..

This should never happen.. but it is, and i have no idea how to reproduce, it just happens now and then..

[image]

tablatronix, 2022-08-19:
Anyone still seeing this issue? Not sure i have lately.

societyofrobots, 2022-08-20:
> Anyone still seeing this issue? Not sure i have lately.

While it was never fixed in the 1.x core, it was resolved in 2.0.

Himanshu21358, 2023-03-10:
Sir I am new developer now we are create a ESP8266 as server and serial port receive data and send data successfully. but I have a issue now I am change my Wi-Fi SOFTAP credentials(like SSID, PASWORD) by receive in serial monitor, so please suggest any code ?

PR #5332 (2021-07-16): [backport-v1] Fix softap_config_equal compare auth_mode/cipher
backport from v2 to v1 of fix #5121
fixes #5038
commit 08571094 (2021-07-16): [backport-v1] Fix softap_config_equal compare auth_mode/cipher (#5332)

backport from v2 to v1 of fix #5121
fixes #5038