espressif/arduino-esp32 #5039: HAL/SPI initializes from incompatible pointer type.

atlaste, 2021-04-11:
Compiling 3.0 with platformio. The error message sais it all:

```
Compiling .pio\build\release\FrameworkArduino\esp32-hal-time.c.o
C:\Users\stefan\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-spi.c: In function 'spiTransferBytesNL':
Compiling .pio\build\release\FrameworkArduino\esp32-hal-timer.c.o
C:\Users\stefan\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-spi.c:922:39: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
Compiling .pio\build\release\FrameworkArduino\esp32-hal-touch.c.o
Compiling .pio\build\release\FrameworkArduino\esp32-hal-uart.c.o
                 uint8_t * last_out8 = &result[c_longs-1];
                                       ^
Compiling .pio\build\release\FrameworkArduino\libb64\cdecode.c.o
C:\Users\stefan\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_data8 = &last_data;
                                        ^

```

This breaks compilations that treats all warnings as errors (which is arguably the right way to compile...)

lbernstone, 2021-04-11:
Already fixed in newer code

closed 2021-04-22 by VojtechBartoska

hubbiw, 2022-04-02:
Hello Iberstone. Thanks for the answer. How can I install the newer code? Best regards Hubert

VojtechBartoska, 2022-04-04:
Hello @hubbiw, take  a look here: https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html

You need to change your link in preferences to **development release link** to see 2.0.3-RC1 version in Boards manager.