espressif/arduino-esp32 #5040: Migrate Arduino IDE 2.0

JAICHANGPARK, 2021-04-11:
In Arduino IDE 2.0, information that can be set for each board in the existing IDE is not displayed.

[image]

chansheunglong, 2021-04-13:
Does selecting ESP32 Dev Module show anything? My Arduino IDE 2.0.0 with arduino-ESP32 is showing the configuration normally.

JAICHANGPARK, 2021-04-13:
@chansheunglong 
Thanks to comment 
can you show your IDE tools tab screen??

chansheunglong, 2021-04-14:
[image]

Sure
I install arduino-esp32 via the board manager, haven't tried manual installation.

closed 2021-07-11 by stale[bot]