espressif/arduino-esp32 #5041: analogSetPinAttenuation does not work with Board v1.0.5 and v1.0.6, but it works with board 1.0.4. How can i fix this problem?

cbkaefer, 2021-04-11:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do  
Describe your system (Hardware, computer, O/S, core version, environment)  
Describe what is failing  
Show the shortest possible code that will duplicate the error  
Show the EXACT error message (it doesn't work is not enough)  
Then if someone is interested and knowledgeable you might get a answer.  All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
[ExceptionDecoder](https://github.com/me-no-dev/EspExceptionDecoder)

----------------------------- Remove above -----------------------------

### Hardware:
|||||||
|:---|---|---|---|---|---|
|<B>Board</B>|ESP32 Dev Module|node32|ttgo_lora|ESP32-S2-Saola|Custom w/ ESP32-S2-WROVER 16MB|
|<B>Version/Date</B>|1.0.4|2.0.0|0badbeef|11/jul/2017|today's master|
|<B>IDE name</B>|Arduino IDE|Atom + Platform.io|IDF component|VSCode|
|<B>Flash Frequency</B>|40Mhz|80Mhz|
|<B>PSRAM enabled</B>|yes|no|
|<B>Upload Speed</B>|115200|
|<B>Computer OS</B>|Windows 10|Mac OSX|Ubuntu|

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

Triangulix, 2021-05-22:
Looks like they completely broke the ADC API. Non-blocking reading is also gone.

ssieb, 2021-06-25:
There's a bug in https://github.com/espressif/arduino-esp32/pull/3377.

jesserockz, 2021-06-25:
Specifically a bug with 
https://github.com/espressif/arduino-esp32/blob/21947ebe76cf6380fe06aa885390b86cc477709a/cores/esp32/esp32-hal-adc.c#L132 

being called every time the adc pin is attached to, which overrides the `analogSetPinAttenuation` called from the code above

PR #5399 (2021-07-16): Fix pin attenuation being overwritten
Fixes: #5041
commit a9bd39de (2021-07-16): Fix pin attenuation being overwritten (#5399)

Fixes: #5041
closed 2021-07-16 by me-no-dev