espressif/arduino-esp32 #5043: The ESP32 processor hangs on a second call to Serial.begin(115200) when compiled for ESP32 Dev Module.

ZinggJM, 2021-04-11:
```
// SerialBeginTwice test by Jean-Marc Zingg
//
// Purpose: test issue with ESP32 compiled for ESP32 Dev Module

void setup()
{
  Serial.begin(115200);
  delay(100);
  Serial.println("going to call Serial.begin(115200); a second time. Should tell success ...");
  delay(100);
  Serial.begin(115200);
  delay(100);
  Serial.println("... success, returned from second call to Serial.begin(115200); Bye and out.");
  delay(100);
}

void loop() {}
```

I verified this on a ESP32 Dev Module clone, on a Wemos LOLIN32 Lite, and on the Waveshare e-Paper ESP32 Driver Board.

The GxEPD2 examples call Serial.begin(115200) in setup() to Serial.print("setup"), and again in display.init(115200).

Related post in Arduino Forum Displays: https://forum.arduino.cc/index.php?topic=487007.msg4952538#msg4952538

lbernstone, 2021-04-12:
Please follow the [ISSUE_TEMPLATE](https://github.com/espressif/arduino-esp32/blob/master/docs/ISSUE_TEMPLATE.md).

ZinggJM, 2021-04-12:
Never mind if you are not interested. All relevant information is there, can be verified easily.

OS: Windows 10 - not relevant
Arduino IDE: 1.8.13
package: actual
target: 3 mentioned - not relevant
Error message: none
expected behavior: see both lines in Serial Monitor
actual behavior: only first line, processor hangs.

lbernstone, 2021-04-12:
Can you please indicate what version of the code you are running?

ZinggJM, 2021-04-12:
1.0.6 - is this the version you mean?

lbernstone, 2021-04-12:
Well, there's an easy workaround.  I'll put in a PR and see if someone with JTAG can figure out what is actually causing the hang.

ZinggJM, 2021-04-12:
I just repeated the test with ESP32-Wrover on LOLIN D32 PRO:
works when compiled for LOLIN D32 PRO
fails when compiled for ESP32 Wrover Module

lbernstone, 2021-04-12:
If you set core debugging to verbose it will work.  This is a timing issue.

ZinggJM, 2021-04-12:
Ok, thank you, I will try. But how to explain this to users of my library?

ZinggJM, 2021-04-12:
Verbose didn't help, on LOLIN D32 PRO.

ZinggJM, 2021-04-12:
Thank you for the fast response. Glad the issue is verified and a solution is found.

closed 2021-07-11 by stale[bot]

ZinggJM, 2021-11-21:
This issue seems still to be present. See e.g. https://forum.arduino.cc/t/esp32-e-paper-display-upload-but-no-output/927375/3

DriekdeGadgetfreak, 2022-01-27:
Here is the fix:.
In uartBegin the output buffer is flushed:
uartFlush(uart);
Change this to 
uartFlushTxOnly(uart,false);  // also flush the input buffer:

ZinggJM, 2022-08-16:
**Maybe someone should make the old download link invalid, remove the version pointed to!**
I still get issues reported by users of this outdated version.

Jean-Marc

PR #8634 (2023-09-15): removes delay(10) from HardwareSerial::end()
## Description of Change
Back in April, 2021, a workaround PR #5047 was merged to fix #5043 by adding a `delay(10)` into `HardwareSerial::end()`
This is useless as pointed in https://github.com/espressif/arduino-esp32/discussions/8627

This PR removes such `delay()`

## Tests scenarios
Simple test with ESP32

## Related links
Closes #8627