espressif/arduino-esp32 #5046: Example does not seem to compile

mars000, 2021-04-12:
Hi
I'm trying to compile the example:
`arduino-esp32/libraries/ESP32/examples/GPIO/GPIOInterrupt/GPIOInterrupt.ino`

using PlatformIO on latest esp32 arduino build but get following errors, am I doing something wrong ?:
[image]

my` platformio.in`i file is:
[image]

felmue, 2021-04-12:
Hello @mars000 

about 7 days ago the example has been modified to work with ESP32S2 it seems. However those changes have not made it into PlatformIO yet. Or in other words, the example from github is too new for the esp32 Arduino library that comes with PlatformIO.

The two recent changes replaced `IRAM_ATTR` with `ARDUINO_ISR_ATTR`. So you could try to change them back.

Thanks
Felix

mars000, 2021-04-13:
thanks Felix ;-)

closed 2021-07-11 by stale[bot]