espressif/arduino-esp32 #5048: Task watchdog got triggered. The following tasks did not reset the watchdog in time "Error"

hicham-vv, 2021-04-12:
### Hardware:
Board:	az-delivery-devkit-v4					
?
IDE name:					 Platform.io
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
hey, i am trying to run two sketch on two core, in first time i thought that my code is wrong, but then i clear all the code and keep only the configuration of the two core and the infinity loop, 
but i got the same problem, i don't know why the watchdog got triggered.
i read some reply about adding delay functions but not work!
Any help, thank you 

### Sketch: 
```cpp
#include<Arduino.h>
TaskHandle_t Task1;
TaskHandle_t Task2;

void Task1code( void * pvParameters );
void Task2code( void * pvParameters );

void setup() {
  Serial.begin(115200);
  while(!Serial);
  enableCore1WDT();

  //create a task that will be executed in the Task1code() function, with priority 1 and executed on core 0
  xTaskCreatePinnedToCore(
                    Task1code,   /* Task function. */
                    "Task1",     /* name of task. */
                    10000,       /* Stack size of task */
                    NULL,        /* parameter of the task */
                    1,           /* priority of the task */
                    &Task1,      /* Task handle to keep track of created task */
                    0);          /* pin task to core 0 */                  
  delay(500); 

  //create a task that will be executed in the Task2code() function, with priority 1 and executed on core 1
  xTaskCreatePinnedToCore(
                    Task2code,   /* Task function. */
                    "Task2",     /* name of task. */
                    10000,       /* Stack size of task */
                    NULL,        /* parameter of the task */
                    1,           /* priority of the task */
                    &Task2,      /* Task handle to keep track of created task */
                    1);          /* pin task to core 1 */
    delay(500); 
}

void Task1code( void * pvParameters ){
  Serial.print("Task1 running on core ");
  Serial.println(xPortGetCoreID());

  for(;;){

  } 
}

void Task2code( void * pvParameters ){
  Serial.print("Task2 running on core ");
  Serial.println(xPortGetCoreID());

  for(;;){

  }
}

void loop() {

}

```

### Debug Messages:
```
E (10189) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (10189) task_wdt:  - IDLE0 (CPU 0)
E (10189) task_wdt: Tasks currently running:
E (10189) task_wdt: CPU 0: Task1
E (10189) task_wdt: CPU 1: loopTask
E (10189) task_wdt: Aborting.
abort() was called at PC 0x400df617 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x40084ea0:0x3ffbe960 0x40085115:0x3ffbe980 0x400df617:0x3ffbe9a0 0x40083cf9:0x3ffbe9c0 0x400d0c6a:0x3ffb46d0 0x40086115:0x3ffb46f0

Rebooting...
```

lbernstone, 2021-04-12:
Tasks must give up focus for multitasking to happen.  Put a delay(1) in your infinite loops.

smarq8, 2021-04-12:
You can not run task loops without any kind of delay function. There have to be any minimal time for IDLE task, if IDLE task does not run in about 4sec (not sure exact value for esp32) then wdt trigger occur.
In example below it look like delay may stall CPU but its not true, it just switch context. Its not easy for beginner to understand how to write properly tasks, if wdt trigger or anything other then you write it wrong and it's not because freertos. I recommend to study any freertos tutorials for understeding how to handle it.

```
void Task1code( void * pvParameters ){
  Serial.print("Task1 running on core ");
  Serial.println(xPortGetCoreID());

  for(;;){
    vTaskDelay(1); // at least 1 tick (in this case 1 tick==1ms)
    vTaskDelay(pdMS_TO_TICKS(numberOfMsToSleep));
  } 
}

void Task2code( void * pvParameters ){
  Serial.print("Task2 running on core ");
  Serial.println(xPortGetCoreID());
  TickType_t xTimeIncrement = pdMS_TO_TICKS(intervalMs);
  TickType_t xLastWakeTime = xTaskGetTickCount();
  for(;;){
    vTaskDelayUntil(&xLastWakeTime,xTimeIncrement);
  }
}
```

lbernstone, 2021-04-12:
Brevity == first post :smile:

hicham-vv, 2021-04-12:
Thank you guys, i will try my best

closed 2021-07-11 by stale[bot]