espressif/arduino-esp32 #5049: Priority on task not high enough for time critical uart reading

Gamelauncher, 2021-04-12:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Mac OSX

### Description:
Running a time critical task for rs485 decoding with timeout setting of 3 chars gets interrupted when performing OTA. While the bin is uploading, the timing gets off no matter what I try. Changing the interrupt flag to 3, priority to 50, core 0 or core 1, it still interferes with the RX. Is there a way to raise the priority even more?

It's no problem that after OTA the ESP32 reboots, the reboot is within a second, but the 10 to 15 seconds or so during OTA upload causes allot of RX error's.

### Sketch:
```  
  uart_driver_install(RS485, 70 * 3, 70 * 3, 10, &RS485_queue, ESP_INTR_FLAG_LEVEL3);
  uart_set_mode(RS485, UART_MODE_RS485_HALF_DUPLEX);
  uart_set_rx_timeout(RS485, 3);
   xTaskCreatePinnedToCore(RS485_event,"RS485_event",6144,NULL, 50 | portPRIVILEGE_BIT ,NULL,1);
```

chegewara, 2021-04-12:
1) MAX task priority is 24 or 25
2) if i remember correct, write to flash, like OTA, is disabling INTR for short time
3) it is good to disable other tasks to perform OTA, as you will reset device anyway after OTA finished, but not required of course

Gamelauncher, 2021-04-12:
1. I've lowered the priorities to 24 and lower to be on the safe side. 
2. I am guessing there is no way to avoid that?
3. All tasks are stopped when OTA is beginning except for the RS485

I am fine with OTA taking 15 minutes as long as it's not interfering :-)

chegewara, 2021-04-12:
Sorry, but i dont have deep to the bottom knowledge, but the thing is that write to flash takes some time and this opperation cant be interrupted by anything, including ISR.
Maybe @atanisoft @lbernstone or @me-no-dev can explain it better

There is many opened issues on esp-idf where OTA and BLE may lead to crash

chegewara, 2021-04-17:
This is from esp-idf docs:

> Please note that the ISR is disabled during flash operation by default. To keep sending transactions during flash operations, enable CONFIG_SPI_MASTER_ISR_IN_IRAM and set ESP_INTR_FLAG_IRAM in the member spi_bus_config_t::intr_flags.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_master.html#cache-miss

Gamelauncher, 2021-04-17:
Thanks for this! Not really sure how it affects uart operation though ...

My knowledge about this isn't that great either, I've been fiddling around with this for a while now and can't seem to figure out what it is with IRAM directive.

Setting this: ESP_INTR_FLAG_LEVEL3 appears to be the highest for the driver. The ESP_INTR_FLAG_IRAM is not supported on the driver itself.

I found some example where IRAM_ATTR is used on the RX event task but without changing anything else, the IRAM_ATTR seems to have no effect, the task still gets delayed during flash operations ( OTA and noticed it now too with LittleFS ). https://github.com/bw38/wiog_v3/blob/08bc86cc3db63250f9a77676ed7ca19b19936d82/gateway/main/rs232.c

I found another example where the interrupt routine gets set differently with ESP_INTR_FLAG_IRAM in the uart_isr_register(). But will have to move around some code to do a test since it's not using a queue anymore.

As for the CONFIG_SPI_MASTER_ISR_IN_IRAM, I cannot find a place where to set this using the Arduino ide ....

Gamelauncher, 2021-04-17:
I've made a test:

```#include <WiFi.h>

extern "C" {
  #include "freertos/timers.h"
  #include "freertos/queue.h"
  #include "freertos/task.h"
}

#include "driver/uart.h"

const int RS485_RX_PIN = 34;    // -> TX RS485
const int RS485_TX_PIN = 32;    // -> RX RS485

#define RS485 UART_NUM_2
#define RS485_BUF_SIZE 130
#define RS485_TIMEOUT 3.5

uint8_t rxbuf[RS485_BUF_SIZE];
uint16_t urxlen;

TaskHandle_t RS485_Process;

void ResetRXBuffer() {
  urxlen = 0;
}

void IRAM_ATTR RS485_RX_ISR(void * arg) {
  
  if(UART2.int_st.rxfifo_ovf) {
    uart_flush_input(RS485);
    uart_clear_intr_status(RS485, UART_RXFIFO_OVF_INT_CLR | UART_RXFIFO_TOUT_INT_CLR);
    Serial.print("UART receive error!\n");
    ResetRXBuffer();
    UART2.int_ena.brk_det = 0;
    return;
  }
  
  if (UART2.int_st.at_cmd_char_det) {
    uart_clear_intr_status(RS485, UART_AT_CMD_CHAR_DET_INT_CLR);
      
    while (UART2.status.rxfifo_cnt) {
      rxbuf[urxlen] = READ_PERI_REG(UART_FIFO_REG(RS485));
      urxlen++;
    }
    
    xTaskNotifyFromISR(RS485_Process, 0, eSetValueWithOverwrite, NULL);

    uart_flush_input(RS485);
  }
  
  if (UART2.int_st.rxfifo_full || UART2.int_st.brk_det) {

    while (UART2.status.rxfifo_cnt) {
      rxbuf[urxlen] = READ_PERI_REG(UART_FIFO_REG(RS485));
      urxlen++;
    }

    uart_flush_input(RS485);
    uart_clear_intr_status(RS485, UART_RXFIFO_FULL_INT_CLR | UART_BRK_DET_INT_CLR);
  }
}

void Task_RS485_Process(void * pvParameters) {
  while (1) {

    if (xTaskNotifyWait(0, 0, NULL, portMAX_DELAY) == pdFAIL) {
      ResetRXBuffer();
      continue;
    }

    Serial.print("Len: ");
    Serial.print(urxlen);
      Serial.print(" ");
    for (int i=0; i<urxlen; i++) {
      Serial.print(urxlen, HEX);
      Serial.print(" ");
    }
    Serial.println();
    
    ResetRXBuffer();

    vTaskDelay(1);
  }

}

void setup(){

  xTaskCreatePinnedToCore(Task_RS485_Process, "RS485_Process", 2048, NULL, 12, &RS485_Process, NULL);
  
  uart_config_t uart_config_rs485 = {
    .baud_rate = 9600,
    .data_bits = UART_DATA_8_BITS,
    .parity = UART_PARITY_DISABLE,
    .stop_bits = UART_STOP_BITS_1,
    .flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
    .rx_flow_ctrl_thresh = 0
  }; 

  uart_param_config(RS485, &uart_config_rs485);
  uart_set_pin(RS485, RS485_TX_PIN, RS485_RX_PIN, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);
  uart_driver_install(RS485, RS485_BUF_SIZE, 0, 0, NULL, ESP_INTR_FLAG_LEVEL3);
  uart_set_mode(RS485, UART_MODE_RS485_HALF_DUPLEX);

  uart_isr_free(RS485);
  uart_isr_register(RS485, RS485_RX_ISR, NULL, ESP_INTR_FLAG_IRAM, NULL);
  uart_enable_rx_intr(RS485);
  
//  uart_set_rx_timeout(RS485, RS485_TIMEOUT);
}

void loop(){

  vTaskDelay(1);
}
```

Unfortunately this crashes immediately with:

```Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x400842a2  PS      : 0x00050034  A0      : 0x400ea62e  A1      : 0x3ffbe7d0  
A2      : 0x00000000  A3      : 0x00000040  A4      : 0x00800000  A5      : 0x3ffbe7b0  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800898d0  A9      : 0x4008acac  
A10     : 0x00000003  A11     : 0x00060823  A12     : 0x00000001  A13     : 0x3ffba264  
A14     : 0x00000015  A15     : 0x3ffbd4e0  SAR     : 0x0000001f  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x400d340a  EPC2    : 0x00000000  EPC3    : 0x400ea62e  EPC4    : 0x400842a2

Backtrace: 0x400842a2:0x3ffbe7d0 0x400ea62b:0x3ffbc570 0x400d3407:0x3ffbc590 0x40089f3e:0x3ffbc5b0 0x40088a4d:0x3ffbc5d0```

```PC: 0x400842a2
EXCVADDR: 0x00000000

Decoding stack results
0x400ea62b: esp_pm_impl_waiti at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/pm_esp32.c line 492
0x400d3407: esp_vApplicationIdleHook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/freertos_hooks.c line 63
0x40089f3e: prvIdleTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 3382
0x40088a4d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143```

me-no-dev, 2021-04-17:
just FYI, pins 34+ are InputOnly on ESP32

Gamelauncher, 2021-04-17:
Thanks! I am using 34 as the input. I have it working just fine with the top code, only that it gets interrupted to much by OTA / Flash operations. I am doing some emulation onto the 485 and the master goes nuts when I respond out of time for more then 2 seconds. It's the only bit of code that I need to avoid to be delayed.

chegewara, 2021-04-17:
Like i said, and like mentioned in docs, when you are using SPI flash operations, which include write OTA packet to flash, the ISR is disabled for that moment, and there is solution for this.

Gamelauncher, 2021-04-17:
There are 2 files in ~/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk referencing the CONFIG_SPI_MASTER_ISR_IN_IRAM and both are auto generated:

sdkconfig contains: CONFIG_SPI_MASTER_ISR_IN_IRAM=y
include/config/sdkconfig.h contains:  #define CONFIG_SPI_MASTER_ISR_IN_IRAM 1

It's correct this is included during compile time?

The next bit it references: spi_bus_config_t::intr_flags

Could you explain where or how to set it? 
```
spi_bus_config_t spi_config
```

Searching for a function that uses it I only found spi_bus_initialize in "spi_master.h"

atanisoft, 2021-04-17:
When flash operations are run all non-IRAM ISRs will be halted, this will include UART. You will need to move to using an IRAM ISR for your UART usage and add appropriate flags to the ISR attach function. The task may also be suppressed during flash writes as well.

Gamelauncher, 2021-04-20:
I've been trying different approaches now but can't seem to get it to work like I need ...

I am getting data through to the task, but the .rx_timeout_thresh = 3 in the isr config doesn't seem to work. Data gets cut off, but at the wrong section and not where the actual gap is and there is an additional byte added at the beginning of each.

Any suggestions on what I am missing?

Output capture:
```
0FDA8
0FDA81111A105DBA
0FD41AF
0FD41AFA1011F80FF000B4
0FD4907244
0FD411F410C0
0FD106C0
0FD11FEBA
0FD11251A1EA
0FD11251A1EA1111EC13C000007C
0FDA8
0FDA8251A1EA1111EC1311A18716FA3E73FDD2A5
0FD21AD
0FD21AD1EA1111EC1311A18716FA3E73FDD2A511F1FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF35390001C
0FD85131
0FD851311EA1111EC1311A18711F7000000000035FF7
0FD90641
0FD9011F410C0
0FD106C0
0FD11FEBA
0FD85131
0FD8511FE0BA
0FDA8
0FDA81111A105DBA
0FD106C0
0FD11FEBA
0FD11251A1EA
```

Output proper:
```
02 01 ad 
11 f1 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 35 39 00 00 00 1c
```

Code:
```
#include <WiFi.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <driver/timer.h>
#include <driver/uart.h>

const int RS485_RX_PIN = 34;
const int RS485_TX_PIN = 32;

#define RS485 UART2
#define RS485_NUM UART_NUM_2
#define RS485_BUF_SIZE 64

static TaskHandle_t RS485_Process;
static intr_handle_t RS485_Intr;
static QueueHandle_t RS485_queue;

static void IRAM_ATTR RS485_RX_ISR(void * arg);
static void RS485_Process_TASK (void * arg);

typedef struct {
  uint8_t data[RS485_BUF_SIZE];
  uint8_t size;
} RS485_Packet;

void setup(){
  Serial.begin(115200);
  Serial.println();

  RS485_queue = xQueueCreate( 2, sizeof(RS485_Packet) );
  xTaskCreatePinnedToCore(RS485_Process_TASK, "RS485_Process", 2048, NULL, 12, &RS485_Process, NULL);
  
  uart_config_t uart_config_rs485 = {
    .baud_rate = 9600,
    .data_bits = UART_DATA_8_BITS,
    .parity = UART_PARITY_DISABLE,
    .stop_bits = UART_STOP_BITS_1,
    .flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
    .rx_flow_ctrl_thresh = 0
  }; 

  uart_intr_config_t uart_intr_config_rs485 = {
    .intr_enable_mask = UART_RXFIFO_FULL_INT_ENA_M
                      | UART_RXFIFO_TOUT_INT_ENA_M
                      | UART_FRM_ERR_INT_ENA_M
                      | UART_RXFIFO_OVF_INT_ENA_M
                      | UART_BRK_DET_INT_ENA_M
                      | UART_PARITY_ERR_INT_ENA_M,
    .rx_timeout_thresh = 3,
    .txfifo_empty_intr_thresh = 1,
    .rxfifo_full_thresh = RS485_BUF_SIZE
  };
  
  uart_param_config(RS485_NUM, &uart_config_rs485);
  uart_set_pin(RS485_NUM, RS485_TX_PIN, RS485_RX_PIN, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);
  uart_driver_install( RS485_NUM, RS485_BUF_SIZE * 3, 0, 0, NULL, 0 );
//  uart_set_rx_timeout(RS485_NUM, 3);

  uart_isr_free(RS485_NUM);
  uart_intr_config(RS485_NUM, &uart_intr_config_rs485);
  uart_isr_register(RS485_NUM, RS485_RX_ISR, NULL, ESP_INTR_FLAG_LOWMED | ESP_INTR_FLAG_IRAM, &RS485_Intr);
  uart_enable_rx_intr(RS485_NUM);

  uart_flush(RS485_NUM);
}

void loop(){

  vTaskDelay(1);
}

void IRAM_ATTR RS485_RX_ISR(void * arg) {

  RS485_Packet queue;

  uint16_t numberOfBytesToRead;
  BaseType_t xHigherPrioritTaskWoken = pdFALSE;

  volatile uart_dev_t* uart = &RS485;

  numberOfBytesToRead = RS485.status.rxfifo_cnt; // number of bytes in UART buffer
  queue.size = numberOfBytesToRead;

  while ( uart->status.rxfifo_cnt ) {
      queue.data[queue.size] = uart->fifo.rw_byte;
      queue.size++;
  }

  xQueueSendFromISR (RS485_queue, &queue, &xHigherPrioritTaskWoken );

  uart_clear_intr_status( RS485_NUM, UART_RXFIFO_TOUT_INT_CLR | UART_RXFIFO_FULL_INT_CLR | UART_RXFIFO_TOUT_INT_CLR );
}

static void RS485_Process_TASK(void * pvParameters) {
  
  RS485_Packet queue;

  byte parity;
  
  for (;;) {
    if (xQueueReceive(RS485_queue, &queue, portMAX_DELAY)) {

      for (int i=1; i<queue.size; i++) {
        Serial.print(queue.data[i], HEX);
      }
      Serial.println();
    }
    vTaskDelay(1);
  }

  vTaskDelete( NULL );
}
```

Gamelauncher, 2021-04-24:
Solved, alter the way to reset the flags.

closed 2021-04-24 by Gamelauncher