espressif/arduino-esp32 #5050: ESP32-S2 PWM for a Servo pulse

StuartsProjects, 2021-04-13:
ESP32-S2 PWM settings do not work as expected. 

The code bellow when run on an ESP32 WROOM Development board, generates 1ms and 2ms pulses at 50hz, works as intended. It correctly produces a pulse to control a servo that shifts between 1m2 and 2mS.  

```
/*
On ESP32 generates a sweeping pulse of 1mS to 2mS at 50hz
*/

const int servoPin = 17;

int dutyCycle = 0;

const int PWMFreq = 50;                                //50hz
const int PWMChannel = 0;
const int PWMResolution = 12;                       //12 bits 0-4095
 
void setup()
{  
  ledcSetup(PWMChannel, PWMFreq, PWMResolution);
  ledcAttachPin(servoPin, PWMChannel);
  ledcWrite(PWMChannel, dutyCycle);
}

void loop()
{
    ledcWrite(PWMChannel, 200);                     //approx 1ms
    delay(2000);    
  
    ledcWrite(PWMChannel, 400);                     //approx 2mS
    delay(2000);    
}
```

When the above code is run on a ESP32-S2 Saola board the pulse produced is not correct, it has a period of circa 1500mS and periods of 80mS and 160mS. 

If in the code I change PWMFreq  from 50 to 4096, I end up with a pulse that is 1ms to 2ms @ 50hz on the ESP32-S2.

lbernstone, 2021-04-13:
It seems like something is wrong at the higher resolutions.  Can you test at 10 bit and scale your timings to fit?

StuartsProjects, 2021-04-13:
Using this;

const int PWMFreq = 50;                             //50hz
const int PWMChannel = 0;
const int PWMResolution = 10;                      
 
Gives me a pulse period of 250uS. 

I need to set PWMFreq = 4096; to get back to a 20mS period.

Dlloydev, 2021-04-18:
In testing an analogWrite library I wrote (using a ESP32S2-Saola-1M) that uses the latest commit here (it wraps the ledc functions) here from 2 days ago, I find that the set frequency is off by a factor of 80. Need to use 80000 to get 1kHz PWM signal.
```
  analogWriteFrequency(35, 80000);
  analogWriteResolution(35, 8);
  analogWrite(35, 31);
```
The bit resolution and duty value work as expected, but not the frequency.

[image]

PR #5452 (2021-07-26): Fixes ESP32-S2 LEDC PWM #5375 #5050
Fixes: #5375
Fixes: #5050
commit 65eafd16 (2021-07-26): Fixes ESP32-S2 LEDC PWM #5375 #5050 (#5452)

Fixes: #5375
Fixes: #5050
closed 2021-07-11 by stale[bot]