espressif/arduino-esp32 #5051: Empty project, binary size incredibly large

Reneg973, 2021-04-13:
I just have an empty arduino sketch. I compile it for the SAMD21 (Arduino-samd core), the binary is < 50 kByte, its large, but ok.

Now I take the same empty sketch, compile it for ESP32 and binary size is 240 kByte. What's going wrong?

chansheunglong, 2021-04-13:
SAMD21 and ESP32 are two **different** MCU, their board package is also different, nothing is wrong.
You got tones of program storage space in ESP32, nothing to worry about.

Reneg973, 2021-04-13:
Tons? I think you overstate it?
I've inserted WiFi library with Web server/client, json dispatching/serial communication. My sketch now is 970 kB, without a display library or the other things I want to communicate with.

chansheunglong, 2021-04-13:
I wrote a LoRa Gateway with Web Interface and local decoder under 2MB, maybe your code need optimization? If space is important, going for 16MB variant may be better. Also, if OTA is not needed, using a custom partition table can also give you more usable space. If your project contain web page, storing them in the SPIFFS or FFat partition is also another option, with that I can include almost 1.5MB of CSS and JS files without hurting the programming space and performance.

Reneg973, 2021-04-13:
It would be good if it would be my code. Unfortunately I didn't wrote one line of code, just added 2 3rd  party libraries.

lbernstone, 2021-04-13:
A default installation of Windows 10 is almost 30GB.  As mentioned, esp32 is not a samd.  There is an operating system and lots of libraries behind your "empty" sketch.

closed 2021-07-11 by stale[bot]