espressif/arduino-esp32 #5053: Hi all,

Shane87F3Ng, 2021-04-13:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3?
IDE name:						?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					?115200?
Computer OS:                                        ?Windows 10? ?Mac OSX? ?Ubuntu?

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

societyofrobots, 2021-04-19:
Try NimBLE.

This is what I use:
https://github.com/h2zero/NimBLE-Arduino/tree/master/examples/Refactored_original_examples/BLE_uart

Shane87F3Ng, 2021-04-21:
> Try NimBLE.
> 
> This is what I use:
> https://github.com/h2zero/NimBLE-Arduino/tree/master/examples/Refactored_original_examples/BLE_uart

Thanks for your reply!
As i try to use BT-serial, i think it is not possible with BLE? Thought i have to use BT-classic?

societyofrobots, 2021-04-22:
I use NimBLE for passing serial.

Use this to declare it a serial port:
`pAdvertising->addServiceUUID((uint16_t)0x1101);`

before you start advertising:
`BLEDevice::startAdvertising();`

Metaln00b, 2021-05-01:
Same problem here with 1.0.6:
No connection possible. Using 1.0.4 no problem.
```
02:22:10.444 -> ets Jun  8 2016 00:22:57
02:22:10.444 -> 
02:22:10.444 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
02:22:10.492 -> flash read err, 1000
02:22:10.492 -> ets_main.c 371 
02:22:10.828 -> ets Jun  8 2016 00:22:57
02:22:10.828 -> 
02:22:10.828 -> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
02:22:10.828 -> configsip: 0, SPIWP:0xee
02:22:10.828 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
02:22:10.828 -> mode:DIO, clock div:1
02:22:10.876 -> load:0x3fff0018,len:4
02:22:10.876 -> load:0x3fff001c,len:1044
02:22:10.876 -> load:0x40078000,len:10124
02:22:10.876 -> load:0x40080400,len:5856
02:22:10.876 -> entry 0x400806a8
02:22:11.161 -> Setup
02:22:11.161 -> [D][ESP32-BT.ino:66] ledc_init(): ledc_channel_config returned 0
02:22:11.209 -> [D][ESP32-BT.ino:75] ledc_init(): ledc_timer_set: divider: 0x3d090 duty_resolution: 10
02:22:11.209 -> 
02:22:11.209 -> [D][ESP32-BT.ino:66] ledc_init(): ledc_channel_config returned 0
02:22:11.209 -> [D][ESP32-BT.ino:75] ledc_init(): ledc_timer_set: divider: 0x3d090 duty_resolution: 10
02:22:11.209 -> 
02:22:12.692 -> [I][BluetoothSerial.cpp:571] _init_bt(): device name set
02:22:12.692 -> [I][BluetoothSerial.cpp:243] esp_spp_cb(): ESP_SPP_INIT_EVT
02:22:12.692 -> [I][BluetoothSerial.cpp:246] esp_spp_cb(): ESP_SPP_INIT_EVT: slave: start
02:22:12.692 -> [I][BluetoothSerial.cpp:347] esp_spp_cb(): ESP_SPP_START_EVT

ASSERT_WARN(103 24), in lc_task.c at line 877
ASSERT_WARN(103 24), in lc_task.c at line 877
ASSERT_WARN(103 24), in lc_task.c at line 877
ASSERT_WARN(103 24), in lc_task.c at line 877
ASSERT_WARN(103 24), in lc_task.c at line 877
ASSERT_WARN(103 24), in lc_task.c at line 877
ASSERT_WARN(103 24), in lc_task.c at line 877
...

```

Shane87F3Ng, 2021-05-04:
> Same problem here with 1.0.6:
> No connection possible. Using 1.0.4 no problem.
> ```
> 02:22:10.444 -> ets Jun  8 2016 00:22:57
> 02:22:10.444 ->
> 02:22:10.444 -> rst:0x1 (POWERON_RESET),boot:0x13 [...]

Yes, 1.0.4 is doing the trick, so it seems there is a problem with the newer versions

Shane87F3Ng, 2021-07-15:
Will this issued bei focused in a future update? Maybe in 2.0.0 ?

alastaira, 2022-01-20:
Faced this issue today (or a very similar one, at least) - I can successfully use SerialBT to pair ESP32 to another, pin-protected, device (HC-05) and receive a stream of data just fine. 
But, periodically (every few seconds), I receive an error printed to the serial output: 
ASSERT_WARN(1 8), in lc_task.c at line 1394

All Google searches seem to point to this thread as being the only useful reference, and the only solution (to compile against v1.0.4 rather than v1.0.6), which feels unsatifactory, but does resolve the issue.

Shane87F3Ng, 2022-03-15:
I could not find any other solution than just use 1.0.4 
Newer versions are still not working

zvikaf, 2022-03-15:
WOW bummer...  :-(

alastaira, 2022-03-19:
> Did you find any solution ? @alastaira ?

Nope, unfortunately not - just seems to be a regression bug in later versions - I had to go back to v1.0.4.

Shane87F3Ng, 2022-03-20:
Yes, i really hope they will fix it i a future update

zvikaf, 2022-03-20:
Went back to 1.0.4 warning indeed stopped, but the ESP now falsely acknowledge connection by address or by name .   
checked 1.0.5, correctly connects, behaves like the 1.0.6

Shane87F3Ng, 2022-03-30:
Thanks for your answer!!

I can do that on next monday, until then i am away, but maybe one of the others can check it earlier :)

alastaira, 2022-04-01:
@VojtechBartoska I just tried compiling against the latest 2.0.3-RC1 but unfortunately the issue remains for me. The only difference is the line number on which it occurs ;)

`ASSERT_WARN(1 8), in lc_task.c at line 1409`

alastaira, 2022-04-01:
@VojtechBartoska And thankyou for your attention - it's much appreciated :)

nabe-abk, 2022-04-29:
One way to solve this problem is to change the sdk build options.

**Original**
```
# CONFIG_BTDM_CTRL_MODE_BLE_ONLY is not set
# CONFIG_BTDM_CTRL_MODE_BR_EDR_ONLY is not set
CONFIG_BTDM_CTRL_MODE_BTDM=y
```

**Solved**
```
# CONFIG_BTDM_CTRL_MODE_BLE_ONLY is not set
CONFIG_BTDM_CTRL_MODE_BR_EDR_ONLY=y
# CONFIG_BTDM_CTRL_MODE_BTDM=y
```

ricardoquesada, 2022-05-04:
Might be a duplicate from this one: https://github.com/espressif/arduino-esp32/issues/6193

Shane87F3Ng, 2022-05-04:
@ricardoquesada yes it is, as @alastaira already wrote before, these problems are definetly related and it does not matter what kind of tasks you try to do, it happens even while purely connecting, if you look at my example code above (first post)

@VojtechBartoska could it be related to the esp-wroom revision we are using?

Shane87F3Ng, 2022-05-04:
> One way to solve this problem is to change the sdk build options.
>
> **Original**
> ```
> # CONFIG_BTDM_CTRL_MODE_BLE_ONLY is not set
> # CONFIG_BTDM_CTRL_MODE_BR_EDR_ONLY is not set [...]

How can it be done?

ricardoquesada, 2022-05-04:
@Shane87F3Ng ty.

That can be done from `idf.py menuconfig`  -> "components" -> "bluetooth" and only select BR/EDR (not dual stack)

but it didn't work for me. I'm still getting the `ASSERT_WARN(103 23), in lc_task.c at line` messages with BR/EDR only stack.

nabe-abk, 2022-05-08:
I rewrote "sdkconfig" directly without using "menuconfig" for rebuilding ESP-IDF v4.4 SDK.
It works on Windows10 with BT5.0 adapter on Arudino-esp32 v2.0.x (replaced sdk with it).

closed 2025-01-16 by Parsaabasi