espressif/arduino-esp32 #5055: attachInterrupt GPIO fires randomly when using WiFi (example full code included to repeat issue)

mars000, 2021-04-13:
Board: LOLIN D32
Visual Studio Code with latest version of PlatformIO

See platformio file and sketch (sorry the format being off).
Its a basic sketch.
I have a simple flowsensor attached to pin 36 on ESP32.
[image]

**RESULTS OF TESTS:**
-  **[TEST 1]** If you compile it with `#define wifiON` not commented out you should see pulseCounts > 0 even if the flow sensor is not spinning, i.e. no GPIO pulses active on pin_sensorPIN on ESP32. It seems like it just firing interrupt randomly
- **[TEST 2]** If you compile it with `//#define wifoON `  i.e.. commented then pulseCounts = 0 when no air or water flowing through flow sensor i.e. interrupt it firing when it should be
- **[TEST 3]** If you compile if with `#define wifiON` not commented out and with platform=espressif32@2.1.0 it work just fine with pulseCounts = 0 when no GPIO unput received on pin_sensorPIN

I hope that makes sense. It look to me that WiFi library somehow interferes with attatchInterrupt using GPIO

**TEST 1 Serial Output: (WiFi on)**
```
[env:lolin_d32]
`platform = espressif32`
```
[image]

**TEST 2 Serial Output (WiFi off)**
```
[env:lolin_d32]
platform = espressif32
```
[image]

**TEST 3 Serial Output (using platformIO version 2.1.0 of esp32 library)**
```
[env:lolin_d32]
platform = espressif32@2.1.0
```
[image]

Full `platformio.ini` file:
```
[platformio]
default_envs = lolin_d32

[env:lolin_d32]
platform = espressif32  
framework = arduino
board = lolin_d32
lib_ldf_mode = chain+
upload_port = COM11
monitor_speed = 115200
build_type = release
lib_deps = jfturcot/SimpleTimer
```

Sketch:
```cpp
#include <Arduino.h>

#define pin_sensorPin 36

#define wifiON   // switch to include or exclude WiFi

void IRAM_ATTR pulseCounter();

// BEGIN: code for WiFi and NTP Time Server connectivity
#ifdef wifiON
#include <WiFi.h>
#include "time.h"
#endif

#include <SimpleTimer.h>

#ifdef wifiON
const char *ssid = "xxxxx";
const char *password = "xxxx";

const char *ntpServer = "pool.ntp.org";
const long gmtOffset_sec = 10 * 3600;
int daylightOffset_sec = 0;
#endif

// END: code for NTP

// the timer object
SimpleTimer timer;

volatile int pulseCount = 0;

// Class Setup : BEGIN
class flowmeterClass
{
	
public:
	//flowmeter() // constructor
	// {}

	//======== Calculate WaterFlow ==========
	void WorkRoutine(void)
	{
		static unsigned long oldTime = millis();
		
		if ((millis() - oldTime) > 1000) // Only process counters once per second
		{
			// Reset the pulse counter so we can start incrementing again given we are not using pulseCount varaible again in this function from here on...
			Serial.print(F("504-[flowmeter_class.h] pulseCount = "));
			Serial.println(pulseCount);
		
			pulseCount = 0;
			oldTime = millis();

		}
	}
};
// Class Setup : END

flowmeterClass flowmeterObject ;

#ifdef wifiON
void printLocalTime()
{
  struct tm timeinfo;
  if (!getLocalTime(&timeinfo))
  {
    Serial.println("Failed to obtain time");
    return;
  }
  Serial.println(&timeinfo, "%A, %B %d %Y %H:%M:%S");
}
#endif

void runWorkRoutine()
{
  flowmeterObject.WorkRoutine();
}

void setup()
{
  // put your setup code here, to run once:
  Serial.begin(115200);
  pinMode(pin_openclose, OUTPUT);
  pinMode(pin_sensorPin, INPUT_PULLUP);
  
  //connect to WiFi
 
#ifdef wifiON
  WiFi.begin(ssid, password);
  Serial.println(F("74"));

  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  Serial.println(" CONNECTED");

  //init and get the time
  configTime(gmtOffset_sec, daylightOffset_sec, ntpServer);
  printLocalTime();
#endif
  
  timer.setInterval(1000, runWorkRoutine);  // run every second
  attachInterrupt(pin_sensorPin, pulseCounter, FALLING);  // setup interrupt
}

void incrementCounter()
{
    pulseCount++; // increment everytime interrupt called  
}

void IRAM_ATTR pulseCounter()
{
  static long last_millis = 0;
  if (abs(millis() - last_millis) > 1)
  {
 incrementCounter();
}
last_millis = millis();
}

void loop()
{
timer.run();
 }
```

lbernstone, 2021-04-13:
https://github.com/espressif/esp-idf/commit/d890a516a1097f0a07788e203fdb1a82bb83520e

mars000, 2021-04-14:
thanks for reply.
Unfortunately I tried a different pin, in fact I tried a bunch of other, GPIO 34, 26, 27 but same problem.
Work perfect when using `espresiff32@2.1.0` library but 3.0.0 or higher results in GPIO interrupt firing :-(
Really stuck with this.

closed 2021-04-24 by mars000

gslongo, 2024-03-06:
@mars000 can you can how you managed to solve your issue ?
I know it was a long time ago but let's try

Thanks !