espressif/arduino-esp32 #5058: 好,ESP32 ADC 校准问题

hzhh110, 2021-04-14:
好,我这边想要获得继电器的外部电压,发现analogRead获得的并不准确,非线性的,如何校准呢?

hzhh110, 2021-04-14:
> 好,我这边想要获得继电器的外部电压,发现analogRead获得的并不准确,非线性的,如何校准呢?

closed 2021-04-14 by hzhh110

chansheunglong, 2021-04-14:
You can reference this image from [this ](https://esp32.com/viewtopic.php?t=13089)link.
ESP32's ADC is non-linear, quote from a forum member

> SAR is always going to be biased towards the ends of the scale, and the esp32 seems particularly insensitive there.

Take a look at this issue too: [https://esp32.com/viewtopic.php?t=13089](https://esp32.com/viewtopic.php?t=13089)

You will need to do a offset either using the official method in the documentation or simply just apply a offset/equation for your ADC reading in your code.

ESP32的ADC是非線性的, 較正方法可以參考官方文檔或者在軟件進行手動較正.

[image]