espressif/arduino-esp32 #5064: Problems using arduino-ESP32 as ESP-IDF component

lgLindstrom, 2021-04-15:
Hi

I am trying to follow this instruction [https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md)

This line gives me problem:

  ` make menuconfig`

The make command seems to be missing and als the menuconfig file:

My process.

I am running VsCode on Windows 10.
I have installed VsCode ESP-IDF extension.
I have configured the ESP-IDF extension. Downloaded version 3.5
I have installed the Blink example from the ESP-IDG.
I have compiled the example and it worked well.

THEN,
- In the folder of the Blink example, I using Powershell , I tried to follow the installation instruction above and got this error with make menuconfig.
- After that the Blink example no longer compiles with the following error:

```
Executing task: cmake -G Ninja .. <

-- Checking Python dependencies...
Python requirements from C:/Users/lglin/esp/esp-idf\requirements.txt are satisfied.
-- Building for target esp32
-- ccache will be used for faster builds
CMake Error at C:/Users/lglin/OneDrive/experimentalProjects/esp-idf/blink/blink/components/arduino/CMakeLists.txt:171 (idf_component_register):
  Unknown CMake command "idf_component_register".
Call Stack (most recent call first):
  C:/Users/lglin/esp/esp-idf/tools/cmake/scripts/expand_requirements.cmake:107 (include)
  C:/Users/lglin/esp/esp-idf/tools/cmake/scripts/expand_requirements.cmake:114 (expand_component_requirements)
  C:/Users/lglin/esp/esp-idf/tools/cmake/scripts/expand_requirements.cmake:217 (expand_component_requirements)

CMake Error at C:/Users/lglin/esp/esp-idf/tools/cmake/idf_functions.cmake:339 (message):
  Failed to expand component requirements
Call Stack (most recent call first):
  C:/Users/lglin/esp/esp-idf/tools/cmake/project.cmake:173 (idf_import_components)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/lglin/OneDrive/experimentalProjects/esp-idf/blink/blink/build/CMakeFiles/CMakeOutput.log". 
See also "C:/Users/lglin/OneDrive/experimentalProjects/esp-idf/blink/blink/build/CMakeFiles/CMakeError.log".  
The terminal process "C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake -G Ninja .." terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

So, what to do?

lbernstone, 2021-04-15:
In IDF 4+, all commands should use idf.py instead of make
`idf.py menuconfig`
Please follow the instructions at [esp-idf](https://github.com/espressif/esp-idf#setup-build-environment) to get a functional environment.

lgLindstrom, 2021-04-15:
I read this..

[image]

so I assumed that I should install ESP-IDF 3.5

You are saying it will work with 4.0,, ???

lbernstone, 2021-04-15:
Sorry, I assumed you were compiling with idf 4.
I cannot help you with Windows compiles.  This looks like you have a problem with python.

chegewara, 2021-04-15:
No matter which esp-idf, logs shows you are using ninja, so you should use `idf.py menuconfig`.

The problem is with this part `Unknown CMake command "idf_component_register".`, which means you have something not installed or wrong version.
I saw this once, but honestly i dont remember what was the cause, but you can fix it when you change to old cmake style with `register_component()`

I think you may get better and faster support with this issue when you post it in esp-idf repo.

dobairoland, 2021-04-16:
Maybe it would be worth trying to remove the project's `build` directory and re-run `make menuconfig` again. Also please post the full output you are getting. The issue report contains cmake output.

me-no-dev, 2021-04-16:
Current Arduino master is building with IDF master. There is a release/v1.0 branch that works with IDF 3.3.5. Instructions need updating. I would recommend you to use the lib builder: https://github.com/espressif/esp32-arduino-lib-builder
master branch builds with IDF master and release/v3.3 branch builds with IDF 3.3.5 (and Arduino release/v1.0)

brandonros, 2021-04-20:
Something is for sure broken somewhere. It's like hot potato on where the latest bug is trying to build this.

## esp-idf version

```
$ pwd
/Users/brandonros/Desktop/esp32-arduino-lib-builder/esp-idf
Brandons-MacBook-Air:esp-idf brandonros$ git show
commit cf457d412a7748139b77c94e7debe72ee86199af (HEAD -> master, origin/master, origin/HEAD, 0a)
Merge: db67047eb 370fedf96
Author: Michael (XIAO Xufeng) <xiaoxufeng@espressif.com>
Date:   Fri Apr 16 04:22:57 2021 +0000

    Merge branch 'feature/update_spi_programming_guide_on_c3' into 'master'
    
    spi: update esp32c3 programming guide
    
    Closes IDF-2343
    
    See merge request espressif/esp-idf!12490
```

## project version

```
$ pwd
/Users/brandonros/Desktop/esp32-arduino-lib-builder/esp-idf/examples/get-started/sample_project

$ idf.py build
Executing action: all (aliases: build)
Running cmake in directory /Users/brandonros/Desktop/esp32-arduino-lib-builder/esp-idf/examples/get-started/sample_project/build
Executing "cmake -G 'Unix Makefiles' -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /Users/brandonros/Desktop/esp32-arduino-lib-builder/esp-idf/examples/get-started/sample_project"...
-- Building ESP-IDF components for target esp32
CMake Error at /Users/brandonros/Desktop/esp32-arduino-lib-builder/esp-idf/tools/cmake/build.cmake:194 (message):
  Failed to resolve component 'arduino_tinyusb'.
Call Stack (most recent call first):
  /Users/brandonros/Desktop/esp32-arduino-lib-builder/esp-idf/tools/cmake/build.cmake:225 (__build_resolve_and_add_req)
  /Users/brandonros/Desktop/esp32-arduino-lib-builder/esp-idf/tools/cmake/build.cmake:508 (__build_expand_requirements)
  /Users/brandonros/Desktop/esp32-arduino-lib-builder/esp-idf/tools/cmake/project.cmake:361 (idf_build_process)
  CMakeLists.txt:13 (project)

-- Configuring incomplete, errors occurred!
See also "/Users/brandonros/Desktop/esp32-arduino-lib-builder/esp-idf/examples/get-started/sample_project/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
```

## arduino component version

```
$ pwd
/Users/brandonros/Desktop/esp32-arduino-lib-builder/esp-idf/examples/get-started/sample_project/components/arduino
Brandons-MacBook-Air:arduino brandonros$ git log -1
commit 82161bebe357b78d1343fa0878e17c1fc8c3b315 (HEAD -> master, origin/master, origin/HEAD)
Author: me-no-dev <hristo@espressif.com>
Date:   Mon Apr 19 14:25:09 2021 +0300

    Use cpu_hal_get_cycle_count for all chips
```

If I need to be pegged on some very specific branch/release of:

* esp32-arduino-lib-builder
* esp-idf
* arduino component

can the winning example please be documented somewhere? if you follow the current documentation 1:1, you will run into bugs about:

* esp-rainmaker
* esp_littlefs/src/littlefs
* `esp32-arduino-lib-builder/build.sh` will throw `esp32-arduino-lib-builder/esp-idf' is not buildable`, but if you `source tools/install-esp-idf.sh` (which build.sh does... but you need to do it again randomly, undocumented), then it'll work

brandonros, 2021-04-20:
I started from scratch and got the same result:

```
git clone -b v4.3-beta2 --recursive https://github.com/espressif/esp-idf.git esp-idf-v4.3-beta2
./install.sh
. ./export.sh 
cd examples/get-started/sample_project
mkdir components
cd components
git clone -b 2.0.0-alpha1 --recursive https://github.com/espressif/arduino-esp32.git arduino
cd arduino
git submodule update --init --recursive 
cd ../../
idf.py build
```

```
$ idf.py build
Executing action: all (aliases: build)
Running cmake in directory /Users/brandonros/Desktop/esp-idf-v4.3-beta2/examples/get-started/sample_project/build
Executing "cmake -G 'Unix Makefiles' -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /Users/brandonros/Desktop/esp-idf-v4.3-beta2/examples/get-started/sample_project"...
-- Found Git: /usr/bin/git (found version "2.24.3 (Apple Git-128)") 
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/brandonros/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/brandonros/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/brandonros/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
CMake Error at /Users/brandonros/Desktop/esp-idf-v4.3-beta2/tools/cmake/build.cmake:189 (message):
  Failed to resolve component 'arduino_tinyusb'.
Call Stack (most recent call first):
  /Users/brandonros/Desktop/esp-idf-v4.3-beta2/tools/cmake/build.cmake:220 (__build_resolve_and_add_req)
  /Users/brandonros/Desktop/esp-idf-v4.3-beta2/tools/cmake/build.cmake:435 (__build_expand_requirements)
  /Users/brandonros/Desktop/esp-idf-v4.3-beta2/tools/cmake/project.cmake:396 (idf_build_process)
  CMakeLists.txt:8 (project)

-- Configuring incomplete, errors occurred!
See also "/Users/brandonros/Desktop/esp-idf-v4.3-beta2/examples/get-started/sample_project/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
```

brandonros, 2021-04-20:
Then you need to do

```
git submodule add https://github.com/joltwallet/esp_littlefs.git
git submodule update --init --recursive 
```

Then it will fail with

```
WARNING: Missing submodule examples/get-started/sample_project/components/esp_littlefs/littlefs...
Attempting 'git submodule update --init examples/get-started/sample_project/components/esp_littlefs/littlefs' in esp-idf root directory...
error: pathspec 'examples/get-started/sample_project/components/esp_littlefs/littlefs' did not match any file(s) known to git
make: *** [/Users/brandonros/Desktop/esp-idf-v4.3-beta2/examples/get-started/sample_project/components/esp_littlefs/littlefs/.git] Error 1
```

you need to change `COMPONENT_SUBMODULES += $(PROJECT_PATH)/components/esp_littlefs/littlefs` to `COMPONENT_SUBMODULES += $(PROJECT_PATH)/components/esp_littlefs/src/littlefs` in `build/esp_littlefs/component_project_vars.mk`

brandonros, 2021-04-20:
Then it will fail with this:

```
CXX build/arduino/libraries/RainMaker/src//RMaker.o
In file included from /Users/brandonros/Desktop/esp-idf-v4.3-beta2/examples/get-started/sample_project/components/arduino/libraries/RainMaker/src/RMakerParam.h:4,
                 from /Users/brandonros/Desktop/esp-idf-v4.3-beta2/examples/get-started/sample_project/components/arduino/libraries/RainMaker/src/RMakerDevice.h:4,
                 from /Users/brandonros/Desktop/esp-idf-v4.3-beta2/examples/get-started/sample_project/components/arduino/libraries/RainMaker/src/RMakerNode.h:4,
                 from /Users/brandonros/Desktop/esp-idf-v4.3-beta2/examples/get-started/sample_project/components/arduino/libraries/RainMaker/src/RMaker.h:5,
                 from /Users/brandonros/Desktop/esp-idf-v4.3-beta2/examples/get-started/sample_project/components/arduino/libraries/RainMaker/src/RMaker.cpp:1:
/Users/brandonros/Desktop/esp-idf-v4.3-beta2/examples/get-started/sample_project/components/arduino/libraries/RainMaker/src/RMakerType.h:4:10: fatal error: esp_rmaker_core.h: No such file or directory
 #include <esp_rmaker_core.h>
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [libraries/RainMaker/src//RMaker.o] Error 1
make: *** [component-arduino-build] Error 2

```

brandonros, 2021-04-20:
Then it will fail with this:

```
$ make EXTRA_COMPONENT_DIRS=components/esp-rainmaker/components
Toolchain path: /Users/brandonros/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: esp-2020r3
Compiler version: 8.4.0
Python requirements from /Users/brandonros/Desktop/esp-idf-v4.3-beta2/requirements.txt are satisfied.

App "sample_project" version: v4.3-beta2-dirty
Arduino libraries in build: ArduinoOTA AsyncUDP BLE BluetoothSerial DNSServer EEPROM ESP32 ESPmDNS FFat FS HTTPClient HTTPUpdate HTTPUpdateServer LITTLEFS NetBIOS Preferences README.md RainMaker SD SD_MMC SPI SPIFFS SimpleBLE Ticker USB Update WebServer WiFi WiFiClientSecure WiFiProv Wire
CXX build/arduino/libraries/WiFi/src//ETH.o
/Users/brandonros/Desktop/esp-idf-v4.3-beta2/examples/get-started/sample_project/components/arduino/libraries/WiFi/src/ETH.cpp: In member function 'bool ETHClass::begin(uint8_t, int, int, int, eth_phy_type_t)':
/Users/brandonros/Desktop/esp-idf-v4.3-beta2/examples/get-started/sample_project/components/arduino/libraries/WiFi/src/ETH.cpp:174:23: error: 'esp_eth_phy_new_ksz8081' was not declared in this scope
             eth_phy = esp_eth_phy_new_ksz8081(&phy_config);
                       ^~~~~~~~~~~~~~~~~~~~~~~
/Users/brandonros/Desktop/esp-idf-v4.3-beta2/examples/get-started/sample_project/components/arduino/libraries/WiFi/src/ETH.cpp:174:23: note: suggested alternative: 'esp_eth_phy_new_ksz8041'
             eth_phy = esp_eth_phy_new_ksz8081(&phy_config);
                       ^~~~~~~~~~~~~~~~~~~~~~~
                       esp_eth_phy_new_ksz8041
make[1]: *** [libraries/WiFi/src//ETH.o] Error 1
```

brandonros, 2021-04-20:
[image]

Replace `esp_eth_phy_new_ksz8081` with `esp_eth_phy_new_ksz8041` manually in 4 files. New error.

```
CXX build/arduino/libraries/WiFiClientSecure/src//ssl_client.o
/Users/brandonros/Desktop/esp-idf-v4.3-beta2/examples/get-started/sample_project/components/arduino/libraries/WiFiClientSecure/src/ssl_client.cpp:23:4: error: #error "Please configure IDF framework to include mbedTLS -> Enable pre-shared-key ciphersuites and activate at least one cipher"
 #  error "Please configure IDF framework to include mbedTLS -> Enable pre-shared-key ciphersuites and activate at least one cipher"
    ^~~~~
/Users/brandonros/Desktop/esp-idf-v4.3-beta2/examples/get-started/sample_project/components/arduino/libraries/WiFiClientSecure/src/ssl_client.cpp: In function 'int start_ssl_client(sslclient_context*, const char*, uint32_t, int, const char*, const char*, const char*, const char*, const char*, bool)':
/Users/brandonros/Desktop/esp-idf-v4.3-beta2/examples/get-started/sample_project/components/arduino/libraries/WiFiClientSecure/src/ssl_client.cpp:164:15: error: 'mbedtls_ssl_conf_psk' was not declared in this scope
         ret = mbedtls_ssl_conf_psk(&ssl_client->ssl_conf, psk, psk_len,
               ^~~~~~~~~~~~~~~~~~~~
/Users/brandonros/Desktop/esp-idf-v4.3-beta2/examples/get-started/sample_project/components/arduino/libraries/WiFiClientSecure/src/ssl_client.cpp:164:15: note: suggested alternative: 'mbedtls_ssl_conf_sni'
         ret = mbedtls_ssl_conf_psk(&ssl_client->ssl_conf, psk, psk_len,
               ^~~~~~~~~~~~~~~~~~~~
               mbedtls_ssl_conf_sni
make[1]: *** [libraries/WiFiClientSecure/src//ssl_client.o] Error 1
make: *** [component-arduino-build] Error 2
```

brandonros, 2021-04-20:
new error lol

```
/Users/brandonros/Desktop/esp-idf-v4.3-beta2/examples/get-started/sample_project/components/esp-rainmaker/components/button/button/button.c: In function 'iot_button_create':
/Users/brandonros/Desktop/esp-idf-v4.3-beta2/examples/get-started/sample_project/components/esp-rainmaker/components/button/button/button.c:65:40: error: 'CONFIG_IO_GLITCH_FILTER_TIME_MS' undeclared (first use in this function); did you mean 'BUTTON_GLITCH_FILTER_TIME_MS'?
 #define BUTTON_GLITCH_FILTER_TIME_MS   CONFIG_IO_GLITCH_FILTER_TIME_MS
                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/brandonros/Desktop/esp-idf-v4.3-beta2/examples/get-started/sample_project/components/esp-rainmaker/components/button/button/button.c:233:32: note: in expansion of macro 'BUTTON_GLITCH_FILTER_TIME_MS'
     btn->tap_rls_cb.interval = BUTTON_GLITCH_FILTER_TIME_MS / portTICK_PERIOD_MS;
                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/brandonros/Desktop/esp-idf-v4.3-beta2/examples/get-started/sample_project/components/esp-rainmaker/components/button/button/button.c:65:40: note: each undeclared identifier is reported only once for each function it appears in
 #define BUTTON_GLITCH_FILTER_TIME_MS   CONFIG_IO_GLITCH_FILTER_TIME_MS
                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/brandonros/Desktop/esp-idf-v4.3-beta2/examples/get-started/sample_project/components/esp-rainmaker/components/button/button/button.c:233:32: note: in expansion of macro 'BUTTON_GLITCH_FILTER_TIME_MS'
     btn->tap_rls_cb.interval = BUTTON_GLITCH_FILTER_TIME_MS / portTICK_PERIOD_MS;
                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/brandonros/Desktop/esp-idf-v4.3-beta2/examples/get-started/sample_project/components/esp-rainmaker/components/button/button/button.c: In function 'iot_button_set_evt_cb':
/Users/brandonros/Desktop/esp-idf-v4.3-beta2/examples/get-started/sample_project/components/esp-rainmaker/components/button/button/button.c:65:40: error: 'CONFIG_IO_GLITCH_FILTER_TIME_MS' undeclared (first use in this function); did you mean 'BUTTON_GLITCH_FILTER_TIME_MS'?
 #define BUTTON_GLITCH_FILTER_TIME_MS   CONFIG_IO_GLITCH_FILTER_TIME_MS
                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/brandonros/Desktop/esp-idf-v4.3-beta2/examples/get-started/sample_project/components/esp-rainmaker/components/button/button/button.c:298:36: note: in expansion of macro 'BUTTON_GLITCH_FILTER_TIME_MS'
         btn->tap_psh_cb.interval = BUTTON_GLITCH_FILTER_TIME_MS / portTICK_RATE_MS;
                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[1]: *** [button/button.o] Error 1
make: *** [component-button-build] Error 2
```

lbernstone, 2021-04-20:
I have docker containers which have all the resources and are ready to build. https://hub.docker.com/r/lbernstone/esp32-arduino-lib-builder

brandonros, 2021-04-21:
> I have docker containers which have all the resources and are ready to build. https://hub.docker.com/r/lbernstone/esp32-arduino-lib-builder

@lbernstone 

```
$ docker run -it -v $(pwd)/out:/opt/esp/lib-builder/out lbernstone/esp32-arduino-lib-builder:1.0.6
WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested
```

Apple M1 isn't supported on 1.0.6. Any plans to make a Docker hub image for espressif/arduino-esp32:2.0.0-alpha1?

cvanelteren, 2021-04-22:
Is there a fix for this?

lbernstone, 2021-04-22:
> Apple M1 isn't supported on 1.0.6. Any plans to make a Docker hub image for espressif/arduino-esp32:2.0.0-alpha1?
> 
You can open a new issue on this, but it is going to depend on the upstream guys to build a toolchain.  What is the arch on M1 (`uname -a`)?

lbernstone, 2021-04-22:
[2.0.0alpha1](https://hub.docker.com/layers/lbernstone/esp32-arduino-lib-builder/2.0.0alpha1/images/sha256-ab1d821c1119a152969f96336597a17efde6c20d9c277713f4c4e6742070cb72?context=explore)
esp-face is handled in a hacky way, but it compiles

ul-gh, 2021-04-26:
@brandonros: Thank you for sharing your experience!

I just ran into the same issues when following the [official documentation building arduino-ESP32 as ESP-IDF component](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md).

I was looking forward to finally having a working version with the recent WiFi fixes (e.g. commit ad4cf1461bea3fb1378402b572383813ea9c2a06).

Can we expect the Arduino as ESP-IDF component to work for the upcoming 2.0?

brandonros, 2021-04-26:
I was able to get it fully working but boy was it not fun lol

https://gist.github.com/brandonros/74cae1694c34368a1181ecad801895de

it was something like this, let me know if i need to make any updates/changes

ul-gh, 2021-05-10:
Thanks @brandonros,

yes, I also had a [different compilation issue](https://github.com/espressif/esp-idf/issues/6968) for my specific project, but your recipe works for me with "release/v4.3" branch of ESP-IDF.

For the arduino-esp32 CMakelists.txt I added: (https://github.com/ul-gh/arduino-esp32/commit/a2fbaf81dc6109faade85b2d9ddf2e81a0da7587)

I also tried ESP-IDF master a few days ago, which compiled, too back then (the cherry-picked ksz8081 patch was not necessary any more in master)

AlexeyYukhin, 2021-05-11:
@brandonros Hi. Thanks for your job! Just a little question, I struggled with error "Failed to resolve component 'arduino_tinyusb'." when running 
platformio run -t menuconfig in MSVisualStudio Code 

And you suggested to  run
make menuconfig,
but than it says:
make: *** No rule to make target 'menuconfig'.  Stop.

I tried to run this command just from the project folder.

Any advices will be appreciated.

lbernstone, 2021-05-11:
https://github.com/espressif/arduino-esp32/issues/5064#issuecomment-820396911

qt1, 2021-05-19:
Failed to resolve component 'arduino_tinyusb'.

It seems like arduino_tinyusb exists only arduino-esp32 and not in esp-idf

sieren, 2021-05-22:
why is it so hard to have at least one official arduino-esp32 library that works with any kind of v4+ version of ESP-IDF?
I'm also running into this FYI.

lbernstone, 2021-05-22:
https://github.com/espressif/arduino-esp32/pull/5123

sieren, 2021-05-22:
Well that's good, but it doesn't work with ESP-IDF Master or the referenced version of the last alpha release.
Does anyone know of a combination of this repo and ESP-IDF that works? I'd gladly take commit hashes for both.

lbernstone, 2021-05-22:
https://github.com/lbernstone/arduino-esp32-dockerfiles/blob/master/Dockerfile-2.0.0a

sieren, 2021-05-22:
Thanks. I think it's a good example to visualise what's wrong with this repo and what's needed to even get a build up and running but it doesn't apply to my use-case as I'm not working with sketchbooks/Arduino IDE but merely use this as a library via CMake.

lbernstone, 2021-05-22:
Please actually look at what I have provided.  It has the commit hashes.

erichousetkc, 2021-05-24:
> Please actually look at what I have provided. It has the commit hashes.

@lbernstone Thanks for the docker file.  I checked the instruction on your [docker hub](https://hub.docker.com/r/lbernstone/esp32-arduino-lib-builder). I tried to modify the sdkconfig using "idf.py menuconfig" but there is an error:
`CMakeLists.txt not found in project directory /opt/esp/lib-builder`
I am very new to ESP32 development so I may have missed something. Your suggestions are appreciated.

lbernstone, 2021-05-24:
v1.0.x images will use `make menuconfig` rather than idf.py (and cmake)

akhil3223, 2021-06-16:
change idf_component_register() in CMakeLists.txt to register_component().
Now try "idf.py build"

Sterpa, 2021-06-17:
Unfortunately, the bug still exists.
On branches, arduino-esp32 (master) and esp-idf (release/4.3) I get an error:
```
../components/arduino/libraries/WiFi/src/ETH.cpp: In member function 'bool ETHClass::begin(uint8_t, int, int, int, eth_phy_type_t, eth_clock_mode_t)':
../components/arduino/libraries/WiFi/src/ETH.cpp:291:23: error: 'esp_eth_phy_new_ksz8081' was not declared in this scope
             eth_phy = esp_eth_phy_new_ksz8081(&phy_config);
                       ^~~~~~~~~~~~~~~~~~~~~~~
../components/arduino/libraries/WiFi/src/ETH.cpp:291:23: note: suggested alternative: 'esp_eth_phy_new_ksz8041'
             eth_phy = esp_eth_phy_new_ksz8081(&phy_config);
                       ^~~~~~~~~~~~~~~~~~~~~~~
                       esp_eth_phy_new_ksz8041
```
Replacing 81 with 41 solves the problem [issue](https://github.com/espressif/arduino-esp32/issues/5064#issuecomment-822989386), but I'm not sure if this is the right way...

**arduino-esp32 (2.0.0-alpha1) is no longer compiled with esp-idf (release/4.3), even the `idf.py menuconfig` does not start...**
It is also no longer possible to repeat the @brandonros [issue](https://github.com/espressif/arduino-esp32/issues/5064#issuecomment-826488860), because arduino-esp32 (master) branches already contain the proposed fixes, but this has no effect.

Does anyone get to compile arduino-esp32 (???) with esp-idf (release/4.3) version without replacing 81 with 41?

ul-gh, 2021-06-17:
Am 17.06.21 um 12:44 schrieb Sterpa:
> Does anyone get to compile arduino-esp32 (???) with esp-idf
> (release/4.3) version without replacing 81 with 41?

Hi,

I guess it is best to stay with a fixed commit or tag as long as it works.

There should be a new bug report filed on arduino-esp32 with the title:
"please tag at least one version that actually compiles with current
esp-idf".

lbernstone, 2021-06-18:
me-no-dev builds it to what works.  It is up to mere mortals to backport his work into "older" aka stable branches.

aeljandro2, 2021-06-30:
I could compile changing the esp_eth_phy_new_ksz8081

lgLindstrom, 2021-09-22:
I have this working now. Internet is full of instructions about how to use arduino-esp in a esp-idf environment. As far as I can tell a lot of those instructions is either faulty and or outdateded.
This is my setup/workflow.

- Using VsCode I installed the esp-idf add-on.
- From Vscode I installed  esp-idf using the esp-idf add-on.
- From VsCode i created a new project using esp-idf add-on. I selected the  Aduino template.
- I followed this instruction https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html with some modifications.
```
//if not exists mkdir components
cd components && \
git clone https://github.com/espressif/arduino-esp32.git arduino && \
cd arduino && \
git checkout tags/2.0.0 && \
git submodule update --init --recursive && \
// no need to build ....
```

There you have it,, 
It works nicely from within VsCode using the esp-idf provided commands. 
To use components from the arduino-esp you must include arduino CmakeLists.txt

i```
df_component_register(
    SRCS ${sources}
    INCLUDE_DIRS ${include}
    REQUIRES arduino 
)
```

VojtechBartoska, 2021-09-22:
@lgLindstrom Do you consider to contribute to Arduino ESP32 online Documentation? We will appreciate your feedback and your modifications to make the docs more straight forward for users. 

Thanks!

k3rn3lp4n1c-1337, 2021-10-01:
Thx man this run fine by me with esp32 modules.
But when i use a project with the esp32c3 module i have a compiler error:
`../components/arduino/cores/esp32/libb64/cencode.c:46:20: warning: this statement may fall through [-Wimplicit-fallthrough=]
             result = (fragment & 0x003) << 4;
             ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
../components/arduino/cores/esp32/libb64/cencode.c:47:9: note: here
         case step_B:
         ^~~~
../components/arduino/cores/esp32/libb64/cencode.c:56:20: warning: this statement may fall through [-Wimplicit-fallthrough=]
             result = (fragment & 0x00f) << 2;
             ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
../components/arduino/cores/esp32/libb64/cencode.c:57:9: note: here
         case step_C:
         ^~~~
[942/1099] Building C object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/esp32-hal-uart.c.obj
../components/arduino/cores/esp32/esp32-hal-uart.c: In function 'uartStartDetectBaudrate':
../components/arduino/cores/esp32/esp32-hal-uart.c:507:17: warning: unused variable 'hw' [-Wunused-variable]
     uart_dev_t *hw = UART_LL_GET_HW(uart->num);
                 ^~
[952/1099] Building CXX object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/Esp.cpp.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
`

VojtechBartoska, 2022-03-29:
Hello, I'm closing this issue. Right now ESP-IDF v4.4 is supported for using Arduino as ESPIDF component. Related issue #5819.

closed 2022-03-29 by VojtechBartoska