espressif/arduino-esp32 #5068: BLE_Client crashing at "registerForNotify"

cards2022, 2021-04-17:
### Hardware:
Board:							M5 Stick C Plus (esp32 Pico) & M5 Stack Core 2 (esp32)
Core Installation version:			i'm not sure how to find this
IDE name:						Arduino IDE
Flash Frequency:					i'm not sure how to find this
PSRAM enabled:                                     i'm not sure how to find this
Upload Speed:				       1500000
Computer OS:                                        Windows 10

### Description:
Hello, I have come across a peculiar issue. I am using an M5 Stick C Plus (esp32 Pico) to send data using BLE to an M5 Stack Core 2 (esp32). When I use the examples BLE_notify and BLE_client to send one characteristic, everything works fine. The problem occurs when I try to send two characteristics - I get a "Guru Meditation Error". I am using a modified code of BLE_Client to receive data, and the exact same code works just fine when I use an Arduino to send two characteristics over BLE. The problem only arises when I try to send two characteristics with an esp32 Pico. However, I can read the data on my phone using a BLE app, so I am not sure what side of this is messed up. Below, I have added the 1) BLE Notify Code, 2) BLE Client Code, 3) BLE Client Code's Error Message, and 4) Exception Decoder Results. I have come across some similar issues on this forum, and tried a few things, but with no luck. What exactly does this mean? Is this an issue my BLE Notify Code or my BLE Client Code?

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```Modified BLE Notify

#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>

BLEServer* pServer;
BLECharacteristic* pCharacteristic1;
BLECharacteristic* pCharacteristic2;

bool deviceConnected = false;
bool oldDeviceConnected = false;
uint32_t value = 0;

float pitch = 0.0F;
float roll  = 0.0F;
float yaw   = 0.0F;

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID        "1101"
#define CHARACTERISTIC1_UUID "2101"
#define CHARACTERISTIC2_UUID  "3101"

class MyServerCallbacks: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
      deviceConnected = true;
    };

    void onDisconnect(BLEServer* pServer) {
      deviceConnected = false;
    }
};

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");

  // Create the BLE Device
  BLEDevice::init("ESP32");

  // Create the BLE Server
  pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());

  // Create the BLE Service
  BLEService *pService = pServer->createService(SERVICE_UUID);

  // Create a BLE Characteristic
  pCharacteristic1 = pService->createCharacteristic(
                      CHARACTERISTIC1_UUID,
                      BLECharacteristic::PROPERTY_READ   |
                     // BLECharacteristic::PROPERTY_WRITE  |
                      BLECharacteristic::PROPERTY_NOTIFY //|
                     // BLECharacteristic::PROPERTY_INDICATE
[... 246 lines cut ...]
```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d54bf  PS      : 0x00060330  A0      : 0x800d4dee  A1      : 0x3ffc8640  
A2      : 0x00000000  A3      : 0x3ffc867e  A4      : 0x00000002  A5      : 0x00000000  
A6      : 0x3ffd499c  A7      : 0x3f0186c0  A8      : 0x800d4b60  A9      : 0x3ffc85f0  
A10     : 0x3ffc863c  A11     : 0xf594a6b6  A12     : 0x3ffc52ac  A13     : 0xf594a6b6  
A14     : 0x0000004e  A15     : 0xff000000  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000030  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x400d54bf:0x3ffc8640 0x400d4deb:0x3ffc8670 0x400d20f2:0x3ffc86c0 0x400d218e:0x3ffc8720 0x400d7ec9:0x3ffc8760 0x400904f9:0x3ffc8780
```

``` Decoder
PC: 0x400d54bf: BLERemoteDescriptor::writeValue(unsigned char*, unsigned int, bool) at C:\Users\Andrew\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\1.0.7\libraries\BLE\src\BLERemoteDescriptor.cpp line 134
EXCVADDR: 0x00000030

Decoding stack results
0x400d54bf: BLERemoteDescriptor::writeValue(unsigned char*, unsigned int, bool) at C:\Users\Andrew\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\1.0.7\libraries\BLE\src\BLERemoteDescriptor.cpp line 134
0x400d4deb: BLERemoteCharacteristic::registerForNotify(void (*)(BLERemoteCharacteristic*, unsigned char*, unsigned int, bool), bool) at C:\Users\Andrew\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\1.0.7\libraries\BLE\src\BLERemoteCharacteristic.cpp line 473
0x400d20f2: connectToServer() at C:\Users\Andrew\Downloads\BLE_client/BLE_client.ino line 110
0x400d218e: loop() at C:\Users\Andrew\Downloads\BLE_client/BLE_client.ino line 166
0x400d7ec9: loopTask(void*) at C:\Users\Andrew\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\1.0.7\cores\esp32\main.cpp line 19
0x400904f9: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

chegewara, 2021-04-17:
It is crashing, because BLEClient is trying to write non existing 0x2902 descriptor.
You have 2 characteristics which have this property: `BLECharacteristic::PROPERTY_NOTIFY`, but no CCC descriptor in any.

cards2022, 2021-04-17:
thank you so much for the quick response. I am now sending a descriptor for each characteristic and it was working great. Thank you.

VojtechBartoska, 2021-04-21:
@cards2022 Can I consider it solved? :)

cards2022, 2021-04-22:
@VojtechBartoska yes, thank you.

cards2022, 2021-04-22:
@VojtechBartoska yes thank you

closed 2021-04-22 by cards2022