espressif/arduino-esp32 #5069: Initializing Serial in separate thread using `xTaskCreate...` does not work.

ketan, 2021-04-17:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.6
IDE name:						Arduino IDE and Platform.io
Flash Frequency:					40Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        MacOS Catalina

### Description:

I'm using an ESP32 board with ESP32 framework version `1.0.6`.

I'm trying to initialize and read from a serial port (gps NMEA sentences). However, to improve boot time, I'm initializing this in a separate thread at startup.

The code snippet below fails when using Serial1/Serial2 in a separate thread. Does anyone have a workaround/fix?

### Sketch

```cpp
#include <Arduino.h>

#define GPS_RX 32
#define GPS_TX 33
#define GPS_PORT Serial1

TaskHandle_t backgroundJob = NULL;

void backgroundTask(void* pvParameters) {
  GPS_PORT.begin(0, SERIAL_8N1, GPS_RX, GPS_TX);

  while (true) {
    if (GPS_PORT.available()) {
      Serial.print((char) GPS_PORT.read());
    }
  }
}

void setup() {
  Serial.begin(115200);
  Serial.printf("Beginning...");

  xTaskCreatePinnedToCore(backgroundTask,
                          "GPS",
                          1024 * 5,
                          NULL,
                          10,
                          &backgroundJob, 
                          0 // changing this to 1 does not work either
                         );
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Messages:
```
Beginning...[V][HardwareSerial.cpp:91] end(): pins 33 32
$GNRX,99.99,99.99,99.99*2E
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*2E
$GPGSV,1,1,00*79
$GLGSV,1,1,00*65
$GNGLL,,,,,,V,N*7A
$GNRMC,,V,,,,,,,,,,N*4D
$GNVTG,,,,,,,,,N*2E
$GNGGA,,,,,,0,00,99.99,,,,,,*56
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*2E
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*2E
$GPGSV,1,1,00*79
$GLGSV,1,1,00*65
$GNGLL,,,,,,V,N*7A
$GNRMC,,V,,,,,,,,,,N*4D
$GNVTG,,,,,,,,,N*2E
$GNGGA,,,,,,0,00,99.99,,,,,,*56
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*2E
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*2E
$GPGSV,1,1,00*79
$GLGSV,1,1,00*65
$GNGLL,,,,,,V,N*7A
$GNRMC,,V,,,,,,,,,,N*4D
$GNVTG,,,,,,,,,N*2E
$GNGGA,,,,,,0,00,99.99,,,,,,*56
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*2E
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*2E
$GPGSV,1,1,00*79
$GLGSV,1,1,00*65
$GNGLL,,,,,,V,N*7A
$GNRMC,,V,,,,,,,,,,N*4D
$GNVTG,,,,,,,,,N*2E
$GNGGA,,,,,,0,00,99.99,,,,,,*56
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*2E
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*2E
$GPGSV,1,1,00*79
$GLGSV,1,1,00*65
$GNGLL,,,,,,V,N*7A
$GNRMC,,V,,,,,,,,,,N*4D
$GNVTG,,,,,,,,,N*2E
$GNGGA,,,,,,0,00,99.99,,,,,,*56
$GNGSA,A,1,,,,,,,,,E (11765) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (11765) task_wdt:  - IDLE0 (CPU 0)
E (11765) task_wdt: Tasks currently running:
E (11765) task_wdt: CPU 0: GPS
E (11765) task_wdt: CPU 1: loopTask
E (11765) task_wdt: Aborting.
abort() was called at PC 0x400e1183 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x40084ea0:0x3ffbe4e0 0x40085115:0x3ffbe500 0x400e1183:0x3ffbe520 0x400837f9:0x3ffbe540 0x4000bfed:0x3ffb9700 0x40086f89:0x3ffb9710 0x400860ab:0x3ffb9730 0x400d18d1:0x3ffb9750 0x400d0ce9:0x3ffb9770 0x400d0ccd:0x3ffb9790 0x40086125:0x3ffb97c0

Rebooting...
Beginning...[V][HardwareSerial.cpp:91] end(): pins 33 32
                      $GNR	⸮)⸮r⸮⸮b⸮⸮r⸮⸮b⸮⸮r⸮⸮R⸮*5
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*2E
$GPGSV,1,1,00*79
$GLGSV,1,1,00*65
$GNGLL,,,,,,V,N*7A
$GNRMC,,V,,,,,,,,,,N*4D
$GNVTG,,,,,,,,,N*2E
$GNGGA,,,,,,0,00,99.99,,,,,,*56
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*2E
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*2E
$GPGSV,1,1,00*79
$GLGSV,1,1,00*65
$GNGLL,,,,,,V,N*7A
$GNRMC,,V,,,,,,,,,,N*4D
$GNVTG,,,,,,,,,N*2E
$GNGGA,,,,,,0,00,99.99,,,,,,*56
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*2E
$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*2E
$GPGSV,1,1,00*79
$GLGSV,1,1,00*65
$GNGLL,,,,,,V,N*7A
```

### Stack trace from decoder

```
Decoding stack results
0x40084ea0: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 156
0x40085115: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 171
0x400e1183: task_wdt_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c line 174
0x400d18d8: uartAvailable at /Users/ketanpadegaonkar/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/cores/esp32/esp32-hal-uart.c line 285
0x400d0ce9: HardwareSerial::available() at /Users/ketanpadegaonkar/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/cores/esp32/HardwareSerial.cpp line 116
0x400d0ccd: backgroundTask(void*) at /Users/ketanpadegaonkar/Documents/Arduino/gps-serial2-backgroundthread-bug/gps-serial2-backgroundthread-bug.ino line 13
0x40086125: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

ketan, 2021-04-17:
On another problem. I'm seeing an issue with calling `HardwareSerial.begin()` in the background thread:

```plain
Backtrace: 0x40081b72:0x3ffbfbe0 0x4008dfe9:0x3ffbfc10 0x4000bfed:0x3ffe5510 0x400923f9:0x3ffe5520 0x40164ef2:0x3ffe5540 0x40164f4f:0x3ffe5580 0x400dc84c:0x3ffe55b0 0x400dc913:0x3ffe55d0 0x400dcdf5:0x3ffe55f0 0x400da715:0x3ffe5630 0x400d695e:0x3ffe5670 0x400d6ac9:0x3ffe56a0 0x400d6b3c:0x3ffe56c0 0x40091276:0x3ffe56f0
  #0  0x40081b72:0x3ffbfbe0 in _uart_isr at /Users/ketanpadegaonkar/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:277
  #1  0x4008dfe9:0x3ffbfc10 in _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
  #2  0x4000bfed:0x3ffe5510 in ?? ??:0
  #3  0x400923f9:0x3ffe5520 in vTaskExitCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:4813
  #4  0x40164ef2:0x3ffe5540 in esp_intr_alloc_intrstatus at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/intr_alloc.c:668
  #5  0x40164f4f:0x3ffe5580 in esp_intr_alloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/intr_alloc.c:688
  #6  0x400dc84c:0x3ffe55b0 in uartEnableInterrupt at /Users/ketanpadegaonkar/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:277
  #7  0x400dc913:0x3ffe55d0 in uartAttachRx at /Users/ketanpadegaonkar/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:277
  #8  0x400dcdf5:0x3ffe55f0 in uartBegin at /Users/ketanpadegaonkar/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:219
  #9  0x400da715:0x3ffe5630 in HardwareSerial::begin(unsigned long, unsigned int, signed char, signed char, bool, unsigned long) at /Users/ketanpadegaonkar/.platformio/packages/framework-arduinoespressif32/cores/esp32/HardwareSerial.cpp:190
  #10 0x400d695e:0x3ffe5670 in GPSSupport::initializeSerialPort() at lib/gps_support/gps_support.cpp:328
  #11 0x400d6ac9:0x3ffe56a0 in GPSSupport::setupInBackgroundJob() at lib/gps_support/gps_support.cpp:41
  #12 0x400d6b3c:0x3ffe56c0 in GPSSupport::backgroundTask(void*) at lib/gps_support/gps_support.cpp:32
  #13 0x40091276:0x3ffe56f0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

felmue, 2021-04-17:
Hello @ketan 

I see the watchdog getting triggered as well. Try to add a `delay(10);` in the `while()` loop of the task - this should reset the watchdog. (At least it works for me.)

Thanks
Felix

closed 2021-07-11 by stale[bot]