espressif/arduino-esp32 #5070: WiFiClientSecure doesn’t work in 1.0.5 and later

rezaneam, 2021-04-17:
----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 Pico Kit
Core Installation version:			1.0.5 (and later)
IDE name:						 Arduino IDE and Platform.io ([espressif32@3.1.0 and later](https://github.com/platformio/platform-espressif32))
Flash Frequency:					40Mhz
PSRAM enabled:                                    No
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
I recently upgraded my ESP32 platform to `1.0.5` () and noticed that WiFiClientSecure doesn’t work properly.
I see this error in the console when trying to connect to a socket.

`[E][WiFiClientSecure.cpp:127] connect(): start_ssl_client: -1`

I downgraded the project to 1.0.4 and works just fine.
Kind regards, Reza

Please find the sample code written in Platform.io
https://github.com/rezaneam/temporary

dralisz82, 2021-04-17:
Hi,

It worked for me in 1.0.5-rc2, but I've just upgraded to 1.0.6 and see that https fails for me, too.
**Good news, that I know the solution! ;)**

I assume you also try to connect to a https server without the need to verifying its certificate, so you also don't provide a const char* CAcert.
In older versions, if you skipped CAcert, underlying WifiClientSecure's setInsecure() function was called.
In latest releases this get disabled:
`        WiFiClientSecure& wcs = static_cast<WiFiClientSecure&>(client);
        if (_cacert == nullptr) {
            //wcs.setInsecure();
        } else {
            wcs.setCACert(_cacert);
            wcs.setCertificate(_clicert);
            wcs.setPrivateKey(_clikey);
        }
`

Way to restore the functionality:
Go to line 78 in HTTPClient.cpp and uncomment `wcs.setInsecure();`
That's it! Good luck! ;)

BR,
Szilárd

rohitsam, 2021-04-19:
I have the same problem, I 'm using pubsub client and Wifisecure client to make a mqtt connection, there was no issue when using ver 1.04.
but when using 1.0.5 and 1.0.6 i get this error.

[E][WiFiClientSecure.cpp:133] connect(): start_ssl_client: -1

rohitsam, 2021-04-19:
> Hi,
>
> It worked for me in 1.0.5-rc2, but I've just upgraded to 1.0.6 and see that https fails for me, too.
> **Good news, that I know the solution! ;)**
>
> I assume you also try to connect to a https [...]

wcs.setInsecure() worked for me.
Thanks.

rezaneam, 2021-04-19:
Thank you for your feedback. adding `setInsecure();` fixed the issue.

closed 2021-04-21 by VojtechBartoska