espressif/arduino-esp32 #5071: Compile errors for ESP32-C3

Jason2866, 2021-04-17:
Hi great work for supporting the C3!!

I encountered two errors from core when trying to compile project Tasmota for the C3
1st:
https://github.com/espressif/arduino-esp32/blob/master/tools/platformio-build-esp32c3.py#L74
it seems not supported from riscv32 gcc. Removing did solve
2nd:
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/Esp.h#L114
not working for C3

@me-no-dev can you take a look?

Thx Jason

me-no-dev, 2021-04-17:
@Jason2866 I did not encounter any build errors. Flags are extracted from ESP-IDF's build. As for some things not working... yeah :) still in early alpha

Jason2866, 2021-04-17:
Changing too
```
{
    uint32_t ccount;
    //__asm__ __volatile__("esync; rsr %0,ccount":"=a" (ccount));
    asm volatile ("rdcycle x1" : "=r" (ccount) );
    return ccount;
}
```
in Esp.h solved

lbernstone, 2021-04-17:
That change crashes my C3.  Does `systimer_hal_get_counter_value(SYSTIMER_COUNTER_0);` from hal/systimer_hal.h give you the correct value?  This should ultimately work for all the devices once the hal is functional.

Jason2866, 2021-04-17:
Yes, you are right. Thx for your suggestion. Looks better now. Now crashes later ;-)

commit b45cf11f (2021-04-19): Fixed ESP::getCycleCount for S2/C3 (#5075)

Fixes #5071
closed 2021-04-19 by me-no-dev