espressif/arduino-esp32 #5080: webserver :Please add   chunkedResponseModeStart /  chunkedResponseFinalize

fanfanlatulipe26, 2021-04-19:
### Description:
webserver :Please add   chunkedResponseModeStart /  chunkedResponseFinalize
These functions were added in  a recent release for ESP8266WebServer and it will be nice to have them in the upcomming release for ESP32. If not , porting from ESP8266 to ESP32 becomes cumbersome.
I  developed a small derivated class to get them in rhe ESP32 environment. If it can help .....

### Sketch: 
```cpp

#ifndef FS_WEBSERVER_H
#define FS_WEBSERVER_H

#include <WebServer.h>
class fs_WebServer : public WebServer
{
  public:
    fs_WebServer(int port = 80): WebServer(port) {
    }

    void send_P(int code, PGM_P content_type, PGM_P content) {
      size_t contentLength = 0;

      if (content != NULL) {
        contentLength = strlen_P(content);
      }

      String header;
      char type[64];
      memccpy_P((void*)type, (PGM_VOID_P)content_type, 0, sizeof(type));
      _prepareHeader(header, code, (const char* )type, contentLength);
      _currentClientWrite(header.c_str(), header.length());
      if (contentLength) {  // if rajouté par FS ...........................+++++
        sendContent_P(content);
      }
    }

    bool chunkedResponseModeStart_P (int code, PGM_P content_type) {
      if (_currentVersion == 0)
        // no chunk mode in HTTP/1.0
        return false;
      setContentLength(CONTENT_LENGTH_UNKNOWN);
      send_P(code, content_type, "");
      return true;
    }
    bool chunkedResponseModeStart (int code, const char* content_type) {
      return chunkedResponseModeStart_P(code, content_type);
    }
    bool chunkedResponseModeStart (int code, const String& content_type) {
      return chunkedResponseModeStart_P(code, content_type.c_str());
    }
    void chunkedResponseFinalize () {
      sendContent(emptyString);
    }
};
#endif // FS_WEBSERVER_H
```

VojtechBartoska, 2021-04-21:
Thanks @fanfanlatulipe26 . We will take a look and consider implementation.

tablatronix, 2022-01-23:
Status of this?

fanfanlatulipe26, 2022-02-06:
I just updated the code of the derivative class
   using WebServer:: send_P;
was added.

FrankBoesing, 2023-05-17:
Great news!

AlefRosa, 2023-05-20:
I really need this, I need to answer a gigantic (100mb) text file that is dynamically generated from memory. With esp8266 I sent with sendContent_P a few lines at a time and it worked great. I can't do this with esp32.
Thanks.

fanfanlatulipe26, 2023-05-20:
@AlefRosa 
 In the mean time, why don't you use my trick I gave at the beginning: just include in your sketch a file fs_Webserver.h containing the code I gave above and declare you server as fs_Webserver.
You can see the use in the project [BaliseDGAC_GPS_Logger](https://github.com/fanfanlatulipe26/BaliseDGAC_GPS_Logger) where you will find the file fs_Webserver.h and the server declaration and use in the .ino

FrankBoesing, 2023-05-21:
Do you know how to use the  RequestHandler (server->addHandler(new ... ) with your derived class?

FrankBoesing, 2023-05-21:
An other missing method is:

void send(int code, const char \*content\_type, const char \*content, _**size\_t contentLength**_)

Without, it uses an often superlfues strlen(), which is not optimal (esp with large content)

fanfanlatulipe26, 2023-05-21:
Sorry, I never used RequestHandler . I had a look at the source code for RequestHandler .h and the ESP8266 version is really more C++ that the ESP32 one. I must admit that I am not C++ expert ... Didn't find any examples.

AlefRosa, 2023-05-21:
@fanfanlatulipe26 thank you very much, i got it yesterday. As I need to use asyncwebserver, I studied it further and managed to use request response using chunk method. thank you so much again! Now I'm managing to generate 10mb files from memory registers and send them in parts as I make them.

chconnor, 2024-01-22:
Also eagerly awaiting this important feature!

Thanks @fanfanlatulipe26 -- that code worked great. I had to add an additional constructor at the top:

`    fs_WebServer(IPAddress addr, int port): WebServer(addr, port){ }
`

fanfanlatulipe26, 2024-01-24:
Thanks @chconnor for the tip. I updated the code in the repository [BaliseDGAC_GPS_Logger](https://github.com/fanfanlatulipe26/BaliseDGAC_GPS_Logger)  that I referenced above. It may help some one.

FarokhReza, 2024-02-17:
Thanks for sharing it is great for solving some issues in the progress of migration from esp8266 to esp32!!

VojtechBartoska, 2024-02-20:
Postponing to 3.1.0 Milestone, we will cover this during Webserver refactoring

Parsaabasi, 2025-01-16:
Hello, 

Due to the overwhelming volume of issues currently being addressed, we have decided to close the previously received tickets. If you still require assistance or if the issue persists, please don't hesitate to reopen the ticket.

Thanks.

chconnor, 2025-01-17:
@VojtechBartoska -- did you folks happen to get this done in the Webserver refactoring, or should it be reopened? (I didn't see it in the closed issues for the milestones linked above, or for 3.2.0.)

chconnor, 2025-01-20:
I have confirmed that WebServer.h still does not have this included. I don't have permission to reopen this issue -- perhaps someone else can?

commit 9d44061c (2025-10-08): feat(webserver): add support for chunked HTTP responses Fixes #5080 (#11894)

* Add chunkedResponseModeStart, sendChunk, chunkedResponseFinalize
to WebServer library for chunked HTTP responses

* feat(webserver): Chunk example for ESP32 HTTP server

This example demonstrates how to send an HTTP response using chunks with an ESP32 server. It includes setup for WiFi, MDNS, and handles HTTP requests with chunked responses.

* feat(webserver): Add ci.json for ChunkWriting example configuration

* fix(example): not necessary code line

* ci(pre-commit): Apply automatic fixes

---------