espressif/arduino-esp32 #5083: why the code start with #if 0

chenguoping76, 2021-04-19:
CameraWebServer/CameraWebServer.ino
//Disable Example for now
#if 0

when change '0' to '1' enable this example, can't compile it, who can resolute it ?

Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3?
IDE name:						?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					?115200?
Computer OS:                                        ?Windows 10? ?Mac OSX? ?Ubuntu?

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

lbernstone, 2021-04-19:
If you are using the experimental master branch (IDF 4), then the camera driver is not currently working.  When it does, that block will be removed.  You can run the [example](https://github.com/espressif/arduino-esp32/blob/1.0.6/libraries/ESP32/examples/Camera/CameraWebServer/CameraWebServer.ino) on the stable version (1.0.6).

VojtechBartoska, 2021-04-21:
Thanks @lbernstone for answer. Can I consider it solved @chenguoping76 ?

chenguoping76, 2021-04-24:
got, it is solved

closed 2021-04-26 by VojtechBartoska